openwrt/openwrt #9963: Impossible to connect my S21 Ultra to wifi wrt1900ac-v2 mixed WPA2/WPA3 PSK, SAE (CCMP) since upgrading to 22.03 (tested rc1 and rc3)

zuzzurro, 2022-05-29:
Since upgrading to 22.03 I cannot connect my S21 Ultra to my wifi using a Linksys wrt1900ac-v2 (it worked fine in in all 21.02 releases).

There's an endless series of connection attempts resulting in the following line in the logs, but never succeeding.

```
Sun May 29 19:47:05 2022 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.10.10.162 xx:xx:xx:xx:xx:xx S21-Ultra
Sun May 29 19:47:14 2022 daemon.notice hostapd: SAE: xx:xx:xx:xx:xx:xx indicates support for SAE H2E, but did not use it
Sun May 29 19:47:14 2022 daemon.notice hostapd: SAE: xx:xx:xx:xx:xx:xx indicates support for SAE H2E, but did not use it
Sun May 29 19:47:25 2022 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Sun May 29 19:47:25 2022 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)
Sun May 29 19:47:25 2022 daemon.notice hostapd: wlan1: Prune association for xx:xx:xx:xx:xx:xx
Sun May 29 19:47:27 2022 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Sun May 29 19:47:27 2022 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)
Sun May 29 19:47:27 2022 daemon.notice hostapd: wlan1: Prune association for xx:xx:xx:xx:xx:xx
Sun May 29 19:47:29 2022 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Sun May 29 19:47:29 2022 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)
Sun May 29 19:47:29 2022 daemon.notice hostapd: wlan1: Prune association for xx:xx:xx:xx:xx:xx
```

Let me know how I can help troubleshooting this.

pkgadd, 2022-05-29:
mwlwifi is known to be broken with IEEE802.11w and WPA3/SAE, disable it completely in favour of the older WPA2/CCMP. Given the driver/ firmware situation, this is not going to improve.

zuzzurro, 2022-05-29:
But is this a regression? As I said it worked perfectly in 21.02.

zuzzurro, 2022-05-29:
Anyway, I can confirm that moving to WPA2 brings the wireless back to life. 😢

pkgadd, 2022-05-29:
> But is this a regression?

That is something only you can confirm (I don't have any mwlwifi hardware), ideally by git-bisecting between 21.02 and 22.03~. Marvell/ NXP has killed off support for these chipsets, it never went mainline and development is dead - if you can pin point which change 'broke' it, someone might be able to figure out why and eventually fix it (chances are that it's in the proprietary firmware though), but it would need your debugging/ bisecting.

That aside, WPA3 is known to be broken in mwlwifi, up to crashing the wireless core, so I wouldn't expect much or anything in this regard - even if it might have worked in 21.02.x 'by accident'.

zuzzurro, 2022-06-15:
Just a heads up. I tested it with rc4 and it still behaves the same. I think it may be time to start looking for a new router...