openwrt/openwrt #9981: firewall4: design shortcomings in fw4 custom filter chains

Shine-, 2022-06-02:
### Problem description:

fw4 custom filter chains ("10-custom-filter-chains.nft" or "*.nft") have some logic/design issues.
Below are two example cases, and I'm sure there are more.

### Case 1:

Background: Firewall rules should always be processed from top to bottom. If one rule is matched, processing should end, so the matched rule can come into effect.

This works fine as long as you remain within **only one nft chain**.

As soon as you add additional chains, nft behaves differently. A "Drop" rule in any chain will render all "Allow" rules in all previous (lower priority) chains void. This is due to nft's behavior to continue processing any later chains, even if a rule matches, **until** a Drop is found. So the "top to bottom" approach suddenly applies only to "Drop" rules, not to "Allow" rules anymore.

Consequently, this renders fw4's "*.nft" files - which can add additional chains, but not any additional rules in the default chain - quite useless, because they can only be used for "Drop" rules, not for "Allow" rules.

### Problem for Case 1:

Due to the nft behavior with multiple chains, at this time, the user doesn't have a possibility to add custom "Allow" rules to fw4.

This especially prohibits the user from configuring fw4 like a proper network firewall should be configured: Deny everything unless explicitly allowed - since the "Deny all" last rule will override any previous "Allow" rules from any lower priority chains!

### Case 2:

Background: In a stateful firewall, flows that are already handled by a firewall rule should not be sent through the whole rulebase processing again, as this would mean wasting CPU cycles and therefore slowing down firewall performance.

This works fine, as long as the default fw4 chains are the **only** chains used. These chains contain an "Accept" rule for "established,related".

As soon as the user adds additional chains (without own "established,related" rules), these complete chains are processed for every packet. As already described in Case 1, this re-processing will repeat for each and every chain, as an "Accept" rule will not stop processing later chains.

### Problem for Case 2:

By adding simple additional rules as separate nft chains, performance will suffer.

### Suggestion:

"*.nft" files aren't currently useful for the purpose one would suppose: quickly adding additional firewall rules that aren't supported in uci/LuCI syntax.

While Case 2 is just "inconvenient", but can be solved by adding "established,related" rules to each chain, I don't see how Case 1 can be solved in the current state of fw4 at all.

The possibility of adding complete additional nft chains might be useful for special purposes. It is, however, not a feasible way of quickly adding custom firewall rules. Therefore, it should not be the **only** possible way of customizing the fw4 ruleset.

Rather, the user should be given an easy possibility of customizing the fw4 default chains, without having to think of all side effects and limits that he might have when using separate nft chains.

After all, the standard OpenWRT user, or even the OpenWRT Power User, is not a nft expert!

spolack, 2022-06-04:
We are also affected by that issue. Our usecase is, that we are running multiple routers in a active-active setup, while we have a inbound firewall, which connection tracking table needs to synced across the gateways. Since conntrack-sync is constrained for near destinations (because the first TCP SYN-ACK might arrive before the state is synced, we are allowing this kind of packets rate-limited.

Currently this is archieved by the following:

```
root@saarbruecker-gw:~# cat /etc/firewall.user 
# Allow established packets (hopefully) belonging to flows which are not being synced yet (rate limited to prevent DoS)
ip6tables -A forwarding_uplink_rule -p tcp -m tcp --tcp-flags ACK ACK -m limit --limit 5000/sec --limit-burst 2500 -m comment --comment "Allow established connections which not made it in the conntrack sync yet" -j ACCEPT
ip6tables -A forwarding_uplink_rule -p tcp -m tcp --tcp-flags RST RST -m limit --limit 1000/sec --limit-burst 500 -m comment --comment "Allow established connections which not made it in the conntrack sync yet" -j ACCEPT

# Provide a rule to lookup the packets which were dropped because they were more than the limit.
# lookup via `ip6tables -nvL | grep exceeded`
ip6tables -A forwarding_uplink_rule -p tcp -m tcp --tcp-flags ACK ACK -m comment --comment "Limit exceeded ACK"
ip6tables -A forwarding_uplink_rule -p tcp -m tcp --tcp-flags RST RST -m comment --comment "Limit exceeded RST"
```

jow-, 2022-06-04:
Shouldn't it just be a matter of adding more includes to various places of the default ruleset?

dave14305, 2022-06-04:
There was some discussion in the forums about alternative user includes.

https://forum.openwrt.org/t/firewall4-nftables-tips-and-tricks/113704/39

PolynomialDivision, 2022-06-05:
> To quickly test this, you can edit `/usr/share/ucode/firewall4/templates/ruleset.uc` on the device and add further include statements in the various places you need them. Ideally we should group those in a directory hierarchy like

Can you give me a more detailed explanation? The file `/usr/share/ucode/firewall4/templates/ruleset.uc` is not existing?

Strasedon, 2022-06-05:
> Can you give me a more detailed explanation? The file `/usr/share/ucode/firewall4/templates/ruleset.uc` is not existing?

Try looking at `/usr/share/firewall4/templates/ruleset.uc`.

notgood, 2022-06-05:
> * others which I can't think of right now

Ability to include rules outside of `table inet` context would be great as well. Things like `table bridge`, `table arp` and `table netdev`.

Here is a usecase [example](https://forum.openwrt.org/t/firewall4-nftables-rules-for-android-ipv6-clients/128295/10?u=notgood)

dave14305, 2022-06-05:
Start by saving the output of `fw4 print` and decide where you would want to insert new `include` statements (e.g. at top and bottom of a chain). Once you know where you want them, updating ruleset.uc isn’t as daunting, especially if it’s just adding the literal include text and nothing has to be interpreted by ucode.

jow-, 2022-06-06:
> Why so complicated? Just add "include" uci directives like fw3 had (just not for shell scripts but for nft rulesets). 

Because I dislike the concept of having to specify a native include file in uci. Packages that want to amend the firewall should be able to drop includes into predefined locations and not be required to programmatically modify /e/c/firewall to register their includes (which will destroy comments, change formatting and potentially commit uncommitted changes without the users consent)

> It should be enough to define empty pre- and post- chains (with predictable names) by fw4 and jump to them from standard chains. 

I really would like to avoid that approach. That would almost triple the amount of chains in the default ruleset and bloat it with stuff which is not needed by the majority of users.

Shine-, 2022-06-06:
> Because I dislike the concept of having to specify a native include file in uci. Packages that want to amend the firewall should be able to drop includes into predefined locations and not be [...]

Wouldn't a package that just drops firewall rulesets into a directory, which are active immediately - not only without user consent, but without the user **even knowing** - be much much worse?

When someone's trying to figure out why something isn't working correctly (or wants to get rid of the remnants of a package), he does a `uci show` - in which the altered firewall rule set wouldn't even show up. Nor would there be any other hint for the user to see that someone or something modified his firewall...

Troubleshooting nightmare, security nightmare...

jow-, 2022-06-06:
> Wouldn't a package that just drops firewall rulesets into a directory, which are active immediately - not only without user consent, but without the user even knowing - be much much worse?

I don't think so. The user needs to be aware of it either way and opt-out manually, regardless of whether the package adds an uci include (which destroys all user comments, reformats the firewall config and might commit changes not meant to be committed) or a nft include file (which does none of the above).

Another problem with uci-configured includes is that they're unreliable. If a user restores an old version of the firewall configuration without required includes, services relying on them will not work (e.g. miniupnpd for its chain setup).

It boils down to the fact that (package provided) includes are not user configuration but implementation details. If you don't want upnp functionality, don't install the miniupnpd package. Installing miniupnpd but not wanting it to automatically manage the firewall integration is a niche use-case, and users wanting that are unlikely to run fw3 or fw4 in the first place. If for whatever reason you do run fw3/fw4 and do not want automatic integration of miniupnpd with the firewall you would still be able to delete it's `/etc/nftables.d/whatever.nft` file.

In the end you do have to trust the packages you install from the feeds. If you don't trust them, you can't trust them to use uci includes either, you will have to inspect all potential scripting places for manual rule injection anyway.

> When someone's trying to figure out why something isn't working correctly (or wants to get rid of the remnants of a package), he does a uci show - in which the altered firewall rule set wouldn't [...]

I don't think that this is an fw4 specific issue. Packages could also stage rules through different means, e.g. by shipping custom init scripts or hotplug handlers invoking manual iptables (now nft commands). Those would not show up in uci either. And even worse, due to that you don't even have a common well known place to look for such things, you have to inspect any file a package is shipping anyway.

There's also the ability for procd services to stage "virtual" firewall rules through ubus service data objects, those do not show up in uci as either.

Shine-, 2022-06-07:
> [...] (which destroys all user comments, reformats the firewall config and might commit changes not meant to be committed) [...]

Well, with that logic you are voting for complete removal of uci from OpenWRT. Back to myriads of config files, cluttered all around the file system. Welcome to the Linux disease. :-(

Shine-, 2022-06-07:
Becoming more constructive again. Sorry, but couldn't resist. Having all config settings in one place with uci, is **the** reason why I'm using OpenWRT, not a random standard distro on a random SBC.

How about a 2-way approach:
1. Something like `firewall.defaults[0].autoinclude=1` to enable automatic includes from `/etc/nftables.d/`.
2. `firewall.include[x]` like fw3, just with nft rule sets, not shell scripts. Or even, with the option to choose between shell script, run at the end of fw4 start (like `firewall.user`), or a nft rule set, included before/after fw4 section X or Y.

Best of both worlds? You could have your auto-includes, and I can disable them if I want.

notgood, 2022-06-07:
> myriads of intransparent config files,

Sorry, have to disagree. In my own experience, having to manage tens of various hosts, mostly automatic with ansible, "/etc/config.d/" approach is a blessing.

You can rely on upstream project and distribution to provide sensible, secure and tested config file default and slightly adjust/override it according to your needs with "config.d" snippets. And distribution can push upgrades to main configuration, without messing up your adjustments.

Just a quick grep from my ansible projects, dropping snippets in these directories. I would go crazy writing direct config file modifications code, if these didn't exist.
```
/etc/profile.d/
/etc/sysctl.d/
/etc/sudoers.d/
/etc/modules-load.d/
/etc/modprobe.d/
/etc/ssh/sshd_config.d/
/etc/udev/rules.d/
/etc/systemd/*.conf.d/
```

Shine-, 2022-06-07:
> > myriads of intransparent config files,
> 
> Sorry, have to disagree. In my own experience, having to manage tens of various hosts, mostly automatic with ansible, "/etc/config.d/" approach is a blessing.

Let's not get off-topic here.

We're **not** talking about Linux/UNIX servers that you are performing fully automated, professional management on, using ansible, salt stack, puppet or whichever management automation you can think of.

We're talking about OpenWRT and uci. About manual configuration from the command line or web interface. About a single device, managed by a single, non-professional person with only basic UNIX knowledge.

So again, please let's not drift off-topic. This is OpenWRT, uci and firewall4.

champtar, 2022-06-07:
@Shine- why should OpenWrt not adopt what works for other distros ?
Example, packages ship their firewall conf in `/usr/lib/fw4.d/mypkg.nft`, and if you want to override it you create `/etc/fw4.d/mypkg.nft`

jow-, 2022-06-15:
Explicit include support added in https://git.openwrt.org/11256ff037 - package bump will follow later.

stintel, 2022-06-20:
Bumped in e8433fb4336b72e44a40cb667ee2eb06e0a31109. Please test and close this issue or give feedback.

ptpt52, 2022-06-21:
> Explicit include support added in https://git.openwrt.org/11256ff037 - package bump will follow later.

seems not respected on `option reload 0` `option family any` or something like in the firewall include rule-set

jow-, 2022-06-21:
Yes, this is expected. nftables (fw4) does not differentiate between reloads and restarts anymore and the family is meaningless as the ruleset generation is af agnostic, so it is expected that these options are not supported.

PolynomialDivision, 2022-06-21:
Can someone drop an example how it works? I tried multiple things, however fw4 does not show any prepend nor append of my chains. I tried to add them under `/etc/nftables.d/table-prepend`?

PolynomialDivision, 2022-06-21:
Thanks. :) I thought that the chain includes to a table would also be directly jumped at from the chain that hooks into it. So a table include needs also to hook itself to be executed (or I insert a jump with a chain include)?

jow-, 2022-06-21:
Yes. These includes are simply pasting the contents of the include files verbatim into the set to be fed to the nft parser, you do need to take care of adding the appropriate jumps (potentially through another include). What do you try to achieve exactly?

Shine-, 2022-06-23:
@jow- 
Thank you very much, and especially thanks for the example above, since there wasn't any documentation in the commits. Just came back from vacation, will test it asap and report back.

On an side note, also **many many** thanks for the `option log` - I was missing that forever, in fw3.
Would there be a possibility to also add a global option `option log_unhandled` for all traffic that doesn't match any rule and therefore receives default behavior (drop/reject/accept)? I know such option from some iptables-based pro firewalls, it massively helps with troubleshooting, sometimes.

notgood, 2022-06-30:
Cherry-picked package bump commit into 22.03 build and tested "ruleset-append" include, together with a several nftables.d/*.nft snippets.

Working as expected, thank  you!

Shine-, 2022-07-14:
I'm too busy with work and real life right now to test it out myself thoroughly. Since others have tested and confirmed it's working as expected, and the updates have also landed in 22.03-rc5 already, I'm closing this for now and consider it fixed.
@jow- Thanks again for your great work!

closed 2022-07-14 by Shine-