openwrt/openwrt #9982: Firewall4

jhthomas, 2022-06-02:
As noted previously by user "debaba" there is an extra space in the dependency entry file for Firewall4 that prevents the dependency check from working properly and thus Luci fails to install the new version.   The extra space occurs in the phrase '> = 20220322' (note extra space between > and =) when it should be '>= 20220322' (no space).  Thanks.

dave14305, 2022-06-02:
What branch and version are you seeing this on? Snapshot, 22.03-Snapshot, 22.03-rc1, etc?

dave14305, 2022-06-02:
Link to referenced issue:

https://github.com/openwrt/asu/issues/345

jhthomas, 2022-06-02:
22.03.0-rc3

jhthomas, 2022-06-02:
I should probably mention that I'm on a Linksys WRT32X and that this
happens on 22.03-rc1 as well as 22.03-rc3.  (Did not try rc2).

dave14305, 2022-06-02:
Is this what you see in LuCI when attempting to upgrade firewall4? Or somewhere else?

Description
This package provides an nftables-based implementation of the UCI firewall sharing the same configuration format.

The installed version of package ucode is not compatible, require > = 2022-03-22 while 2022-05-20-081871e1-1 is installed.

dave14305, 2022-06-02:
Moving the `>` after the `>=` in this line 574 of `/www/luci-static/resources/view/opkg.js` seems to fix the issue for me.
Old:
```
if (deps[i].match(/^(.+)\s+\((<=|<|>|>=|=|<<|>>)(.+)\)$/)) {
```
New:
```
if (deps[i].match(/^(.+)\s+\((<=|<|>=|>|=|<<|>>)(.+)\)$/)) {
```

jhthomas, 2022-06-03:
A very logical update however, it didn't work for me.   I edited the .js
file, rebooted, and refreshed the software packages.   Same result as
before.

The new line (that didn't work) reads:
if(deps[i].match(/^(.+)\s+\((<=|<|>=|>|=|<<|>>)(.+)\)$/)){dep=RegExp.$1.trim();
no other changes to the file.

dave14305, 2022-06-03:
Did the change survive the reboot? No need to reboot after making the change. Just refresh the Software page in LuCI. I've tested with Upgrading firewall4 since it's already installed.

jhthomas, 2022-06-03:
Yes, the changes survived the reboot.  I tried the .js changes
several times with increasing levels of refresh.  The most complete cycle
was the one I quoted.  Bottom line: I have confirmed the change in the .js
file, reset in various ways including reboot, and finally refreshed the
software before trying the update.

Again, to be precise, I am stopping the install at the warning about
incompatible versions.  I am not actually taking the next step which would
be to install the Firewall4 update, anyway.

dave14305, 2022-06-03:
Try an incognito/private browser window, to avoid any caching issues.

aparcar, 2022-06-03:
I can't find the mentioned issue anywhere, can you point me to actual code please?
https://github.com/openwrt/openwrt/commit/0481a5a35a72177495ad32b69c0067d65809c223

dave14305, 2022-06-03:
> I can't find the mentioned issue anywhere, can you point me to actual code please?
> [0481a5a](https://github.com/openwrt/openwrt/commit/0481a5a35a72177495ad32b69c0067d65809c223)

https://github.com/openwrt/luci/blob/24723a006482c3b664ddffe0f2752386085cbe0f/applications/luci-app-opkg/htdocs/luci-static/resources/view/opkg.js#L574

aparcar, 2022-06-03:
All right so then this is a LuCI issue and should move over there, right?

closed 2022-06-03 by jow-

jhthomas, 2022-10-11:
> completed via ***@***.***

I flashed a fresh copy of -rc3 on a backup router, manually updated the .js
file with the changes described by Jo-Philipp Wich and it worked.  The
Firewall4 upgrade went through the Luci filters without complaint.   So the
changes look good to the common user.  Thanks.