openwrt/openwrt #9983: latest snapshot `/etc/config/fstab` config ignored - not mounting any devices

mwoz123, 2022-06-02:
I've just (attended) system upgrade for Netgear xr500 to latest snapshot (from previous Snapshot (mid-may)) version:
```
 OpenWrt SNAPSHOT, r19726-b21b98627d
```

I was stuggling with Luci Mount Points not detecting devices, but in console `blkid` did.
Manual mounting also works. 

[image]

Searching for similar issue found [exactly the same issue](https://forum.openwrt.org/t/error-loading-shared-library-libubus-so-20210215-when-trying-to-run-block/88909) but for different version of libubus. Mine is `libubus.so.20220601` 

My log:
```
root@OpenWrt:~# blkid info
root@OpenWrt:~# block-mount
ash: block-mount: not found
root@OpenWrt:~# opkg install block-mount
Package block-mount (2022-06-02-93369be0-1) installed in root is up to date.
root@OpenWrt:~# block info
Error loading shared library libubus.so.20220601: No such file or directory (needed by /sbin/block)
Error relocating /sbin/block: ubus_free: symbol not found
Error relocating /sbin/block: ubus_connect: symbol not found
Error relocating /sbin/block: ubus_invoke_fd: symbol not found
Error relocating /sbin/block: ubus_lookup_id: symbol not found
root@OpenWrt:~# block mount
Error loading shared library libubus.so.20220601: No such file or directory (needed by /sbin/block)
Error relocating /sbin/block: ubus_free: symbol not found
Error relocating /sbin/block: ubus_connect: symbol not found
Error relocating /sbin/block: ubus_invoke_fd: symbol not found
Error relocating /sbin/block: ubus_lookup_id: symbol not found
root@OpenWrt:~# opkg list-installed | grep ubus
libubus-lua - 2022-02-28-584f56a2-2
libubus20220228 - 2022-02-28-584f56a2-2
ubus - 2022-02-28-584f56a2-2
ubusd - 2022-02-28-584f56a2-2
ucode-mod-ubus - 2022-05-31-ae62d722-1
uhttpd-mod-ubus - 2022-02-07-2f8b1360-1
root@OpenWrt:~# opkg list-installed | grep block-mount
block-mount - 2022-06-02-93369be0-1
root@OpenWrt:~# ls -rtal /sbin/libubus*
ls: /sbin/libubus*: No such file or directory
root@OpenWrt:~# ls -rtal /sbin/block*
-rwxr-xr-x    1 root     root         36875 Jun  2 14:50 /sbin/block
root@OpenWrt:~# block-mount
ash: block-mount: not found
```

I hope /etc/fstab should work as workaround replacement for /etc/config/fstab. I'll let know.

mwoz123, 2022-06-02:
btw I forgot to paste - marked as solution in about link didn't work:

```
root@OpenWrt:~# opkg install libubus
Package libubus20220228 (2022-02-28-584f56a2-2) installed in root is up to date.
root@OpenWrt:~# block info
Error loading shared library libubus.so.20220601: No such file or directory (needed by /sbin/block)
Error relocating /sbin/block: ubus_free: symbol not found
Error relocating /sbin/block: ubus_connect: symbol not found
Error relocating /sbin/block: ubus_invoke_fd: symbol not found
Error relocating /sbin/block: ubus_lookup_id: symbol not found
root@OpenWrt:~# 
```

mwoz123, 2022-06-03:
`/etc/fstab/` works but not for UUIDs :(
```
root@OpenWrt:/home# cat /etc/fstab 
# <file system> <mount point> <type> <options> <dump> <pass>
UUID=bf436d1b-2d12-4dfe-8863-75dc5485a385       /home/marcin    ext4    defaults        0       0
#/dev/sdb1      /home/marcin    ext4    defaults        0       0
#/dev/mapper/crypt      /media/ ext4    defaults        0       0

root@OpenWrt:/home# mount -a
mount: mounting UUID=bf436d1b-2d12-4dfe-8863-75dc5485a385 on /home/marcin failed: No such file or directory
root@OpenWrt:/home# vim /etc/fstab 
root@OpenWrt:/home# cat /etc/fstab 
# <file system> <mount point> <type> <options> <dump> <pass>
#UUID=bf436d1b-2d12-4dfe-8863-75dc5485a385      /home/marcin    ext4    defaults        0       0
/dev/sdb1       /home/marcin    ext4    defaults        0       0
#/dev/mapper/crypt      /media/ ext4    defaults        0       0

root@OpenWrt:/home# mount -a
root@OpenWrt:/home# mount |grep /home
/dev/sdb1 on /home/marcin type ext4 (rw,relatime)
root@OpenWrt:/home# 

```

mwoz123, 2022-06-03:
Dumb me! there's a package named `libubus20220601` in repo. 
after 
`opkg install libubus20220601`

block info & mounting by UUID works fine:)

mwoz123, 2022-06-03:
Shouldn't it be dependency of the required by mount or another package ?

schuettecarsten, 2022-06-08:
With 22.03.0-rc1 and todays package sources, the error is a bit different:

```
root@Openwrt:~# block detect
Error loading shared library libubus.so.20220228: No such file or directory (needed by /sbin/block)
Error relocating /sbin/block: ubus_free: symbol not found
Error relocating /sbin/block: ubus_lookup_id: symbol not found
Error relocating /sbin/block: ubus_connect: symbol not found
Error relocating /sbin/block: ubus_invoke_fd: symbol not found
```

Unfortunately, libubus.so.20220228 is no longer available in the package repositories.

poranje, 2022-06-08:
work around until block-mount is rebuild against current version of libubus:
`ln -s /lib/libubus.so.20220601 /lib/libubus.so.20220228`
After that /sbin/block runs fine. Beware: that may not be true for other programs that also have been build against the old libubus.

schuettecarsten, 2022-06-09:
It still does not work with 22.03-rc3, but works wit today's snapshot.

schuettecarsten, 2022-06-09:
In todays 22.03-rc1 packages, there are several other packages that need older versions of libraries which are removed:

```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubox20211120 for umbim
 * pkg_hash_check_unresolved: cannot find dependency libubox20211120 for uqmi
 * pkg_hash_check_unresolved: cannot find dependency libblobmsg-json20211120 for uqmi
 * pkg_hash_check_unresolved: cannot find dependency libubox20211120 for mtd
```

brianmercer, 2023-09-23:
More block shenanigans now that libubus is updated to libubus.so.20230605 but block wants libubus.so.20220615