openwrt/openwrt #9986: ImageBuilder git detection is fragile; broken on Guix

attila-lendvai, 2022-06-03:
#### what

```
Checking 'which'... ok.
Checking 'ldconfig-stub'... ok.

Build dependency: Please install Git (git-core) >= 1.7.12.2

Prerequisite check failed. Use FORCE=1 to override.
make[1]: *** [Makefile:92: staging_dir/host/.prereq-build] Error 1
make: *** [Makefile:242: image] Error 2
```

i tried `FORCE=1 ./my-script-calling-image-builder.sh` and also `export FORCE=1`, but it didn't help.

#### why

on Guix some programs are "wrapped" by a script to set up the necessary runtime environment. `git-core/git-submodule` is also wrapped, which breaks its detection.

this is the content of that file on Guix:

```
#!/gnu/store/chfwin3a4qp1znnpsjbmydr2jbzk0d6y-bash-minimal-5.1.8/bin/bash
export PATH="/gnu/store/hy6abswwv4d89zp464fw52z65fkzr7h5-perl-5.34.0/bin${PATH:+:}$PATH"
exec -a "$0" "/gnu/store/zn0mq7f1iv49azmnjw271hrq9avyi4fy-git-2.35.1/libexec/git-core/.git-submodule-real" "$@"
```

i'm not well versed in the autoconf universe, but maybe this could work instead: `git submodule --help | grep -- --recursive`?

Ansuel, 2022-06-04:
The thing is... will the git command work correctly in the buildroot? AFAIK the git prereq check is not complex at all...
How do you call git command on Guix? 

For reference this is the check done
https://github.com/openwrt/openwrt/blob/9be35180f43a4916f53430d8c93437d33896e860/include/prereq-build.mk#L180

attila-lendvai, 2022-06-06:
@Ansuel the prereq check greps for the **contents** of the `git-submodule` executable, which on Guix is a wrapper script that sets some variables and calls the original executable.

i call it exactly the same on Guix as anywhere else, but its contents are different.

commit 7e223a89 (2022-06-27): prereq-build: add additional git detection

With some OS (Guix) the git submodule command is wrapped in
a script. Current logic parse the git submodule script directly.
If it's wrapped the prereq check wrongly fails while 'git submodule
--recursive' is actually available.
Add an additional check that try to directly use the 'git submodule'
command to check if the prereq is satisfied.

Fixes: #9986
Reported-by: Attila Lendvai <attila@lendvai.name>
Suggested-by: Attila Lendvai <attila@lendvai.name>

(cherry picked from commit ab1f3a87063b94f485bd582f32a066b80007bac9)
closed 2022-06-09 by jow-