openwrt/openwrt #9990: The compilation is complete, there is no squashfs-factory.bin, are you using squashfs-firmware.bin to flash in PB-BOOT?

ayancn, 2022-06-05:
Compilation is complete, there is no squashfs-factory.bin, only initramfs-kernel.bin , squashfs-firmware.bin , squashfs-sysupgrade.bin Please use which to flash in PB-BOOT, thank you

jospezial, 2022-06-05:
Which device? Maybe the squashfs-factory.bin would exceed maximum size. You should have a red warning on the end of compilation then. Some devices have split profile (factory/openwrt flash layout). Choose in config.

ayancn, 2022-06-05:
> Which device? Maybe the squashfs-factory.bin would exceed maximum size. You should have a red warning on the end of compilation then. Some devices have split profile (factory/openwrt flash layout). Choose in config.

Mt7621 / cr6608 compiles initramfs kernel bin , squashfs-firmware. bin , squashfs-sysupgrade. Bin these three do not have squashfs factory Bin is squashfs firmware Bin brush in? Will the BL disappear after the pb-boot is brushed

pepe2k, 2022-06-18:
@ayancn you can find instructions for OpenWrt migration in the support commit description:
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=3343ca7e6837b2ac5f237ea78bf73d50831dea20

Based on that, you should use `squashfs-firmware.bin` image, not `squashfs-factory.bin`.

Cheers,
Piotr

closed 2022-06-18 by pepe2k