openwrt/openwrt #9992: SSR+ of helloworld cannot be compiled successfully error

ayancn, 2022-06-05:
[ feeds.conf.default的 https://github.com/fw876/helloworld ,的luci-app-ssr-plus无法编译完成出错,要如何编译?](https://github.com/fw876/helloworld of feeds.conf.default, the luci-app-ssr-plus of feeds.conf.default could not be compiled and completed error, how to compile?)

NueXini, 2022-06-05:
I compiled it successfully.You'd better post your error log.

ayancn, 2022-06-05:
> 我编译成功了。您最好发布错误日志。

WARNING: Makefile 'package/feeds/other/luci-app-mwan3helper/Makefile' has a dependency on 'pdnsd-alt', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'dns2socks', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'microsocks', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'pdnsd-alt', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'ipt2socks', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'redsocks2', which does not exist

make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:111: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

NueXini, 2022-06-05:
you should take a look at README.md

ayancn, 2022-06-06:
> you should take a look at README.md

Did you understand? Can you teach me? Thank you

Bruce-prs, 2022-06-07:
> > I compiled it successfully.You'd better post your error log.
> 
> WARNING: Makefile 'package/feeds/other/luci-app-mwan3helper/Makefile' has a dependency on 'pdnsd-alt', which does not exist WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'dns2socks', which does not exist WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'microsocks', which does not exist WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'pdnsd-alt', which does not exist WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'ipt2socks', which does not exist WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'redsocks2', which does not exist
> 
> make[2]: *** [package/Makefile:70: package/install] Error 255 make[2]: Leaving directory '/workdir/openwrt' make[1]: *** [package/Makefile:111: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2 make[1]: Leaving directory '/workdir/openwrt' make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2 Error: Process completed with exit code 2.

Plz use google translate and you will understand.

Lancet-z, 2022-06-09:
> > you should take a look at README.md
> 
> Did you understand? Can you teach me? Thank you

简单点就是你编译的时候列出来的那些个依赖包你没加。自己想办法加进去就能编译成功了。

closed 2022-06-18 by pepe2k