openwrt/openwrt #9994: 22.03rc3 broke 2Ghz wifi on my tl-wdr4300-v1

zuzzurro, 2022-06-05:
Here's what I get in the logs forever:

```
Sun Jun  5 12:57:41 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:57:51 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:58:02 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:58:03 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:58:03 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:58:03 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:58:11 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:58:32 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:58:32 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:58:32 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:58:32 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:58:49 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:59:12 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:59:12 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:59:12 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:59:12 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:59:12 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:59:13 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:59:14 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:59:14 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:59:14 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:59:14 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:59:15 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:59:27 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:59:42 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 12:59:42 2022 daemon.notice hostapd: handle_probe_req: send failed
Sun Jun  5 13:00:02 2022 daemon.notice hostapd: handle_probe_req: send failed

It was working fine in rc1

5g works fine as far as I can see

```

zuzzurro, 2022-06-11:
It looks like it's working again on rc4.

pepe2k, 2022-06-18:
> It looks like it's working again on rc4.

Closing based on this comment.

Cheers,
Piotr

closed 2022-06-18 by pepe2k