openwrt/openwrt #9996: GCC 12 won't compile elfutils

LynxChaus, 2022-06-05:
Hi.
GCC 12 need additional [patch](https://gcc.gnu.org/git/?p=gcc.git;a=commitdiff_plain;h=2f714642e574c64e1c0e093cad3de6f8accb6ec7;hp=9c186493a7cc1c911910b9c37220334f9529912d) from upstream [pr#104069](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=104069) or it can not compile elfutils-0.186 (and 0.187 too).

LynxChaus, 2022-06-05:
Oops, my bad/ 12.1.0 already patched. 
Reducing warnings via -Wno-use-after-free for target elfutils helps.

closed 2022-06-05 by LynxChaus

castiel652, 2022-06-20:
Just ran into this with GCC12 and Binutils 2.38

> Oops, my bad/ 12.1.0 already patched. Reducing warnings via -Wno-use-after-free for target elfutils helps.

Is that the proper solution?

LynxChaus, 2022-06-24:
For now - yes. gcc-12 emits to many false-positive errors for now