openwrt/openwrt #10001: DSA switch can't have two bridges on different ports

M95D, 2022-06-06:
Linksys WRT1900ACv1 (mvebu mamba), OpenWrt r19635 built by me with custom config.
```
/etc/config/network :
config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan2'   <-- internal switch port (DSA)
        list ports 'lan3'   <-- internal switch port (DSA)

config device
        option type 'bridge'
        option name 'no_lan_bridge'   <-- this bridge has the problem
        option bridge_empty '1'
        list ports 'eth2'   <-- separate USB network adapter
        list ports 'lan1'   <-- internal switch port (DSA)
```
There is one wlan on first bridge and two wlans on second bridge.
There is a separate network on lan4 only; no bridge.

Result:
OpenWrt config fails to add lan1 to the second bridge. Lan1 port becomes disabled. The link partner reports "cable unplugged".
Bridge works between wlans and the usb adapter (eth2).
First bridge works properly.
```
root@GRAPHRT:~# brctl show
bridge name     bridge id               STP enabled     interfaces
no_lan_bridge           7fff.0000000000e2       no              wlan0
                                                        wlan1
                                                        eth2
br-lan          7fff.b4750e5ff0ea       no              wlan0-1
                                                        lan2
                                                        lan3
```

I tried to manually add lan1 to bridge. It doesn't work.
```
root@GRAPHRT:~# brctl addif no_lan_bridge lan1
brctl: bridge no_lan_bridge: Not supported
```
After this command, dmesg shows:
```
<6>[78684.033797] no_lan_bridge: port 4(lan1) entered blocking state
<6>[78684.039692] no_lan_bridge: port 4(lan1) entered disabled state
<3>[78684.047833] mv88e6085 f1072004.mdio-mii:00: p3: hw VLAN 1 already used by port 1 in br-lan
<3>[78684.057332] mv88e6085 f1072004.mdio-mii:00: p3: hw VLAN 1 already used by port 1 in br-lan
<3>[78684.105660] mv88e6085 f1072004.mdio-mii:00 lan1: failed to initialize vlan filtering on this port
```

This might be the same issue as #9882 . It doesn't look like a kernel issue, more like a configuration issue. It seems that OpenWrt tries to assign the same hw VLAN to both bridges.

I can attach config files on request.
Thanks.

sbocinec, 2022-07-05:
I think, this is also similar or at least related to the issue #7077

LucaRood, 2022-07-19:
Same issue on WRT1200AC with the latest release (21.02.3 r16554-1d4dea6d4f).

Only one bridge comes up, and the other lan ports are shown as disabled, while wlans do work on all bridges. I don't think I have much to add, as the OP seems to have all the relevant info.

I agree that it looks to be the same as #7077.

closed 2024-12-04 by M95D