openwrt/openwrt #10016: Unify 6 LR radio0 down every few days

fpauser, 2022-06-09:
After a few days, radio 0 goes down and I have to restart the device to get the 2,4GHz working again.
The 5GHz wlan (radio1) is not affected and works.

`logread -f` spits out some errors that seem related:

```
Thu Jun  9 05:33:25 2022 kern.err kernel: [1629943.379610] mt7622-wmac 18000000.wmac: Message 0000002a (seq 3) timeout
Thu Jun  9 05:33:25 2022 daemon.notice netifd: Wireless device 'radio0' is now down
Thu Jun  9 05:33:25 2022 daemon.notice netifd: radio0 (27963): sh: acs_survey: out of range
Thu Jun  9 05:33:25 2022 daemon.notice netifd: radio0 (27963): sh: acs_survey: out of range
Thu Jun  9 05:33:25 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Thu Jun  9 05:33:45 2022 kern.err kernel: [1629963.859308] mt7622-wmac 18000000.wmac: Message 0000002a (seq 10) timeout
Thu Jun  9 05:33:45 2022 daemon.err hostapd: Could not set interface wlan0 flags (UP): Operation timed out
Thu Jun  9 05:33:45 2022 daemon.err hostapd: nl80211: Could not set interface 'wlan0' UP
Thu Jun  9 05:33:45 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Thu Jun  9 05:33:45 2022 daemon.err hostapd: nl80211 driver initialization failed.
Thu Jun  9 05:33:45 2022 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Thu Jun  9 05:33:45 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Thu Jun  9 05:33:45 2022 daemon.notice netifd: radio0 (27963): Command failed: Invalid argument
Thu Jun  9 05:33:45 2022 daemon.notice netifd: radio0 (27963): Device setup failed: HOSTAPD_START_FAILED
Thu Jun  9 05:33:45 2022 daemon.notice netifd: Wireless device 'radio0' set retry=0
Thu Jun  9 05:33:45 2022 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Thu Jun  9 05:33:45 2022 daemon.notice netifd: Wireless device 'radio0' is now down
```

OS info:

`cat /etc/os-release`:
```
NAME="OpenWrt"
VERSION="21.02.3"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 21.02.3"
VERSION_ID="21.02.3"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r16554-1d4dea6d4f"
OPENWRT_BOARD="mediatek/mt7622"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 21.02.3 r16554-1d4dea6d4f"
```

Wireless config:

`cat /etc/config/wireless`:
```
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/18000000.wmac'
	option band '2g'
	option disabled '0'
	option htmode 'HT40'
	option channel 'auto'
	option cell_density '0'
	option country 'DE'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'xxxxxxxx'
	option encryption 'psk2'
	option key 'xxxxxxxx'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option band '5g'
	option disabled '0'
	option cell_density '0'
	option country 'DE'
	option htmode 'HE40'
	option channel 'auto'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'xxxxxxxx_5ghz'
	option encryption 'psk2'
	option key 'xxxxxxxx'
```

MOZGIII, 2022-06-11:
I have this too on the latest stable release, but I installed a snapshot a month ago and it works fine. Today's snapshot is broken in a different way though, so be wary of that.

fpauser, 2022-06-12:
@MOZGIII Thanks for pointing out. So the the 22.03.0-rc4 release is not recommended?

MOZGIII, 2022-06-12:
> the 22.03.0-rc4 release is not recommended?

I didn't try that one. Might be better.

pepe2k, 2022-06-18:
@fpauser please, try `22.03-rc4` and/or `snapshot`.

Cheers,
Piotr

hwkrvn, 2022-06-25:
I am experiencing this too on 21.02.3. before I try a RC build, @fpauser @MOZGIII, any success?

fpauser, 2022-06-25:
Just did the update to 22.03-rc4. Right now all works fine (uptime: 0h 22m 48s) ;)

hwkrvn, 2022-06-25:
> Just did the update to 22.03-rc4. Right now all works fine (uptime: 0h 22m 48s) ;)

Confirmed fixed – ship it! 😂

Thanks for testing it out. I'll install today and keep an eye as well.

fpauser, 2022-07-06:
Status update (uptime: 11d 5h 51m 47s): all good, not a single hickup