openwrt/openwrt #10021: package/install target broken on GNU Make < 4.0 (or: 21.02 prereq-build fails to require Make 4.0)

Habbie, 2022-06-09:
1. https://github.com/openwrt/openwrt/commit/3b14ddf8d204ee59533ec76ed6018db01f77d6e7 ( build: fix opkg install step for large package selection ) changes `package/Makefile` to call the `$(file ..)` function. This function was introduced in GNU Make 4.0. RHEL/CentOS 7 ships 3.82.
2. 22.03 and master require Make 4.1 in `include/prereq-build.mk`
3. 21.02 requires 3.82 in `include/prereq-build.mk`

Possible solutions:
1. fix 21.02 prereq to require Make 4.0 (or 4.1)
2. change the `$(file ..)` invocation in 21.02 to work on an older Make. (I have noticed no other issues building 21.02 with Make 3.82)
3. ???

neheb, 2022-06-10:
I would guess option 2 is best.

Ansuel, 2022-06-15:
@neheb wonder if it would be problematic to backport prereq fix... seems less radical than fixing `$(file)`

neheb, 2022-06-15:
Right but that breaks CentOS7. Isn't there some rule about not doing that?