openwrt/openwrt #10031: K2P openwrt-22.03 squashfs image can not boot

skbeh, 2022-06-11:
After flash the image from official buildbot, the router can not boot. It will not reboot automatically.
When I flash initramfs image in **breed**, it does boot, but when flashing any squashfs image via `sysupgrade -n /tmp/openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade.bin` (the file is uploaded), the ssh connection closes and the router reboots. Then on the next boot the system remains **exactly** the same (without squashfs).
The image of openwrt-21.02 boots and works.

DragonBluep, 2022-06-11:
Did you choose 'phicomm 0xa0000' in flash layout? Can you show me the init web page of breed?

I changed spi clock a year ago, maybe it is not compatible with some flash chips.

skbeh, 2022-06-11:
I always chose `公版 (reference) (0x50000)`.
```
CPU	MediaTek MT7621A ver 1, eco 3
内存	128MB DDR3
Flash	GigaDevice GD25Q128 @ 45MHz (16MB)
以太网	MediaTek MT7530 Gigabit switch
时钟频率	CPU: 1080MHz, DDR: 1200MHz, Bus: 270MHz, Ref: 40MHz
编译日期	2021-12-15 [git-f9b74d0]
版本	1.1 (r1337)
```
[Breed Web.html.gz](https://github.com/openwrt/openwrt/files/8883866/Breed.Web.html.gz)

DragonBluep, 2022-06-11:
https://github.com/openwrt/openwrt/blob/0dc056eb66e1b3a4a6797bdf91f7362df6ced9c3/target/linux/ramips/dts/mt7621_phicomm_k2p.dts#L85-L89

To be compatible with the original U-boot, the OpenWrt system partition is located at **0xA0000** (Phicomm Layout). On the other hand, the startup failure may be caused by CPU overclocking. Please try to restore the default frequency to 880MHz and use 0xA0000 layout. If all these do not work, it may be due to spi clock, I'll fix it later.

DragonBluep, 2022-06-11:
Maybe I know the reason, it's because there are two different firmware headers in flash. Please flash this file with `公版 (reference)` and then flash Openwrt with `Phicomm(0xA0000)`.
[k2p_fix.zip](https://github.com/openwrt/openwrt/files/8883890/k2p_fix.zip)

skbeh, 2022-06-11:
The bin file is not a illegal firmware so breed rejected to flash it.
I used to flash initramfs image because the squashfs one can not boot if I flash it by breed.

skbeh, 2022-06-11:
When I flash squashfs image (without flashing k2p_fix.bin) it enters boot loop.

skbeh, 2022-06-11:
Then I flashed squashfs image of 21.02 branch it boots correctly.

DragonBluep, 2022-06-11:
Strange, okay, maybe it's not difficult for you to use a telnet tool like `putty`. 
1. Enter breed by press reset button
2. Open a telnet session with putty (IP:192.168.1.1 port: 23)
3. Execute `flash erase 0x50000 0x50000` in Breed telnet console. 
4. Then Flash `squashfs-sysupgrade.bin`, let's see what happens next.

skbeh, 2022-06-11:
I did the following steps. Unfortunately it can not boot (not boot loop) and the blue light is blinking.

DragonBluep, 2022-06-11:
Please try this snapshots build first (can only be accessed by SSH, without LuCI)
https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade.bin

And in this version, I reduced the SPI clock.
[openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade.zip](https://github.com/openwrt/openwrt/files/8884035/openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade.zip)

Looking forward to your feedback.

skbeh, 2022-06-11:
The firmware I tested before is just from the snapshots build.
Your build version works!

DragonBluep, 2022-06-11:
Thanks for your test, I will push a PR to try to fix this issue later. But it could take weeks because the community needs to do some reviews.

skbeh, 2022-06-11:
Please notice, I also have a K2P hardware which uses flash `mx25l12805d` and its max frequency is 50mhz.

DragonBluep, 2022-06-11:
[openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade.zip](https://github.com/openwrt/openwrt/files/8884145/openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade.zip)

I build a 50MHz version, can you test it if possible?

> Please notice, I also have a K2P hardware which uses flash `mx25l12805d` and its max frequency is 50mhz.

Typically, SPI Flash produced in recent years have a maximum frequency of 133MHz. But the actual maximum frequency that can be achieved is often related to CPU, circuit design and so on. Even the same hardware can sometimes be different.

skbeh, 2022-06-11:
Because it is in usage now, I will test it 30 minutes later.

skbeh, 2022-06-11:
The GD25Q128 one works with the 50mhz build.

skbeh, 2022-06-11:
But I am wondering if there are some convenient ways to test if the spi flash is stable.

DragonBluep, 2022-06-11:
There is a package called `flashrom` that can read and write spi Flash and is able to specify the spi clock. But since the system is installed on flash, this way is not safe. Since I've lost K2P now, I can't do more.

Hey, can you push a PR yourself? This may be more convincing. Just change `8` to `5`
https://github.com/openwrt/openwrt/blob/e3e9eb31a281643737142e8e80c6f49204b5ba18/target/linux/ramips/dts/mt7621_phicomm_k2p.dts#L53

skbeh, 2022-06-11:
I ran a fio data integrity test on GD25Q128 one with 7M test file size and 50Mhz flash frequency and it passed.

DragonBluep, 2022-06-11:
It's funny this guy uses an oscilloscope to observe the noise on Xiaomi MI Router 4AG / 3G V2. 
https://github.com/openwrt/openwrt/pull/4048#issuecomment-813340205

So 50MHz can be a reasonable number for most devices, I believe.

skbeh, 2022-06-11:
I tested the K2P with mx25l12805d, both the 50mhz firmware and the 80mhz one works.

skbeh, 2022-06-11:
Even 60MHz can not boot with GD25Q128, so I think 50MHz is a reasonable frequency.

skbeh, 2022-06-14:
https://github.com/openwrt/openwrt/commit/857ea3f690aba8513b356926d9c430adafc7c50b

closed 2022-06-14 by skbeh