openwrt/openwrt #10034: firewall 4 disappears when selecting IPv6

castillofrancodamian, 2022-06-11:
In a clean build from source, I haven't been able to select firewall4 as it disappears when disabling IPv6.

pepe2k, 2022-06-18:
@castillofrancodamian this is not a bug, `firewall4` depends on IPv6 support (you can follow dependencies).

Cheers,
Piotr

closed 2022-06-18 by pepe2k

castillofrancodamian, 2022-06-18:
And there is no alternative?

champtar, 2022-06-18:
Just put a rule to drop IPv6 traffic ?

castillofrancodamian, 2022-06-18:
The idea was to remove possibly everything that has to do with IPv6 and related configuration as well. I don't expect to use IPv6 any time soon. It also saves space.

champtar, 2022-06-18:
Maintaining the option to disable IPv6 is a huge amount of work, not that much people uses it, and even just disabling IPv6 at runtime breaks some software, so expect this option to just disappear in a future release.