openwrt/openwrt #10036: Hostapd feature detection broken in master

Thalience, 2022-06-11:
Since commit c35ff1affe8f347b60a7539648a90b45ad43ffef, `ubus call luci getFeatures` always shows every hostapd feature as false. This prevents LuCI from configuring features like WPA3/SAE or 802.11r even when a full variant of hostapd/wpad is installed.

I believe the change to getopt in [package/network/services/hostapd/patches/420-indicate-features.patch](https://github.com/openwrt/openwrt/commit/c35ff1affe8f347b60a7539648a90b45ad43ffef#diff-d28874bb0b856374ecc9aeca80e796ab9e6b8bc28f08afa864c4f9c70dd3cd15) is wrong and causes `/usr/sbin/hostapd -v<feature>` to only print the version information instead of checking the feature.

Expected result:
```
root@router:~# hostapd -vsae && echo sae enabled
sae enabled
```

Actual result:
```
root@router:~# hostapd -vsae && echo sae enabled
hostapd v2.11-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2022, Jouni Malinen <j@w1.fi> and contributors
```

Thalience, 2022-06-11:
@blocktrron Applying the following patch fixes it for me

diff --git a/package/network/services/hostapd/patches/420-indicate-features.patch b/package/network/services/hostapd/patches/420-indicate-features.patch
index 80d945ca16..12edb6bacc 100644
--- a/package/network/services/hostapd/patches/420-indicate-features.patch
+++ b/package/network/services/hostapd/patches/420-indicate-features.patch
@@ -14,7 +14,7 @@
        wpa_supplicant_event_global = hostapd_wpa_event_global;
        for (;;) {
 -              c = getopt(argc, argv, "b:Bde:f:hi:KP:sSTtu:vg:G:q");
-+              c = getopt(argc, argv, "b:Bde:f:hi:KP:sSTtu:vg:G:qv::");
++              c = getopt(argc, argv, "b:Bde:f:hi:KP:sSTtu:g:G:qv::");
                if (c < 0)
                        break;
                switch (c) {

ptpt52, 2022-06-12:
why `hostapd -vsae` exit with code 1?

GeertVisser, 2022-06-12:
I have the same issue on my c2600

Thalience, 2022-06-12:
> why `hostapd -vsae` exit with code 1?

The getopt optstring has `'v'` twice. Only the 2nd one specifies an (optional) argument, so the version handler is first called without optarg and does `show_version(); exit(1);`

jow-, 2022-06-13:
Fixed by f03b20837bf821241acecaf94fc6a26ce49991be

closed 2022-06-13 by jow-