openwrt/openwrt #10041: Tp-link EAP225 V3 keeps rebooting after increase TX power

hussein-87, 2022-06-12:
a device tp-link EAP225 v3 (EU) with OpenWrt 21.02.3 as AP

when I set the TXpwoer above 19 dBm for the 5GHz band, the device keeps rebooting directly after connecting my phones and laptops.

i checked quicly the last line in the system logs befor rebooting:
Fri Jun 10 18:37:21 2022 kern.warn kernel: [ 2036.154251] ath10k_pci 0000:00:00.0: Invalid VHT mcs 15 peer stats

notice: the txpower of 2.4 has no problem when i change it.

my setting for 5G
Channel: 44 Width : 40
Encryption: WPA2 PSK (CCMP)

my setting for 2G
Channel: 13 Width : 20
Encryption: WPA2 PSK (CCMP)
TXpower: 20

neheb, 2022-06-24:
Lots of possibilities. I would try swapping out the power supply.

hussein-87, 2022-06-24:
I solved this problem (not good solution) by decreasing tx power to 18 for both bands .