openwrt/openwrt #10045: extroot: cannot find device with UUID

cmonty14, 2022-06-12:
Hi,
I configured Extroot based on [this](https://openwrt.org/docs/guide-user/additional-software/extroot_configuration) Wiki article.

However after reboot the USB device is not mounted.

[This](https://openwrt.org/toh/mikrotik/rb760igs) is my device, and I have installed OpenWrt 21.02.3.

Here's some output for troubleshooting:
```
root@eddie:~# block info
/dev/mtdblock8: UUID="bd699133-53e1f799-8bcf4b95-ea59fe6e" VERSION="4.0" MOUNT="/rom" TYPE="squashfs"
/dev/mtdblock9: MOUNT="/overlay" TYPE="jffs2"
/dev/sda1: UUID="2f7a4335-5bd2-4e09-9463-c197a92e0c68" VERSION="1.0" TYPE="ext4"
```
```
root@eddie:~# uci show fstab
fstab.@global[0]=global
fstab.@global[0].anon_swap='0'
fstab.@global[0].anon_mount='0'
fstab.@global[0].auto_swap='1'
fstab.@global[0].auto_mount='1'
fstab.@global[0].check_fs='0'
fstab.@global[0].delay_root='15'
fstab.rwm=mount
fstab.rwm.device='/dev/mtdblock9'
fstab.rwm.target='/rwm'
fstab.overlay=mount
fstab.overlay.uuid='2f7a4335-5bd2-4e09-9463-c197a92e0c68'
fstab.overlay.target='/overlay'
```
```
root@eddie:~# logread | sed -n -e "/- preinit -/,/- init -/p"
Sun Jun 12 23:47:46 2022 user.info kernel: [    2.051370] init: - preinit -
Sun Jun 12 23:47:46 2022 kern.info kernel: [    2.639092] usb 1-1: new high-speed USB device number 2 using xhci-mtk
Sun Jun 12 23:47:46 2022 kern.notice kernel: [    2.807117] random: procd: uninitialized urandom read (4 bytes read)
Sun Jun 12 23:47:46 2022 kern.info kernel: [    2.825330] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
Sun Jun 12 23:47:46 2022 kern.info kernel: [    2.833626] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Sun Jun 12 23:47:46 2022 kern.info kernel: [    2.842091] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Sun Jun 12 23:47:46 2022 kern.notice kernel: [    3.042914] random: jshn: uninitialized urandom read (4 bytes read)
Sun Jun 12 23:47:46 2022 kern.notice kernel: [    3.155557] random: jshn: uninitialized urandom read (4 bytes read)
Sun Jun 12 23:47:46 2022 kern.info kernel: [    3.495607] mt7530 mdio-bus:1f lan2: configuring for phy/gmii link mode
Sun Jun 12 23:47:46 2022 kern.info kernel: [    3.502634] 8021q: adding VLAN 0 to HW filter on device lan2
Sun Jun 12 23:47:46 2022 kern.info kernel: [    6.587547] mt7530 mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control rx/tx
Sun Jun 12 23:47:46 2022 kern.info kernel: [    6.595070] IPv6: ADDRCONF(NETDEV_CHANGE): lan2: link becomes ready
Sun Jun 12 23:47:46 2022 kern.notice kernel: [    7.869974] jffs2: notice: (640) jffs2_build_xattr_subsystem: complete building xattr subsystem, 85 of xdatum (81 unchecked, 4 orphan) and 149 of xref (10 dead, 2 orphan) found.
Sun Jun 12 23:47:46 2022 user.info kernel: [    7.897432] mount_root: loading kmods from internal overlay
Sun Jun 12 23:47:46 2022 user.err kernel: [    8.411193] kmodloader: failed to open /tmp/overlay/upper/lib/modules/5.4.188/xhci-hcd.ko
Sun Jun 12 23:47:46 2022 user.err kernel: [    8.419771] kmodloader: failed to open /tmp/overlay/upper/lib/modules/5.4.188/xhci-mtk.ko
Sun Jun 12 23:47:46 2022 user.err kernel: [    8.428309] kmodloader: failed to open /tmp/overlay/upper/lib/modules/5.4.188/xhci-pci.ko
Sun Jun 12 23:47:46 2022 user.err kernel: [    8.436837] kmodloader: failed to open /tmp/overlay/upper/lib/modules/5.4.188/xhci-plat-hcd.ko
Sun Jun 12 23:47:46 2022 user.err kernel: [    8.493250] mount_root: failed to launch kmodloader from internal overlay
Sun Jun 12 23:47:46 2022 user.info kernel: [    8.668707] block: attempting to load /tmp/jffs_cfg/upper/etc/config/fstab
Sun Jun 12 23:47:46 2022 user.info kernel: [    8.686282] block: extroot: device not present, retrying in 15 seconds
Sun Jun 12 23:47:46 2022 kern.notice kernel: [   14.567320] crng init done
Sun Jun 12 23:47:46 2022 kern.notice kernel: [   14.570031] random: 7 urandom warning(s) missed due to ratelimiting
Sun Jun 12 23:47:46 2022 user.err kernel: [   23.831112] block: extroot: cannot find device with UUID 2f7a4335-5bd2-4e09-9463-c197a92e0c68
Sun Jun 12 23:47:46 2022 user.info kernel: [   23.840958] mount_root: switching to jffs2 overlay
Sun Jun 12 23:47:46 2022 kern.warn kernel: [   23.848508] overlayfs: upper fs does not support tmpfile.
Sun Jun 12 23:47:46 2022 user.warn kernel: [   23.857934] urandom-seed: Seeding with /etc/urandom.seed
Sun Jun 12 23:47:46 2022 kern.info kernel: [   23.951129] mt7530 mdio-bus:1f lan2: Link is Down
Sun Jun 12 23:47:46 2022 user.info kernel: [   23.977645] procd: - early -
Sun Jun 12 23:47:46 2022 user.info kernel: [   23.980659] procd: - watchdog -
Sun Jun 12 23:47:46 2022 user.info kernel: [   24.591219] procd: - watchdog -
Sun Jun 12 23:47:46 2022 user.info kernel: [   24.595622] procd: - ubus -
Sun Jun 12 23:47:46 2022 user.info kernel: [   24.656748] procd: - init -
```

This means block device _/dev/sda1_ with _UUID=2f7a4335-5bd2-4e09-9463-c197a92e0c68_ is not mounted although fstab is configured correctly.

This looks like a bug.

I can mount block device /dev/sda1 manually w/o problems after router has booted.
However there's no directory _.extroot-uuid_.
```
root@eddie:~# mount /dev/sda1 /mnt/
root@eddie:~# ls -la /mnt/
drwxr-xr-x    5 root     root          4096 Jun 12 23:40 .
drwxr-xr-x    1 root     root             0 May 14 12:22 ..
lrwxrwxrwx    1 root     root             1 Jun 12 23:40 .fs_state -> 2
drwx------    2 root     root         16384 Apr 17 19:19 lost+found
drwxr-xr-x   11 root     root          4096 Jun 12 23:40 upper
drwxr-xr-x    3 root     root          4096 Jun 12 23:40 work
```

neheb, 2022-06-24:
Sounds related to https://github.com/openwrt/openwrt/commit/f83e927b8744c0642fd61245935c4bb20e0b6f33 . This has not been backported AFAIK.

cmonty14, 2022-06-24:
> Sounds related to [f83e927](https://github.com/openwrt/openwrt/commit/f83e927b8744c0642fd61245935c4bb20e0b6f33) . This has not been backported AFAIK.

This means there's currently no solution available?

neheb, 2022-06-24:
I mean you could just edit the file.

cmonty14, 2022-06-24:
Where is fstab.init stored?
Or is my understanding incorrect about the required modification?

neheb, 2022-06-24:
/etc/init.d/fstab.init

cmonty14, 2022-06-24:
Would this make the workaround with file _/etc/rc.local_
```
export PREINIT=1
mount_root
```
obsolete?

cmonty14, 2022-06-25:
I have implemented [f83e927](https://github.com/openwrt/openwrt/commit/f83e927b8744c0642fd61245935c4bb20e0b6f33) and cleaned _/etc/rc.local_.
After rebooting my device all required drives (Micro SD) are available:
```
root@eddie:~$ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mtdblock0    31:0    0  256K  1 disk 
mtdblock1    31:1    0   60K  1 disk 
mtdblock2    31:2    0    4K  1 disk 
mtdblock3    31:3    0   60K  1 disk 
mtdblock4    31:4    0    4K  1 disk 
mtdblock5    31:5    0    4K  1 disk 
mtdblock6    31:6    0 15.8M  0 disk 
mtdblock7    31:7    0  2.4M  0 disk 
mtdblock8    31:8    0 13.4M  0 disk /rom/rom
mtdblock9    31:9    0 10.9M  0 disk /rom/overlay
mmcblk0     179:0    0  1.9G  0 disk 
├─mmcblk0p1 179:1    0  1.8G  0 part /overlay
└─mmcblk0p2 179:2    0  100M  0 part [SWAP]
```
However I cannot verify in syslog if device with UUID is correctly mounted.
Means there are not relevant entries since reboot, and existing entries are outdated.
```
root@eddie:~$ logread | sed -n -e "/- preinit -/,/- init -/p"
Sun Jun 19 12:51:02 2022 user.info kernel: [    2.060445] init: - preinit -
Sun Jun 19 12:51:02 2022 kern.info kernel: [    2.824690] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
Sun Jun 19 12:51:02 2022 kern.info kernel: [    2.833018] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Sun Jun 19 12:51:02 2022 kern.info kernel: [    2.841500] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Sun Jun 19 12:51:02 2022 kern.notice kernel: [    3.041827] random: jshn: uninitialized urandom read (4 bytes read)
Sun Jun 19 12:51:02 2022 kern.notice kernel: [    3.154692] random: jshn: uninitialized urandom read (4 bytes read)
Sun Jun 19 12:51:02 2022 kern.notice kernel: [    3.226540] random: jshn: uninitialized urandom read (4 bytes read)
Sun Jun 19 12:51:02 2022 kern.info kernel: [    3.502271] mt7530 mdio-bus:1f lan2: configuring for phy/gmii link mode
Sun Jun 19 12:51:02 2022 kern.info kernel: [    3.509317] 8021q: adding VLAN 0 to HW filter on device lan2
Sun Jun 19 12:51:02 2022 kern.info kernel: [    6.588643] mt7530 mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control rx/tx
Sun Jun 19 12:51:02 2022 kern.info kernel: [    6.596169] IPv6: ADDRCONF(NETDEV_CHANGE): lan2: link becomes ready
Sun Jun 19 12:51:02 2022 kern.notice kernel: [    7.885983] jffs2: notice: (638) jffs2_build_xattr_subsystem: complete building xattr subsystem, 108 of xdatum (82 unchecked, 26 orphan) and 172 of xref (31 dead, 2 orphan) found.
Sun Jun 19 12:51:02 2022 user.info kernel: [    7.916172] mount_root: loading kmods from internal overlay
Sun Jun 19 12:51:02 2022 user.err kernel: [    8.448550] kmodloader: failed to open /tmp/overlay/upper/lib/modules/5.4.188/xhci-hcd.ko
Sun Jun 19 12:51:02 2022 user.err kernel: [    8.457120] kmodloader: failed to open /tmp/overlay/upper/lib/modules/5.4.188/xhci-mtk.ko
Sun Jun 19 12:51:02 2022 user.err kernel: [    8.465647] kmodloader: failed to open /tmp/overlay/upper/lib/modules/5.4.188/xhci-pci.ko
Sun Jun 19 12:51:02 2022 user.err kernel: [    8.474158] kmodloader: failed to open /tmp/overlay/upper/lib/modules/5.4.188/xhci-plat-hcd.ko
Sun Jun 19 12:51:02 2022 user.err kernel: [    8.526400] mount_root: failed to launch kmodloader from internal overlay
Sun Jun 19 12:51:02 2022 user.info kernel: [    8.694558] block: attempting to load /tmp/jffs_cfg/upper/etc/config/fstab
Sun Jun 19 12:51:02 2022 user.info kernel: [    8.710635] block: extroot: device not present, retrying in 10 seconds
Sun Jun 19 12:51:02 2022 kern.notice kernel: [   11.092297] crng init done
Sun Jun 19 12:51:02 2022 kern.notice kernel: [   11.095011] random: 7 urandom warning(s) missed due to ratelimiting
Sun Jun 19 12:51:02 2022 user.err kernel: [   18.909504] block: extroot: cannot find device with UUID 2f7a4335-5bd2-4e09-9463-c197a92e0c68
Sun Jun 19 12:51:02 2022 user.info kernel: [   18.919468] mount_root: switching to jffs2 overlay
Sun Jun 19 12:51:02 2022 kern.warn kernel: [   18.928694] overlayfs: upper fs does not support tmpfile.
Sun Jun 19 12:51:02 2022 user.warn kernel: [   18.938484] urandom-seed: Seeding with /etc/urandom.seed
Sun Jun 19 12:51:02 2022 kern.info kernel: [   19.038099] mt7530 mdio-bus:1f lan2: Link is Down
Sun Jun 19 12:51:02 2022 user.info kernel: [   19.068864] procd: - early -
Sun Jun 19 12:51:02 2022 user.info kernel: [   19.071849] procd: - watchdog -
Sun Jun 19 12:51:02 2022 user.info kernel: [   19.672315] procd: - watchdog -
Sun Jun 19 12:51:02 2022 user.info kernel: [   19.677109] procd: - ubus -
Sun Jun 19 12:51:02 2022 user.info kernel: [   19.741041] procd: - init -
```
If `logread | sed -n -e "/- preinit -/,/- init -/p"` prints no output, does this mean there are not errors in boot phase?

cmonty14, 2022-07-02:
After performing some verification checks I can confirm that extroot is working only if
```
root@eddie:~# cat /etc/rc.local 
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.
export PREINIT=1
mount_root
exit 0
```

Any other workaround or tweak is not working.

The error message in system log
`block: extroot: cannot find device with UUID <uuid ext4 partition>`
remains.

SimplyKot, 2023-05-23:
The same problem on Netgear WNR2200.

Tweak /etc/rc.local 
`
export PREINIT=1
mount_root
`
do not  work well, because router reboots then  just open LUCI interface

[image]

testgit77, 2023-07-19:
I was experiencing the same issues.  What got it working for me was addressing the `kmodloader` error messages from `logread | sed -n -e "/- preinit -/,/- init -/p"`.  (See below)  I had uninstalled `ppp/ppp-mod-pppoe` and I theorize (b/c I'm not sure) those errors were causing the `mount_root` to fail?  Anyways, re-installing the packages appeared to fix the issue.

See `mkodloader` errors messages below.
```
root@openwrt:~# logread | sed -n -e "/- preinit -/,/- init -/p"
..........................
..........................
..........................
Wed Jul 19 19:50:46 2023 kern.notice kernel: [   11.654246] UBIFS (ubi0:1): Mounting in unauthenticated mode
Wed Jul 19 19:50:46 2023 kern.notice kernel: [   11.665850] UBIFS (ubi0:1): background thread "ubifs_bgt0_1" started, PID 573
Wed Jul 19 19:50:46 2023 kern.notice kernel: [   11.748784] UBIFS (ubi0:1): recovery needed
Wed Jul 19 19:50:46 2023 kern.notice kernel: [   11.926693] UBIFS (ubi0:1): recovery completed
Wed Jul 19 19:50:46 2023 kern.notice kernel: [   11.935733] UBIFS (ubi0:1): UBIFS: mounted UBI device 0, volume 1, name "rootfs_data"
Wed Jul 19 19:50:46 2023 kern.notice kernel: [   11.951342] UBIFS (ubi0:1): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
Wed Jul 19 19:50:46 2023 kern.notice kernel: [   11.971087] UBIFS (ubi0:1): FS size: 20189184 bytes (19 MiB, 159 LEBs), journal size 1015809 bytes (0 MiB, 8 LEBs)
Wed Jul 19 19:50:46 2023 kern.notice kernel: [   11.991701] UBIFS (ubi0:1): reserved for root: 953584 bytes (931 KiB)
Wed Jul 19 19:50:46 2023 kern.notice kernel: [   12.004645] UBIFS (ubi0:1): media format: w5/r0 (latest is w5/r0), UUID 4550CEB4-780A-4C7B-B48D-E0C48477A706, small LPT model
Wed Jul 19 19:50:46 2023 user.info kernel: [   12.042967] mount_root: loading kmods from internal overlay
---> Wed Jul 19 19:50:46 2023 user.err kernel: [   12.120431] kmodloader: failed to open /tmp/overlay/upper/lib/modules/5.10.146/ppp_async.ko
---> Wed Jul 19 19:50:46 2023 user.err kernel: [   12.137370] kmodloader: failed to open /tmp/overlay/upper/lib/modules/5.10.146/ppp_generic.ko
---> Wed Jul 19 19:50:46 2023 user.err kernel: [   12.154579] kmodloader: failed to open /tmp/overlay/upper/lib/modules/5.10.146/pppoe.ko
---> Wed Jul 19 19:50:46 2023 user.err kernel: [   12.170845] kmodloader: failed to open /tmp/overlay/upper/lib/modules/5.10.146/pppox.ko
---> Wed Jul 19 19:50:46 2023 user.err kernel: [   12.235502] mount_root: failed to launch kmodloader from internal overlay
Wed Jul 19 19:50:46 2023 user.info kernel: [   12.407837] block: attempting to load /tmp/ubifs_cfg/upper/etc/config/fstab
Wed Jul 19 19:50:46 2023 user.info kernel: [   12.431255] block: extroot: device not present, retrying in 5 seconds
Wed Jul 19 19:50:46 2023 user.err kernel: [   17.473979] block: extroot: cannot find device with UUID bd030103-4501-40ef-b409-444eaceda561
Wed Jul 19 19:50:46 2023 user.info kernel: [   17.495361] mount_root: switching to ubifs overlay
..........................
..........................
..........................
```
Once I re-installed the packages, things worked for me again without having to do the `/etc/rc.local` fix:

Below is after re-installing the packages and successfully mounting the USB drive.
```
root@openwrt:~# logread | sed -n -e "/- preinit -/,/- init -/p"

..........................
..........................
..........................
Wed Jul 19 19:52:22 2023 kern.notice kernel: [   11.673556] UBIFS (ubi0:1): Mounting in unauthenticated mode
Wed Jul 19 19:52:22 2023 kern.notice kernel: [   11.685172] UBIFS (ubi0:1): background thread "ubifs_bgt0_1" started, PID 573
Wed Jul 19 19:52:22 2023 kern.notice kernel: [   11.768085] UBIFS (ubi0:1): recovery needed
Wed Jul 19 19:52:22 2023 kern.notice kernel: [   11.954798] UBIFS (ubi0:1): recovery completed
Wed Jul 19 19:52:22 2023 kern.notice kernel: [   11.963830] UBIFS (ubi0:1): UBIFS: mounted UBI device 0, volume 1, name "rootfs_data"
Wed Jul 19 19:52:22 2023 kern.notice kernel: [   11.979431] UBIFS (ubi0:1): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
Wed Jul 19 19:52:22 2023 kern.notice kernel: [   11.999176] UBIFS (ubi0:1): FS size: 20189184 bytes (19 MiB, 159 LEBs), journal size 1015809 bytes (0 MiB, 8 LEBs)
Wed Jul 19 19:52:22 2023 kern.notice kernel: [   12.019798] UBIFS (ubi0:1): reserved for root: 953584 bytes (931 KiB)
Wed Jul 19 19:52:22 2023 kern.notice kernel: [   12.032634] UBIFS (ubi0:1): media format: w5/r0 (latest is w5/r0), UUID 4550CEB4-780A-4C7B-B48D-E0C48477A706, small LPT model
Wed Jul 19 19:52:22 2023 user.info kernel: [   12.071503] mount_root: loading kmods from internal overlay
Wed Jul 19 19:52:22 2023 user.info kernel: [   12.217277] kmodloader: loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
Wed Jul 19 19:52:22 2023 kern.notice kernel: [   12.488817] SCSI subsystem initialized
Wed Jul 19 19:52:22 2023 kern.info kernel: [   12.519253] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
Wed Jul 19 19:52:22 2023 kern.info kernel: [   12.533074] ohci-platform: OHCI generic platform driver
Wed Jul 19 19:52:22 2023 kern.info kernel: [   12.552054] uhci_hcd: USB Universal Host Controller Interface driver
Wed Jul 19 19:52:22 2023 kern.info kernel: [   12.579192] usb-storage 1-2:1.0: USB Mass Storage device detected
Wed Jul 19 19:52:22 2023 kern.info kernel: [   12.592460] scsi host0: usb-storage 1-2:1.0
[... cut ...]
Wed Jul 19 19:52:22 2023 kern.info kernel: [   18.616576] EXT4-fs (sda2): mounted filesystem with ordered data mode. Opts:
Wed Jul 19 19:52:22 2023 kern.notice kernel: [   18.653080] UBIFS (ubi0:1): un-mount UBI device 0
Wed Jul 19 19:52:22 2023 kern.notice kernel: [   18.662556] UBIFS (ubi0:1): background thread "ubifs_bgt0_1" stops
Wed Jul 19 19:52:22 2023 user.info kernel: [   18.678026] mount_root: switched to extroot
..........................
..........................
..........................
```

lrussell887, 2024-01-07:
I can confirm the same problem as @testgit77. I had removed the b43 module on my router prior to setting up extroot which prevented it from mounting. Undoing that change allowed it to work normally, even after removing it again in the new extroot.

```
root@backup:~# block info; uci show fstab; logread | sed -n -e "/- preinit -/,/- init -/p"
/dev/mtdblock4: UUID="241fe5d9-0f82e6f6-689c2746-6d52fd53" VERSION="4.0" MOUNT="/rom" TYPE="squashfs"
/dev/mtdblock5: MOUNT="/overlay" TYPE="jffs2"
/dev/sda1: UUID="4081a3d7-166c-45c4-8cc1-8ae414a721ae" LABEL="extroot" VERSION="1.0" TYPE="ext4"
/dev/sda2: UUID="e938caff-b9e9-4ada-865d-087dfaefc4b6" TYPE="crypto_LUKS"
fstab.@global[0]=global
fstab.@global[0].anon_swap='0'
fstab.@global[0].anon_mount='0'
fstab.@global[0].auto_swap='1'
fstab.@global[0].auto_mount='1'
fstab.@global[0].delay_root='5'
fstab.@global[0].check_fs='0'
fstab.@mount[0]=mount
fstab.@mount[0].target='/mnt/sda1'
fstab.@mount[0].uuid='74c3963a-3026-400f-8be0-4aae6877a0f7'
fstab.@mount[0].enabled='0'
fstab.@mount[1]=mount
fstab.@mount[1].target='/mnt/sda2'
fstab.@mount[1].uuid='e938caff-b9e9-4ada-865d-087dfaefc4b6'
fstab.@mount[1].enabled='0'
fstab.extroot=mount
fstab.extroot.uuid='4081a3d7-166c-45c4-8cc1-8ae414a721ae'
fstab.extroot.target='/overlay'
fstab.rwm=mount
fstab.rwm.device='/dev/mtdblock5'
fstab.rwm.target='/rwm'
Sat Jan  6 06:02:03 2024 user.info kernel: [    4.368848] init: - preinit -
Sat Jan  6 06:02:03 2024 kern.info kernel: [    4.499453] usb 1-1: new high-speed USB device number 2 using ehci-platform
Sat Jan  6 06:02:03 2024 kern.notice kernel: [    5.257310] random: procd: uninitialized urandom read (4 bytes read)
Sat Jan  6 06:02:03 2024 kern.notice kernel: [    6.157458] random: jshn: uninitialized urandom read (4 bytes read)
Sat Jan  6 06:02:03 2024 kern.notice kernel: [    6.318028] random: jshn: uninitialized urandom read (4 bytes read)
Sat Jan  6 06:02:03 2024 kern.notice kernel: [    6.458573] random: jshn: uninitialized urandom read (4 bytes read)
Sat Jan  6 06:02:03 2024 kern.notice kernel: [    9.348396] jffs2: notice: (286) jffs2_build_xattr_subsystem: complete building xattr subsystem, 31 of xdatum (17 unchecked, 0 orphan) and 47 of xref (7 dead, 0 orphan) found.
Sat Jan  6 06:02:03 2024 user.info kernel: [    9.373462] mount_root: loading kmods from internal overlay
Sat Jan  6 06:02:03 2024 user.err kernel: [    9.396941] kmodloader: failed to open /tmp/overlay/upper/lib/modules/5.15.137/b43.ko
Sat Jan  6 06:02:03 2024 user.err kernel: [    9.734048] mount_root: failed to launch kmodloader from internal overlay
Sat Jan  6 06:02:03 2024 user.info kernel: [    9.941355] block: attempting to load /tmp/jffs_cfg/upper/etc/config/fstab
Sat Jan  6 06:02:03 2024 user.info kernel: [    9.952278] block: extroot: device not present, retrying in 5 seconds
Sat Jan  6 06:02:03 2024 user.err kernel: [   15.083247] block: extroot: cannot find device with UUID 4081a3d7-166c-45c4-8cc1-8ae414a721ae
Sat Jan  6 06:02:03 2024 user.info kernel: [   15.094285] mount_root: switching to jffs2 overlay
Sat Jan  6 06:02:03 2024 kern.warn kernel: [   15.107069] overlayfs: upper fs does not support tmpfile.
Sat Jan  6 06:02:03 2024 user.warn kernel: [   15.134097] urandom-seed: Seeding with /etc/urandom.seed
Sat Jan  6 06:02:03 2024 user.info kernel: [   15.250428] procd: - early -
Sat Jan  6 06:02:03 2024 user.info kernel: [   15.253936] procd: - watchdog -
Sat Jan  6 06:02:03 2024 user.info kernel: [   16.215272] procd: - watchdog -
Sat Jan  6 06:02:03 2024 user.info kernel: [   16.228086] procd: - ubus -
Sat Jan  6 06:02:03 2024 kern.notice kernel: [   16.489991] random: ubusd: uninitialized urandom read (4 bytes read)
Sat Jan  6 06:02:03 2024 kern.notice kernel: [   16.514060] random: ubusd: uninitialized urandom read (4 bytes read)
Sat Jan  6 06:02:03 2024 kern.notice kernel: [   16.545382] random: ubusd: uninitialized urandom read (4 bytes read)
Sat Jan  6 06:02:03 2024 user.info kernel: [   16.573713] procd: - init -
```

flipreverse, 2024-11-30:
@cmonty14 @testgit77 @lrussell887 I'm seeing the same issue.
I've made an interesting observation in my log: Right before an attempt in mounting the extroot is made, the usb device disappears:
`Fri Nov 29 15:16:26 2024 kern.info kernel: [    8.927610] usb 3-1: USB disconnect, device number 2`
The mount attempt obviously fails.  A few seconds later the device `/dev/sda1` appears again. Please have a look at your logs whether you also see a disconnect event.

Sammy1Am, 2025-05-02:
Just want to pile on and add that I had the same issue and resolution as @testgit77 and @lrussell887.

Just to clarify my own steps, I:
- Followed the directions from the [wiki here](https://openwrt.org/docs/guide-user/additional-software/extroot_configuration), but on reboot did not have a larger overlay.
- Spotted this kmodloader error in the log:
```
Thu May  1 22:16:54 2025 user.info kernel: [   10.250893] mount_root: loading kmods from internal overlay
Thu May  1 22:16:54 2025 user.err kernel: [   10.506004] kmodloader: failed to open /tmp/overlay/upper/lib/modules/6.6.73/veth.ko
Thu May  1 22:16:54 2025 user.err kernel: [   10.528821] mount_root: failed to launch kmodloader from internal overlay
Thu May  1 22:16:54 2025 user.info kernel: [   10.775074] block: attempting to load /tmp/overlay/upper/etc/config/fstab
Thu May  1 22:16:54 2025 user.info kernel: [   10.792159] block: extroot: device not present, retrying in 15 seconds
Thu May  1 22:16:54 2025 user.err kernel: [   25.931794] block: extroot: cannot find device with UUID 63bb5eb4-66d1-496f-8bb1-b349f86a7720
```
Ran `opkg install kmod-veth`, and rebooted. Notably I *did not* re-copy the current overlay onto the SD Card after reinstalling veth.
- Success! (And kmod-veth doesn't show as installed because it's not on the SD Card overlay)