openwrt/openwrt #10046: WRT3200ACM - Cannot upgrade firmware

Zurd, 2022-06-13:
What I have:
Linksys WRT3200ACM
Powered by LuCI lede-17.01 branch (git-18.180.55366-b78664c)
LEDE Reboot 17.01.5 r3919-38e704be71
Kernel Version 4.4.140

What I would like to have: 
21.02.3
https://openwrt.org/toh/views/toh_fwdownload?dataflt%5BModel*%7E%5D=WRT3200ACM

System / Backup / Flash and error is:
The uploaded image file does not contain a supported format. Make sure that you choose the generic image format for your platform. 

md5sum:
3811dac18b009edd6c2c4c87725f36d2  openwrt-21.02.3-mvebu-cortexa9-linksys_wrt3200acm-squashfs-sysupgrade.bin

Even unchecking the box "Keep settings" give the same error message.

pepe2k, 2022-06-18:
@Zurd I don't think a clean migration from 17.01 to anything above 19.07 is possible. You would need to perform a forced upgrade (or a clean installation) without preserving settings due to DSA support in recent releases (21.02+).

Cheers,
Piotr

closed 2022-06-18 by pepe2k

Zurd, 2022-06-18:
@pepe2k Thank you, that is the information I was missing.

The webpage to get the firmware is broken because I cannot find my image with a filter WRT3200ACM at https://openwrt.org/toh/views/toh_fwdownload with a release 19.07.*

But I was able to find 19.07.10 at https://downloads.openwrt.org/releases/ in mvebu / cortexa9

Now running OpenWrt 19.07.10 !

Indeed, upgrading to a more recent one warns that the config will be wiped out. I will do this another time.

"Device linksys,rango not supported by this image Supported devices: linksys,wrt3200acm armada-385-linksys-rango linksys,rango - Image version mismatch: image 1.1, device 1.0. Please wipe config during upgrade (force required) or reinstall. Reason: Config cannot be migrated from swconfig to DSA Image check failed."