openwrt/openwrt #10057: Switch settings wrong on WR902AC

micw, 2022-06-15:
Hello,

I am reposting https://github.com/openwrt/mt76/issues/678 at this place since ihave read in a different issue in the mt76 that switch settings does belong in the main repo.

--

I'm analyzing an issue with a "freifunk" firmware based on OpenWRT 21.02. The problem is that a switch setting is not applied correctly on my WR902AC.

I can reproduce it with the following commands:

```
swconfig dev switch0 vlan 1062 set ports "4t 6t"
swconfig dev switch0 vlan 1 set ports "4 6t"
swconfig dev set apply
swconfig dev switch0 show
```

The result I expect is that VLAN 1062 is tagged on both ports and VLAN 1 is tagged in port 6 only and untagged on port 4. But the result is:

```
VLAN 1:
	ports: 4 6t 
VLAN 1062:
	ports: 4 6t 
```

When I swap the `swconfig dev switch0 vlan` commands, the result is

```
VLAN 1:
	ports: 4t 6t 
VLAN 1062:
	ports: 4t 6t 
```

So it seems that the port config of the last command is copied to all VLAN ids.

To be sure that it is not causes by the high VLAN id, I tried it with VLAN 9 instead of 1062 - results are the same.

micw, 2022-06-15:
Additional info: The problem also exists on my other device with plain OpenWRT 19.07.9.
The reason why I thought it works is that if the port is tagged and the port on the other end is untagged with the same VLAN ID, it just works (no idea if I should expect this to work).

So _if_ it's a driver issue (how can I figure this out) - in which repo do I need to create the issue?

neheb, 2022-06-24:
This is the repo. Unfortunately, there's not much that can be done.

The ethernet driver is a hacked up version originating from Mediatek. For mt7621, the upstream DSA ethernet driver is used as it is less buggy and better maintained. For mt7628, I believe there is some support upstream for mt7628 and DSA. I don't know its status completely. Maybe someone knows more.