openwrt/openwrt #10062: Arcadyan WG4хх223 routers: getting MACs from the encrypted partitions

csharper2005, 2022-06-15:
This is a continuation of the discussion that started [here](https://github.com/openwrt/openwrt/pull/9805) (and short summary of the findings from telegram group). 
Cc @Ansuel, @musashino205, @SopaDeMacaco-UmaDelicia, @abyrga, @r3d5ky, @MaxS0niX, @kar200

Affected devices
------------------
[Beeline Smartbox Flash](https://openwrt.org/toh/beeline/smartbox_flash) = Arcadyan WG443223
[MTS WG430223](https://openwrt.org/toh/hwdata/mts/mts_wg430223) = Arcadyan WG430223

Issue description
------------------
* MAC addresses are stored on encrypted (cipher aes-128-cbc) partitions **glbcfg** or **glbcfg2**. As a result we couldn't directly get MAC addresses from OpenWrt

Current workaround
----------------------
* Save MAC addresses to U-Boot env before OpenWrt installation (from OEM firmware) - see WiKi device page

Findings
---------
* /tmp/etc/config/.glbcfg (OEM firmware) contains serial number, WiFi preshared keys etc.
* arccfg util unencrypts  the configuration from mtd8(10): glbcfg or glbcfg2 on every boot
```
~ # arccfg 
fgets partition name ok!the partition is [glbcfg] 
fgets the second partition name ok!the the second partition is [glbcfg2] 
The primary glbcfg is mtd8 
The second glbcfg is mtd10
```
* mtd9(11): **board_data** or **board_data2** also contain MAC address (only *base* MAC). Other addresses should be calculated
* @abyrga & @r3d5ky from telegram group [Beeline SmarBox Giga](https://t.me/smartboxgiga/49944)  found a way how to decrypt a partition:
```
$ dd status=none if=mtd9 bs=1 count=368 skip=256 | openssl aes-128-cbc -d -nopad \
-K 2A4B303D7644395C3B2B7053553C5200 -iv 00000000000000000000000000000000
mac=ec:6c:##:##:##:b8
sn=AG##########
ssid=Beeline_2G_######
psk=###########
ssid2=Beeline_5G_######
psk2=###########
guest_ssid=Beeline_2G_######-Guest
guest_psk=###########
guest_ssid2=Beeline_5G_######-Guest
guest_psk2=###########
group_ssid=BH-Beeline_2G_######
group_psk=###########
group_ssid2=BH-Beeline_5G_######
group_psk2=###########
hw=01
password=#####
```
_Dear OpenWrt team, take this opportunity to inform you that 395 members of the mentioned telegram group are still impatiently waiting for your approval of [Beeline SmartBox GIGA](https://github.com/openwrt/openwrt/pull/4195) and [Beeline SmartBox Turbo+](https://github.com/openwrt/openwrt/pull/4108) PRs._ 

Open points
--------------
- [ ] Decide which decryption tool would be better (including but not limited to the above):
    - [ ]  OpenSSL cli tool (and embed it in OpenWrt firmware for those devices)
    - [ ]  Write a new C tool for decryption and make new package for OpenWrt
    - [ ]  Port https://github.com/wolfSSL/wolfCLU#readme and make a package for OpenWrt
- [ ] It is not clear how to calculate the 5g MAC from the base MAC (if we get it from board_data partition). Here are some examples:

  | MTS WG430223 | MTS WG430223 | MTS WG430223 | MTS WG430223 | Beeline   Smartbox Flash
-- | -- | -- | -- | -- | --
Interface | MAC | MAC | MAC | MAC | MAC
label | A4:xx:xx:51:xx:F4 | 8C:xx:xx:34:xx:10 | 30:xx:xx:77:xx:9E | A4:CE:DA:4x:xx:x4 | 30:xx:xx:51:xx:09
LAN | A4:xx:xx:51:xx:F6 | 8C:xx:xx:34:xx:12 | 30:xx:xx:77:xx:A0 | A4:CE:DA:4x:xx:x6 | 30:xx:xx:51:xx:09
WAN | A4:xx:xx:51:xx:F4 | 8C:xx:xx:34:xx:10 | 30:xx:xx:77:xx:9E | A4:CE:DA:4x:xx:x4 | 30:xx:xx:51:xx:06
WLAN_2g | A4:xx:xx:51:xx:F5 | 8C:xx:xx:34:xx:11 | 30:xx:xx:77:xx:9F | A4:CE:DA:4x:xx:x5 | 30:xx:xx:51:xx:07
WLAN_5g | A6:xx:xx:21:xx:F5 | 8E:xx:xx:24:xx:11 | 30:xx:xx:27:xx:9F | A6:CE:DA:2x:xx:x5 | 32:xx:xx:41:xx:07

Any ideas?

Ansuel, 2022-06-15:
@cotequeiroz Is it possible to have wolfCLU? seems a good solution considering we already ship router with wolfssl

Writing a new C tool IMHO is overkill but who knows... 

For 5ghz the rule is to follow how the original device behave so it should be set based on what the device does... we already have function to increment bits of the address at an offset

cotequeiroz, 2022-06-15:
I'll work on packaging wolfCLU.  It appears to be rather trivial.

SopaDeMacaco-UmaDelicia, 2022-06-15:
@Ansuel We would love to make it follow the original firmware behave but the problem with the 5G mac-address is that we don't understand yet how to calculate it. Check fourth pairs of 2g and 5g of the same router, the difference even between same models is kinda random.

Ansuel, 2022-06-15:
@SopaDeMacaco-UmaDelicia think we should check the scripts in the original firmware for that... could also be that it's random... (but I doubt)
are we sure it's not just a static hex increase?

cotequeiroz, 2022-06-15:
If I configure wolfssl with `--enable-wolfclu`, `libwolfssl*_mipsel_24kc.ipk` jumps from 477K to 531K (a 54Kb, or 11.3% increase).  Is this acceptable across the board?

While one could, theoretically, only enable this on a subtarget level, there may be ABI incompatibility problems with packages linked to a library with different options.

Another option, would be to disable unneeded functionality in wolfclu, **if able**.  What `--enable-wolfclu` does is include `ed25519` and `certreq` support.

SopaDeMacaco-UmaDelicia, 2022-06-15:
@Ansuel Nah, we didn't find any signs of how 5G mac is made yet. Here full nanddump https://t.me/smartboxgiga/50060

Ansuel, 2022-06-15:
My god 400 members with this router O_O

SopaDeMacaco-UmaDelicia, 2022-06-15:
@Ansuel nah, not only this router. The group is for merged #4922 #9805 not merged yet #4770 #4195 #4108.

csharper2005, 2022-06-16:
> My god 400 members with this router O_O

That's not all. :) There is also a community on 4 pda forum... And some of users use "golden orb's rooter" based on Xiaomi's NAND layout with breed. https://github.com/openwrt/openwrt/pull/4195 and https://github.com/openwrt/openwrt/pull/4108 are the cheapest and the most common on the market with OpenWrt (unofficial builds only for now).

r3d5ky, 2022-06-16:
> @SopaDeMacaco-UmaDelicia think we should check the scripts in the original firmware for that... could also be that it's random... (but I doubt) are we sure it's not just a static hex increase?

I already looked at the scripts and binaries that are involved in the calculation, and they match with our manual calculation for all MACs, except for that 5G MAC. Part for its calculation is just commented in it, and even if this would be uncommented, it still didn't make sense, because what it produces doesn't match with the factory MAC. It should be static increase of the last byte of base MAC, but it is don't somehow.

cotequeiroz, 2022-06-22:
I was working on adding a wolfssl CLU package, but I don't really think it is worth it.  I'm inclined to say that writing a small C program is better.  Such a program would only be about 15k uncompressed, so much smaller than the 54kb footprint that supporting it would cost.  It does not appear to be stable, or even ready at all.  The last release will not even build with the wolfssl 5.3.0.  Current snapshot does.
It feels like adding wolfCLU to use aes-128-cbc to decrypt a file is "is like buying a 747 for the free peanuts."

This here should do the trick, and would compile with openssl, or with wolfssl using `-DUSE_WOLFSSL`:
``` C
#include <errno.h>
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>

#ifdef USE_WOLFSSL
# include <wolfssl/options.h>
# include <wolfssl/openssl/evp.h>
#else
# include <openssl/evp.h>
#endif

int do_crypt(FILE *in, FILE *out, const char *key, const char *iv, int enc,
	     int padding)
{
    EVP_CIPHER_CTX *ctx;
    unsigned char inbuf[1024], outbuf[1024 + EVP_MAX_BLOCK_LENGTH];
    int inlen, outlen;

    ctx = EVP_CIPHER_CTX_new();
    EVP_CipherInit_ex(ctx, EVP_aes_128_cbc(), NULL, key, iv, enc);
    EVP_CIPHER_CTX_set_padding(ctx, padding);

    for (;;) {
        inlen = fread(inbuf, 1, 1024, in);
	fprintf(stderr, "inlen=%d.\n", inlen);
        if (inlen <= 0)
            break;
        if (!EVP_CipherUpdate(ctx, outbuf, &outlen, inbuf, inlen)) {
            EVP_CIPHER_CTX_free(ctx);
            return -1;
        }
        fwrite(outbuf, 1, outlen, out);
    }
    fprintf(stderr, "Got out of the loop.\n", inlen);
    if (!EVP_CipherFinal_ex(ctx, outbuf, &outlen)) {
        EVP_CIPHER_CTX_free(ctx);
        return -1;
    }
    fwrite(outbuf, 1, outlen, out);

    EVP_CIPHER_CTX_free(ctx);
    return 0;
}

static void check_enc_dec(const int enc)
{
    if (enc == -1)
	return;
    fprintf(stderr, "Error: both -d and -e were specified.\n");
    exit(EXIT_FAILURE);
}

static void show_usage(const char* name)
{
    fprintf(stderr, "Usage: %s: [-d | -e] [-n] -k key -i iv\n"
    		    "-d = decrypt; -e = encrypt; -n = no padding\n", name);
}

int main(int argc, char *argv[])
{
    int enc = -1;
    unsigned char *iv = NULL;
    unsigned char *key = NULL;
    long len;
    int opt;
    int padding = 1;
    int ret;

    while ((opt = getopt(argc, argv, "dei:k:n")) != -1) {
	switch (opt) {
	case 'd':
	    check_enc_dec(enc);
	    enc = 0;
	    break;
	case 'e':
	    check_enc_dec(enc);
	    enc = 1;
	    break;
	case 'i':
	    iv = OPENSSL_hexstr2buf((const char *)optarg, &len);
	    if (iv == NULL || len != 16) {
		fprintf(stderr, "Error setting IV to %s. The IV should be 16 bytes, encoded in hex.\n", 
			optarg);
		exit(EINVAL);
	    }
	    break;
	case 'k':
	    key = OPENSSL_hexstr2buf((const char *)optarg, &len);
	    if (key == NULL || len != 16) {
		fprintf(stderr, "Error setting key to %s. The key should be 16 bytes, encoded in hex.\n", 
			optarg);
		exit(EINVAL);
	    }
	    break;
	case 'n':
	    padding = 0;
	    break;
	default:
	    show_usage(argv[0]);
	    exit(EINVAL);
	}
    }
    if (iv == NULL || key == NULL) {
	fprintf(stderr, "Error: %s not set.\n", key ? "iv" : (iv ? "key" : "key and iv"));
	show_usage(argv[0]);
	exit(EXIT_FAILURE);
    }
    ret = do_crypt(stdin, stdout, key, iv, !!enc, padding);
    if (ret)
	fprintf(stderr, "Error during crypt operation.\n");
    OPENSSL_free(iv);
    OPENSSL_free(key);
    return ret;
}
```

cotequeiroz, 2022-06-22:
Which one do you mean?  The C program, or WolfCLU, or both?

Ansuel, 2022-06-22:
@cotequeiroz honestly shipping a small tool for the task is better than wolfclu... While i would still consider the wolfclu package a nice addition IMHO for the only use of decrypting a partition i think the tool you provided is a better solution.

What do you think?

Ansuel, 2022-06-22:
i would build the c program only for this specific target

MaxS0niX, 2022-06-23:
> As for the C program, is it worth it to add it as package for all arches, or should it be built only for those in need?

So far, we know what is required for only two devices.)

csharper2005, 2022-07-09:
> we already have function to increment bits of the address at an offset

@Ansuel Do we need a function to change the whole byte? Or 4 calls of `macaddr_setbit / macaddr_unsetbit` is ok?

MaxS0niX, 2022-08-03:
My example MAC's Beeline SmartBox Flash - Arcadyan WG443223
OEM FW v.1.00.12
```
LAN(label)  EC:6C:9A:8x:xx:xD
WAN         EC:6C:9A:8x:xx:xA
WLAN_2g     EE:6C:9A:Cx:xx:xB
WLAN_5g     EC:6C:9A:8x:xx:xB
```

csharper2005, 2022-08-03:
@MaxS0niX, thanks. Your example is different from all the others. 2g MAC is calculated according to algorithm 5g MAC and vice versa. Therefore,  now I have no ideas about a common calculation algorithm. We have to keep the current workaround.

> My example MAC's Beeline SmartBox Flash - Arcadyan WG443223 OEM FW v.1.00.12
> 
> ```
> LAN(label)  EC:6C:9A:8x:xx:xD
> WAN         EC:6C:9A:8x:xx:xA
> WLAN_2g     EE:6C:9A:Cx:xx:xB
> WLAN_5g     EC:6C:9A:8x:xx:xB
> ```

@MaxS0niX, thanks. Your example is different from all the others. 2g MAC is calculated according to algorithm 5g MAC and vice versa. Therefore,  now I have no ideas about a common calculation algorithm. We have to keep the current workaround.

csharper2005, 2022-08-16:
[image]
An example from tg user @iii_stan. Mts wg430223.

csharper2005, 2022-08-18:
@Ansuel at the time we have 9 examples confirming the theory and one [strange example ](https://github.com/openwrt/openwrt/issues/10062#issuecomment-1204200046) with flipped 2g / 5g MACs. What would you recommend in this situation?

Ansuel, 2022-08-18:
Is it possible to find more example? The serial is not generated by the
system right?

Woder if we find a reason for the flipped Mac... Different revision?

LuckySpecialist, 2022-08-18:
> An example from tg user @iii_stan. Mts wg430223.

This is data from my router, so if you need to check something additionally, please contact)

csharper2005, 2022-08-18:
> Is it possible to find more example? The serial is not generated by the system right? Woder if we find a reason for the flipped Mac... Different revision?

I managed to reproduce the issue. Need to:
1. Downgrade to the oldest firmware 1.00;
2. Make Factory reset.
After making these steps I got flipped MACs too.

Solution - upgrade to v1.0.15. This reverts normal MAC addresses. Therefore, I think we have dealt with this issue.

@Ansuel Can I count on your support in merging the fix to master / 22.03?
https://github.com/openwrt/openwrt/pull/10233

@LuckySpecialist thanks!

Ansuel, 2022-08-18:
Ok so the firmware patch the Mac address in later version... Good then we
have a correct pattern.

closed 2022-08-19 by jow-

csharper2005, 2022-08-20:
I offer my sincere thanks to everyone who has contributed contributed to fixing the issue! I'm impressed by the level of collaboration. :)

system1357, 2022-08-27:
I have a Arcadyan WG630223 and it needs the same mac fix, but the board_config partition layout and the decryption key are both different.
I've figured out the key and successfully decrypted the partition.
Should I try to add new device support first or add keys and partition info to this script?

Ansuel, 2022-08-28:
Why not do both? Create a pr for new device and include the extension of
the script...

A suggestion would be mod the function to pass the key as a function arg
and use such arg to decrypt the partition.

commit a6b0d080 (2022-08-19): ramips: get MAC addr from the encrypted partition (WG4хх223)

This commit resolves #10062. Adds decryption of the Arcadyan WG4xx223
configuration partition (board_data)to get base MAC address from it.
As a result, after this change the hack with saving MAC addressees to
u-boot-env before installation of OpenWrt is no longer necessary.

This is necessary for the following devices:
- Beeline Smartbox Flash (Arcadyan WG443223)
- MTS WG430223 (Arcadyan WG430223)

Example:
+----------------+-------------------+------------------------+
|                |    MTS WG430223   | Beeline Smartbox Flash | [... cut]
commit ecd8f7e2 (2022-08-19): ramips: get MAC addr from the encrypted partition (WG4хх223)

This commit resolves #10062. Adds decryption of the Arcadyan WG4xx223
configuration partition (board_data)to get base MAC address from it.
As a result, after this change the hack with saving MAC addressees to
u-boot-env before installation of OpenWrt is no longer necessary.

This is necessary for the following devices:
- Beeline Smartbox Flash (Arcadyan WG443223)
- MTS WG430223 (Arcadyan WG430223)

Example:
+----------------+-------------------+------------------------+
|                |    MTS WG430223   | Beeline Smartbox Flash | [... cut]