openwrt/openwrt #10069: make kernel_menuconfig get an error

kylongmu, 2022-06-16:
Build from this copy https://github.com/OpenNuvoton/NUC980-OpenWrt
after 'make kernel_menuconfig' get this error:
```
gcc  -I.   -I/home/mqy/NUC980-OpenWrt/staging_dir/host/include -I/home/mqy/NUC980-OpenWrt/staging_dir/host/usr/include   -O2 -I/home/mqy/NUC980-OpenWrt/staging_dir/host/include -I/home/mqy/NUC980-OpenWrt/staging_dir/host/usr/include  -MT c-stack.o -MD -MP -MF $depbase.Tpo -c -o c-stack.o c-stack.c &&\
mv -f $depbase.Tpo $depbase.Po
In file included from /usr/include/signal.h:328,
                 from ./signal.h:52,
                 from c-stack.c:49:
c-stack.c:55:26: error: missing binary operator before token "("
   55 | #elif HAVE_LIBSIGSEGV && SIGSTKSZ < 16384
      |                          ^~~~~~~~
make[6]: *** [Makefile:1910: c-stack.o] Error 1
make[6]: Leaving directory '/home/mqy/NUC980-OpenWrt/build_dir/host/m4-1.4.18/lib'

```

nbd168, 2022-06-16:
This is a fork of an old version which we can't support.

nbd168, 2022-06-16:
Sorry about that, I clicked just as you closed the issue :)

closed 2022-06-16 by nbd168