openwrt/openwrt #10074: ramips: MT7628: WiFi signal disappear under heavy load

DragonBluep, 2022-06-17:
MT7628 WiFi issue has been here for years, And there have been plenty of reports and discussions:

https://github.com/openwrt/openwrt/issues/9446 Xiaomi Mi Router 4A (MIR4A) 100M - "did not acknowledge authentication response"
https://github.com/openwrt/openwrt/issues/9219 TL-WR841N v13 (SoC MT7628AN) wifi disappears after some usage
https://github.com/openwrt/openwrt/issues/9063 Problem with TL-WR842N v5 wifi and static IP
https://github.com/openwrt/openwrt/issues/8757 MT7688 wireless signal disappears under heavy load
https://github.com/openwrt/openwrt/issues/8314 TL-WR841N v13 (ramips) unstable Ethernet and WiFi in 19.07.4
https://github.com/openwrt/openwrt/issues/8184 MT7628an: lose radio whilst downloading jarge data

https://forum.openwrt.org/t/mt7628-dsa-driver/114772
https://github.com/openwrt/mt76/issues/628

From @Linaro1985's [investigation](https://github.com/openwrt/mt76/issues/628#issuecomment-1001170423), this problem is probably caused by the ethernet driver.

Linaro1985, 2022-06-17:
@DragonBluep this is easily checked by usb<->ethernet adapter on any MT7628 device.
No problem in this chain ethernet<->usb<->wifi.

Pilabol, 2022-06-18:
I have the same problem on c50 v4. Any one kowns how to solve it?

stre425, 2022-06-19:
Since this is only happening under heavy load and it seems to be caused by MCU hangs, could this be a thermal issue? Has anyone tried to attach a heatsink to the MT7628 chip and checked if this improves the stability?

Linaro1985, 2022-06-19:
No. I think it's not a thermal issue. On the upstream mtk kernel driver works fine. There is also no problem with the usb-ethernet adapter. This maybe due to the dma operations or the handling of the netif queue in current driver. I did the following:
1. Separate tx, rx irq handling
2. Add READ_ONCE and WRITE_ONCE
But there are no any improvements.

DragonBluep, 2022-06-20:
It's unlikely the temperature reason, constant stress testing with iperf3 doesn't break it. This problem sometimes occurs when open an online video produces an instant massive data upload and download. 
Using speedtest.net can trigger this bug with a high probability.

Linaro1985, 2022-06-20:
It is easy to reproduce if:
1. no_scan 1 for wifi
2. launch torrents

My wifi client is based on MT7921 chip.

krisux, 2022-07-26:
Same on r6220. The event occurs while watching uhd movies from samba.

wenqingj, 2022-09-07:
Just tried out the newly released v22.03.0, on a TL-WR841N (v13), and the issue remains.
Anyone knows of stock firmware (non-OpenWrt) working well on a MT7628-based router? If not, is it possible that the MT7628 hardware is flawed and hence a good working firmware is not possible?

DragonBluep, 2022-09-07:
> Anyone knows of stock firmware (non-OpenWrt) working well on a MT7628-based router? If not, is it possible that the MT7628 hardware is flawed and hence a good working firmware is not possible?

The stock firmware has always been stable. And this project is also very stable https://bitbucket.org/padavan/rt-n56u.

DragonBluep, 2022-09-16:
The problem still exists under the latest 5.15.68 kernel and 2022.09.15 mt76 driver, tested on Xiaomi router 4C. There is no relevant log output.

sngvy, 2022-09-22:
Yeah, my 4C still crashes on heavy load too. WiFi may be disconnected or my LG TV get stucked on Wink app for example. I think some settings of WiFi point such as fragmentation or RTS make router unstable

ggbruno, 2022-09-24:
I am using an updated version of the binary firmware. It improves my C20v5...
I get this firmware from the last C50v6 stock firmware...
https://github.com/ggbruno/mt76/commit/607db71f98b532842a4f9745a8703a0f3462973b

DragonBluep, 2022-09-24:
@ggbruno Sadly it doesn't make a difference for me, I've been using this new version since last year. https://github.com/DragonBluep/mt76/commits/exp

sngvy, 2022-10-05:
@DragonBluep Do you have any ideas? The problem is not in mac binary firmware...

adelton, 2022-10-25:
> I am using an updated version of the binary firmware. It improves my C20v5...

@ggbruno Would you have some pointer to the description of using the binary firmware, especially in situations when https://downloads.openwrt.org/releases/22.03.0/targets/ramips/mt76x8/openwrt-imagebuilder-22.03.0-ramips-mt76x8.Linux-x86_64.tar.xz is used?

Rexadev, 2022-12-25:
Problem on 22.03.2 not 2.03.1 - TL-WR850N v2

wenqingj, 2023-01-30:
Just tried the latest version (22.03.3) and the issue still remains.
Wondering if this is an issue inside Linux (kernel, as well as firmware/driver for supported hardware) or in OpenWrt itself? I've the impression that it's the former but I'm not sure.

sngvy, 2023-01-31:
> Just tried the latest version (22.03.3) and the issue still remains. Wondering if this is an issue inside Linux (kernel, as well as firmware/driver for supported hardware) or in OpenWrt itself? I've the impression that it's the former but I'm not sure.

mt76 driver is very unstable, try to flash Padavan

DragonBluep, 2023-01-31:
> mt76 driver is very unstable, try to flash Padavan

As discussed earlier. It seems that this problem is not caused by mt76, but by rt3050 ethernet driver.

@lynxis Do you have any ideas about it? Hope I'm not disturbing your work.

roger-, 2023-02-07:
> As discussed earlier. It seems that this problem is not caused by mt76, but by rt3050 ethernet driver.

Is there a straightforward way to disable the Ethernet driver to see if that helps? I have a few wireless only devices that are experiencing similar issues.

DragonBluep, 2023-02-07:
> > As discussed earlier. It seems that this problem is not caused by mt76, but by rt3050 ethernet driver.
> 
> Is there a straightforward way to disable the Ethernet driver to see if that helps? I have a few wireless only devices that are experiencing similar issues.

@roger- Unset these kernel configs should work. This may brick your device. Be careful.
https://github.com/openwrt/openwrt/blob/f28a604df4659a572dc3b147b65365fef6dc90d9/target/linux/ramips/mt76x8/config-5.10#L123-L126

whoizit, 2023-02-21:
This is not about SOC. I have TP-Link tl-wr842nd v2.1 on Atheros AR9341 rev 3 and after upgrade from latest libreCMC to oWRT 22.03.3 I getting wi-fi hangs every day.

DragonBluep, 2023-04-26:
Has anyone tested the latest master branch? It seems that the issue has been fixed.

tomo90, 2023-04-27:
TL-WR840 user here.
A can say the issue really seems to be fixed with 22.03.4 for me.
I tried a typical synchronization task using Syncthing with about 7k files of several kilobytes in size. 
Not a single disconnect.
The same task fails multiple times using previous versions.

Linaro1985, 2023-04-27:
@DragonBluep the issue still exists.

DragonBluep, 2023-04-28:
@Linaro1985 Recently, I used `natflow` module to reduce CPU load. It seems to be helpful for this issue.
[0001-firewall4-drop-depends-on-kmod-nft-offload.patch](https://github.com/openwrt/openwrt/files/11349243/0001-firewall4-drop-depends-on-kmod-nft-offload.patch)
[0002-ramips-mediatek-hwnat-add-natflow-flow-offload-suppo.patch](https://github.com/openwrt/openwrt/files/11349244/0002-ramips-mediatek-hwnat-add-natflow-flow-offload-suppo.patch)
[0003-package-import-packages-from-x-wrt-20230423.patch](https://github.com/openwrt/openwrt/files/11349245/0003-package-import-packages-from-x-wrt-20230423.patch)
[0004-package-natflow-change-default-ipaddr-to-192.168.1.1.patch](https://github.com/openwrt/openwrt/files/11349246/0004-package-natflow-change-default-ipaddr-to-192.168.1.1.patch)
[0005-package-natflow-refine-dependency.patch](https://github.com/openwrt/openwrt/files/11349247/0005-package-natflow-refine-dependency.patch)
[0006-package-natflow-move-config-file-to-natflow-boot.patch](https://github.com/openwrt/openwrt/files/11349248/0006-package-natflow-move-config-file-to-natflow-boot.patch)
This module comes from x-wrt, and I have made some minor changes. Patch 0002 is only valid for mt7621, so you can just skip it. After applying this series of patches, you can select the `natflow-boot` package in menuconfig.

lukasz1992, 2023-05-30:
I wonder if making these changes would help for the issue:
- https://github.com/openwrt/openwrt/blob/main/target/linux/ramips/mt76x8/config-5.15 - enabling CONFIG_NET_VENDOR_MEDIATEK and disabling CONFIG_NET_VENDOR_RALINK
- rewrite dts for devices, like here: https://github.com/openwrt/openwrt/commit/30644bc579e771e4b6dcc165ca966124e0259e1e

DragonBluep, 2023-06-01:
> I wonder if making these changes would help for the issue:
> 
> * https://github.com/openwrt/openwrt/blob/main/target/linux/ramips/mt76x8/config-5.15 - enabling CONFIG_NET_VENDOR_MEDIATEK and disabling CONFIG_NET_VENDOR_RALINK
> * rewrite dts for devices, like here: [30644bc](https://github.com/openwrt/openwrt/commit/30644bc579e771e4b6dcc165ca966124e0259e1e)

Upstream mt7628 driver seems does not support DSA, we may need some additional changes.

Linaro1985, 2023-06-01:
@DragonBluep, just tested your's patch with: iperf3 -p 5001 -i 1 -t 9999 -c srv_ip -P 3 -R
The problem is still exists( I want to try the proprietary ethernet driver (for example from Padavan firmware) when I have free time. I wonder if this will help.

Linaro1985, 2023-06-09:
Hi! @DragonBluep I can't resolve an issue with changes for switch initialization from linux-3.10.14. There is an intresting patch in gardena's repo: https://github.com/rettichschnidi/linux/commit/3d00418d0aa259764b709f63e6855bcab88ab80c . But it doesn't help either. Therefore, I would like to try using the driver from the linux-3.10.14 kernel for simple test. Maybe you have config example for MT7628?

DragonBluep, 2023-06-09:
> Hi! @DragonBluep I can't resolve an issue with changes for switch initialization from linux-3.10.14. There is an intresting patch in gardena's repo: [rettichschnidi/linux@3d00418](https://github.com/rettichschnidi/linux/commit/3d00418d0aa259764b709f63e6855bcab88ab80c) . But it doesn't help either. Therefore, I would like to try using the driver from the linux-3.10.14 kernel for simple test. Maybe you have config example for MT7628?

@Linaro1985 Thank you for your efforts. If you mean the kernel config file, it's here https://github.com/DragonBluep/mtk-openwrt-sdk/blob/main/target/linux/ramips/mt7628/config-3.10. 

You can also directly build these openwrt based codes. Simply run `make menuconfig` to select the mt7628 target and then `make -jn`. `make kernel_menuconfig` should be able to customize the kernel config file. It is worth noting that using older build systems, such as ubuntu16, can reduce compilation errors. https://github.com/DragonBluep/mtk-openwrt-sdk

sngvy, 2023-06-15:
It may seem strange, but for me helped using WPA/WPA2 encryption instead WPA2 on Mi Router 4C with 7628AN CPU

lukasz1992, 2023-06-16:
@sngvy test it more - it will crash

hafizhlf, 2023-07-03:
Hi, i dont know if this will help or not, but i try pcwrt, and it is stable, no wifi crash

https://www.pcwrt.com/downloads/

lukasz1992, 2023-07-03:
These people are thiefs and do not provide source code for what they sell.

They use proprieatery driver - it does not crash

Rexadev, 2023-07-04:
> They use proprieatery driver - it does not crash

They build a driver and then didn't made it open ??

@lukasz1992

lukasz1992, 2023-07-05:
You can find the most recent driver here: http://vonger.cn/misc/vocore2/MT7628_LinuxAP_V4.1.1.0_DPA_20190925.tar.bz2

But they also closed sources for whole OS (including their UI).

Rexadev, 2023-07-05:
What is https://vonger.cn/ ?

@lukasz1992

Rexadev, 2023-07-05:
> They use proprieatery driver - it does not crash

How did they make drivers on there own ?

hasanhabali, 2023-07-23:
There is no official patch, right?

lukasz1992, 2023-07-24:
no fix for opensource drivers

DragonBluep, 2023-07-24:
The worst thing is that there is no relevant log output, so we cannot debug it at all.😑

lukasz1992, 2023-07-24:
Nothing to log about :( from a perspective of driver it is just mcu hang.

DragonBluep, 2023-07-24:
It seems that PandoraBox team solved a similar problem on mt7603 years ago.
https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=192526&ordertype=1
```
修正MT7603E高温掉线问题
```
I think mt7628 issue should be a similar one. Perhaps @981213 has some idea.

lukasz1992, 2023-07-24:
They just do not use mt76 driver.

hasanhabali, 2023-07-24:
> 

are you sure?

lukasz1992, 2023-07-24:
These links do not work, can not check it.

http://downloads.pandorabox.com.cn/pandorabox/
http://bug.pandorabox.com.cn/

hasanhabali, 2023-07-24:
i think china domains and sites not safe. so i am using openwrt

981213, 2023-07-25:
@DragonBluep 

> Perhaps @981213 has some idea.

That build uses the proprietary mediatek driver, not mt76, so any fix there won't apply here even if anyone got access to the driver source code.

uncharted-user, 2023-07-26:
I can also see similar issue on Netgear R6120 with mt76x8. My Wi-Fi connections (5GHz and 2.4GHz) drop even due to playing a YouTube video at 720p on a single device. I have to manually disconnect and reconnect it. I never saw that issue with factory firmware. Is there any solution yet?

hasanhabali, 2023-07-27:
pretty bad situation. it happens when downloading files in router 4c, it gets stuck at exactly 200 mb and the same situation

uncharted-user, 2023-07-29:
Exactly, most of the time files ended up corrupted. There is probably a [workaround](https://forum.openwrt.org/t/wi-fi-connections-drop-even-due-to-playing-a-youtube-video/166908) which although handicap your dual-band, but it seems working for now.

sngvy, 2023-08-13:
Did anyone tried to set RTS/CTS threshold to 1024? It helped don't crash AP on my 4C

DragonBluep, 2023-08-17:
Mission completed. Thank you all!
Fixed by https://github.com/openwrt/mt76/commit/b14c2351ddb8601c322576d84029e463d456caef

closed 2023-08-17 by DragonBluep

wenqingj, 2023-08-29:
I tested OpenWrt 23.05.0-rc3 (latest at the moment) on a TL-WR841N v13 (the latter uses MT7628 chip). The rc3 does not (yet) contain the fix of openwrt/mt76@b14c235, so I set its Fragmentation Threshold to 2346, just like DragonBluep pointed out in a comment in https://github.com/openwrt/mt76/issues/793. Then I tried to transfer a very large file (>10GB) by sftp via wifi. It caused, like before, the wifi to disappear after a short while.

Does that mean openwrt/mt76@b14c235 might not fix the issue, if its effect is same as setting the Frangmentation Threshold value to 2346?

dfateyev, 2023-08-29:
@wenqingj build a firmware for your device from "openwrt-23.05" branch and re-test with it. There was a couple of other changes in the tree.

wenqingj, 2023-09-09:
@dfateyev I flashed today's snapshot and transferred a 10+ GB file with sftp, and it completed well like charm. Looks like the bug is dead for good. :-)

roger-, 2023-09-29:
I’ve had similar issues on the MT7620n — does the recent patch apply to that?

SFAZ38, 2023-09-30:
> I’ve had similar issues on the MT7620n — does the recent patch apply to that?

I think it doesn't. Because it's uses kmod-mt76x2. This patch is for kmod-mt7603. But it's just me, you need to wait for other's to comment your message.

Linaro1985, 2023-10-03:
Hi. @DragonBluep I made new test on OpenWrt 23.05-rc4 and recorded a video with some debug that shows the problem with a beacon stuck https://disk.yandex.ru/i/_9PFpucSFN-62Q
We can see that after the problem occurs, tail in xmit-queue stops changing and is zero. Maybe this will somehow help find the reason of issue.

[30 more replies]