openwrt/openwrt #10076: no ccache for gcc compile

bam80, 2022-06-17:
During gcc build:
```
...
  make[3] -C toolchain/gcc/initial compile
```
, ccache used only at the very beginning (can be monitored with `CCACHE_DIR=.ccache staging_dir/host/bin/ccache -s`).
After that short period of time, ccache is not used at all - i.e. even misses do not get counted.

Considering gcc build time is huge, it's rather sensitive drawback for overall build.

jospezial, 2022-06-18:
~/.cache/ccache/ is used too. But possibly used by the system ccache executable only before the openwrt ccache or gcc executable takes over.

jospezial, 2022-06-18:
2 issues for the same topic?  #10073

pepe2k, 2022-06-18:
@bam80 please, don't file more than one issues about the same problem.
Closing this one, please use https://github.com/openwrt/openwrt/issues/10073.

bam80, 2022-06-19:
> ~/.cache/ccache/ is used too. But possibly used by the system ccache executable only before the openwrt ccache or gcc executable takes over.

Must system ccache executable be first in PATH for that?

bam80, 2022-06-19:
> @bam80 please, don't file more than one issues about the same problem.
> Closing this one, please use #10073.

The issues are actually different, that is why I opened a this one additionally.
The #10073 was updated and solved.

bam80, 2022-06-21:
> ~/.cache/ccache/ is used too. But possibly used by the system ccache executable only before the openwrt ccache or gcc executable takes over.

@jospezial 
Did you mean ~/.ccache/? I see no files added to this directory during compile (even or early stage when in-tree ccache is not yet accessible)

bam80, 2022-06-21:
> > ~/.cache/ccache/ is used too. But possibly used by the system ccache executable only before the openwrt ccache or gcc executable takes over.
> 
> @jospezial Did you mean ~/.ccache/? I see no files added to this directory during compile (even or early stage when in-tree ccache is not yet accessible).
> 
> PS: prepending /usr/lib/ccache to PATH didn't make it work either

bam80, 2022-06-22:
> Should we redefine CC to gcc somewhere in the makefiles?

Just setting it alone had huge performance boost for compilation time: **25min** vs 54min before, and ~70min without cache.