openwrt/openwrt #10077: VDSL - error in call ubus dsl metrics

turboproc, 2022-06-18:
Call `ubus call dsl metrics` shows an error when values get into the high ranges. Issue seems related to unsigned versus signed numbers and therefore showing negative counter values.

Example
```
root@dsl-02:~# dsl_cpe_pipe.sh rtsg 1
nReturn=0 nDirection=1 nRxCorruptedTotal=146140769 nRxUncorrectedProtected=2389737 nRxRetransmitted=0 nRxCorrected=143751032 nTxRetransmitted=4147527141
```

and
```
root@dsl-02:~# ubus call dsl metrics
{
        "api_version": "4.17.18.6",
        "firmware_version": "5.9.1.4.0.7",
        "chipset": "Lantiq-VRX200",
        "driver_version": "1.5.17.6",
        "state": "Showtime with TC-Layer sync",
        "state_num": 7,
        "up": true,
        "uptime": 163412,
        "atu_c": {
                "vendor_id": [
                        181,
                        0,
                        66,
                        68,
                        67,
                        77,
                        178,
                        30
                ],
                "vendor": "Broadcom 178.30",
                "system_vendor_id": [
                        181,
                        0,
                        66,
                        68,
                        67,
                        77,
                        0,
                        0
                ],
                "system_vendor": "Broadcom",
                "version": [
                        49,
                        57,
                        46,
                        48,
                        46,
                        51,
                        57,
                        46,
                        50,
                        32,
                        86,
                        69,
                        95,
                        49,
                        49,
                        95
                ],
                "serial": [
                        65,
                        65,
                        49,
                        53,
                        52,
                        56,
                        70,
                        83,
                        49,
                        76,
                        81,
                        45,
                        52,
                        51,
                        0,
                        0,
                        0,
                        0,
                        0,
                        0,
                        0,
                        0,
                        0,
                        0,
                        0,
                        0,
                        0,
                        0,
                        0,
                        0,
                        0,
                        0
                ]
        },
        "power_state": "L0 - Synchronized",
        "power_state_num": 0,
        "xtse": [
                0,
                0,
                0,
                0,
                0,
                0,
                0,
                2
        ],
        "annex": "B",
        "standard": "G.993.2",
        "profile": "17a",
        "mode": "G.993.2 (VDSL2, Profile 17a, with down- and upstream vectoring)",
        "upstream": {
                "vector": true,
                "trellis": true,
                "bitswap": true,
                "retx": true,
                "virtual_noise": false,
                "interleave_delay": 0,
                "data_rate": 32986000,
                "latn": 15.500000,
                "satn": 15.300000,
                "snr": 5.000000,
                "actps": -90.100000,
                "actatp": 11.500000,
                "attndr": 33411000
        },
        "downstream": {
                "vector": true,
                "trellis": true,
                "bitswap": true,
                "retx": true,
                "virtual_noise": true,
                "interleave_delay": 130,
                "data_rate": 109022000,
                "latn": 13.600000,
                "satn": 13.800000,
                "snr": 5.100000,
                "actps": -90.100000,
                "actatp": 6.900000,
                "attndr": 109494272
        },
        "errors": {
                "near": {
                        "es": 13724,
                        "ses": 6,
                        "loss": 2,
                        "uas": 3662,
                        "lofs": 0,
                        "fecs": 0,
                        "hec": 0,
                        "ibe": 0,
                        "crc_p": 1480750,
                        "crcp_p": 0,
                        "cv_p": 89404320,
                        "cvp_p": 0,
                        "rx_corrupted": 57808668,
                        "rx_uncorrected_protected": 253572,
                        "rx_retransmitted": 0,
                        "rx_corrected": 57555096,
                        "tx_retransmitted": 38430
                },
                "far": {
                        "es": 60,
                        "ses": 22,
                        "loss": 0,
                        "uas": 3662,
                        "lofs": 0,
                        "fecs": 7169121,
                        "hec": 0,
                        "ibe": 0,
                        "crc_p": 0,
                        "crcp_p": 0,
                        "cv_p": 0,
                        "cvp_p": 0,
                        "rx_corrupted": 146140769,
                        "rx_uncorrected_protected": 2389737,
                        "rx_retransmitted": 0,
                        "rx_corrected": 143751032,
                        "tx_retransmitted": -147440155
                }
        },
        "erb": {
                "sent": 681299,
                "discarded": 0
        }
}
```
As can be seen the value `"tx_retransmitted": -147440155` is exactly the 2-complement of `nTxRetransmitted=4147527141`
Obviously this issue becomes visible only in situations where DSLAM counters are in the high range which apparently was the situation in my case.

xdarklight, 2022-06-18:
try changing https://github.com/openwrt/openwrt/blob/5ee1e045174bad1b7f2754bb022346acc60220f5/package/network/config/ltq-vdsl-app/src/src/dsl_cpe_ubus.c#L149 to `blobmsg_add_u64(&b, id, value);`
my understanding is that ubus always considers integers as signed. so if we want 32-bit wide unsigned numbers I think we need to use 64-bits instead.

turboproc, 2022-06-18:
OK, let me try.

dhewg, 2022-06-18:
If you look at `m_array` a few lines below it has the same "mismatch": argument is u8 but uses the wider `blobmsg_add_u16´.
IIRC for the very same reason, so `blobmsg_add_u64` should fix it!

sqrwf, 2022-06-18:
It's a fix, but it's slightly unbeautiful to call `blobmsg_add_u64` within a function named `m_u32`. It can confuse the heck out of someone unfamiliar with the inner workings of ubus (and why it's mostly signed/unsigned-agnostic). So maybe add a descriptive comment in `m_u32`?

turboproc, 2022-06-18:
Applied both fixes with for now a positive result as shown below
```
                        "tx_retransmitted": 4147529067
                }
        },
        "erb": {
                "sent": 4979,
                "discarded": 0
        }
}
root@dsl-02:~# dsl_cpe_pipe.sh rtsg 1
nReturn=0 nDirection=1 nRxCorruptedTotal=146167723 nRxUncorrectedProtected=2416687 nRxRetransmitted=0 nRxCorrected=143751036 nTxRetransmitted=4147529067

root@dsl-02:~#
```

xdarklight, 2022-06-29:
@turboproc can you please submit a pull request with the fix?

turboproc, 2022-06-29:
Sure, it's my very first commit (could avoid so far), hence please double check. It's in the list of pull requests.

commit 456b9029 (2022-08-05): ltq-vdsl-app: Fix counter overflow resulting in negative values

The re-transmit counters can overflow the 32 bit representation resulting
in negative values being displayed. Background being that the numbers are
treated at some point as signed INT rather than unsigned INT.
Change the counters from 32 bit to 64 bit, should provide sufficient room
to avoid any overflow. Not the nicest solution but it works

Fixes: #10077

Acked-by: Martin Blumenstingl <martin.blumenstingl@googlemail.com>
commit bd9bb130 (2022-08-06): ltq-vdsl-app: Fix counter overflow resulting in negative values

The re-transmit counters can overflow the 32 bit representation resulting
in negative values being displayed. Background being that the numbers are
treated at some point as signed INT rather than unsigned INT.
Change the counters from 32 bit to 64 bit, should provide sufficient room
to avoid any overflow. Not the nicest solution but it works

Fixes: #10077

Acked-by: Martin Blumenstingl <martin.blumenstingl@googlemail.com>
(cherry picked from commit 456b9029d764e69f390ee26bca24883b12eb83c2)