openwrt/openwrt #10080: FriendlyARM NanoPi R1 not booting in 22.03.01-rc4

mmeelix, 2022-06-18:
FriendlyARM NanoPi R1 not booting in 22.03.01-rc4, 22.03.01-rc1 or snapshot
All is well on 21.02.3 (r16554-1d4dea6d4f)

liverun, 2022-11-19:
I think there is indeed something wrong with Openwrt 22 on this particular device. I suspect there is an issue with the network configuration.

**Tested version: OpenWrt 22.03.2, r19803-9a599fee93**

On my test setup the WAN interface seems to pick an IPv6 address but there is no IPv4 connectivity:
```
root@OpenWrt:/# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP qlen 1000
    link/ether e2:9e:6e:6a:b9:9f brd ff:ff:ff:ff:ff:ff
    inet6 2a02:2788:1004:909:e09e:6eff:fe6a:b99f/64 scope global dynamic noprefixroute 
       valid_lft 1209591sec preferred_lft 604791sec
    inet6 fe80::e09e:6eff:fe6a:b99f/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 8c:f7:10:46:a0:8c brd ff:ff:ff:ff:ff:ff
4: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel master br-lan state DOWN qlen 1000
    link/ether 5e:fe:f0:4b:1c:fe brd ff:ff:ff:ff:ff:ff
5: br-lan: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN qlen 1000
    link/ether 5e:fe:f0:4b:1c:fe brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fd2d:5058:a0ab::1/60 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::5cfe:f0ff:fe4b:1cfe/64 scope link 
       valid_lft forever preferred_lft forever
```

Contents of /etc/config/network:
```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd2d:5058:a0ab::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth1'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'wan'
        option device 'eth0'
        option proto 'dhcp'

config interface 'wan6'
        option device 'eth0'
        option proto 'dhcpv6'
```
The file looks wrong to me. Isn't the bridge section supposed to have two interfaces?

However, I have become aware of an important change between Openwrt 21.x and 22.x (not sure if it's related but mention it just in case):

> OpenWrt 22.03 and later ships with firewall4 by default, which uses nftables as a backend (instead of iptables).

Source: [Firewall configuration /etc/config/firewall](https://openwrt.org/docs/guide-user/firewall/firewall_configuration)

I have noticed no problem on other devices like the Nanopi R4S.

AiyionPrime, 2022-12-12:
@mmeelix, @pepe2k  Must have overlooked this before, as I opened #11104 in october.

@linxcow opened #11469 yesterday. Where do we want to continue the bughunt? :)

linxcow, 2022-12-12:
Yes for me it wasnt clear if the cpu frequency problem was also causing the problem i experienced with not handing out an ip address.

linxcow, 2022-12-12:
Oh thx for pointing that out , i didnt realize it.

stintel, 2022-12-22:
Sorry, used the wrong issue in the commit message. The idea is to backport bf06a7c865531cf20735cbdfac46d4947fcc18a6 to 22.03 early 2023 if there is no fallout reported.

linxcow, 2023-03-29:
Hi , I've tried the 22.03.3 from file selector for my nanopi R1 hoping it was fixed but it seems it still has the same issue.( no ip given)
[https://firmware-selector.openwrt.org/?version=22.03.3&target=sunxi%2Fcortexa7&id=friendlyarm_nanopi-r1](url)

When can we expect a fix from files downloaded from firmware-selector ?

commit bf06a7c8 (2022-12-22): uboot-sunxi: use UUID of bootdev and bootpart

Several sunxi devices come with multiple mmc devices. On such devices,
the mmc device order is unpredictable, so using /dev/mmcblk0p2 as root
device doesn't always work, which results in unbootable devices.

For the Banana Pi BPI-R3 in the mediatek target, this has been solved by
defining aliases for the mmc devices in the DTS. Ideally we would do the
same here, but for sunxi-a64 we already use UUID probing, so let's start
with that (5f2ff607e286 ("uboot-sunxi: a64: allow booting directly from
eMMC")).

Since we're building and including u-boot in [... cut]