openwrt/openwrt #10082: CPE510 still on low tx power table

mokkin, 2022-06-18:
Unfortunately the TP-Link CPE510 ist still operating on a low power table when using a stock OpenWRT due to this revert https://github.com/openwrt/openwrt/commit/c2e0c41842895ba47819fa98b785c76a2524628b
I'm not sure if there has been any investigation in the basic problem meanwhile. To use the devices with OpenWRT or other OpenWRT based distributions like in Freifunk we always have to build our own images and rerevert the patch.
Here at https://github.com/freifunk-gluon/gluon/issues/1023 was already a discussion about the topic that is still existing. If user doesn't know about this, he thinks the devices are extremly weak, but they aren't.

If I remember correctly, there was an issue while using the regular transmit power table on "upper" channels with higher transmit power. Do you remember about this @dangowrt ? If I don't mix things, you had one of these destroyed devices with this problem at this time.

dangowrt, 2022-06-19:
Yes correct. The device died within 1h after applying this patch and operating on channel above 5.5 MHz. We did a post-mortem and figured that one of the RF amplifier chips had died and in it's dead condition would short-circuit the 5V rail to ground, turning the device into a heater which would no longer be able to power it's own CPU. This could of course have been a coincident and may not even be related to this change.
IF everybody is sure that using the high power table is really fine, please test that on high channels and have the device do some transmissions. If it survives the day, lets switch to use that table.

ddreamer75, 2022-07-15:
Is it possible to change / test it on runnig firmware without patch ?