openwrt/openwrt #10084: MicroSD card not identified with device MikroTik RB760iGS (hEX S)

cmonty14, 2022-06-19:
Hello,
this is a follow-up issue of [#10044](https://github.com/openwrt/openwrt/issues/10044); the other issue was closed, but the provided solution is not working.

After installing package _kmod-sdhci_ I verified that relevant kernel modules are loaded:
```
thomas@eddie:~$ sudo opkg install kmod-sdhci
Installing kmod-sdhci (5.4.188-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.3/targets/ramips/mt7621/packages/kmod-sdhci_5.4.188-1_mipsel_24kc.ipk
Installing kmod-mmc (5.4.188-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.3/targets/ramips/mt7621/packages/kmod-mmc_5.4.188-1_mipsel_24kc.ipk
Configuring kmod-mmc.
Configuring kmod-sdhci.

thomas@eddie:~$ sudo lsmod | grep hci
mmc_core               85088  2 mmc_block,sdhci
sdhci                  33456  1 sdhci_pltfm
sdhci_pltfm             2368  0 
usb_common              2704  3 xhci_plat_hcd,xhci_hcd,usbcore
usbcore               141312  5 usb_storage,xhci_plat_hcd,xhci_pci,xhci_mtk,xhci_hcd
xhci_hcd              115744  3 xhci_plat_hcd,xhci_pci,xhci_mtk
xhci_mtk                4928  0 
xhci_pci                3744  0 
xhci_plat_hcd           5856  0 
```
However there's no SD card connected as a new drive; drive sda is another 8GB USB stick:
```
thomas@eddie:~$ sudo lsblk
Password: 
NAME      MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda         8:0    1  7.5G  0 disk 
├─sda1      8:1    1  6.5G  0 part /overlay
└─sda2      8:2    1    1G  0 part [SWAP]
mtdblock0  31:0    0  256K  1 disk 
mtdblock1  31:1    0   60K  1 disk 
mtdblock2  31:2    0    4K  1 disk 
mtdblock3  31:3    0   60K  1 disk 
mtdblock4  31:4    0    4K  1 disk 
mtdblock5  31:5    0    4K  1 disk 
mtdblock6  31:6    0 15.8M  0 disk 
mtdblock7  31:7    0  2.4M  0 disk 
mtdblock8  31:8    0 13.4M  0 disk /rom/rom
mtdblock9  31:9    0 10.9M  0 disk /rom/overlay
```

Please advise how to fix this issue.

DragonBluep, 2022-06-19:
Please install kmod-sdhci-mt7620 and try again.

cmonty14, 2022-06-19:
Great.
SD card is now identified correctly:
```
root@eddie:~# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    1  7.5G  0 disk 
├─sda1        8:1    1  6.5G  0 part /overlay
└─sda2        8:2    1    1G  0 part [SWAP]
mtdblock0    31:0    0  256K  1 disk 
mtdblock1    31:1    0   60K  1 disk 
mtdblock2    31:2    0    4K  1 disk 
mtdblock3    31:3    0   60K  1 disk 
mtdblock4    31:4    0    4K  1 disk 
mtdblock5    31:5    0    4K  1 disk 
mtdblock6    31:6    0 15.8M  0 disk 
mtdblock7    31:7    0  2.4M  0 disk 
mtdblock8    31:8    0 13.4M  0 disk /rom/rom
mtdblock9    31:9    0 10.9M  0 disk /rom/overlay
mmcblk0     179:0    0  1.9G  0 disk 
├─mmcblk0p1 179:1    0  1.8G  0 part 
└─mmcblk0p2 179:2    0  100M  0 part 
```

closed 2022-06-19 by cmonty14