openwrt/openwrt #10086: ath79: ZTE MF286RA built-in modem missing /dev/cdc-wdm0

braveheartleo, 2022-06-19:
Hi @Leo-PL,

Great work on bringing this device into the OpenWRT mainline. I have been following your progress closely, and now I have a ZTE MF286RA rebranded by an ISP. I have successfully flashed ``https://downloads.openwrt.org/snapshots/targets/ath79/nand/openwrt-ath79-nand-zte_mf286r-squashfs-sysupgrade.bin`` and its been running great on this model variant, except for the LTE modem.

There is no ``/dev/cdc-wdm0`` device upon boot. I can't figure out what is wrong, so I hope you could help me get the modem working with this device, and in the process bring complete support to this model variant if there may be subtle changes from the MF286R one.

Here is the ``dmesg``:
>[    0.000000] Linux version 5.10.120 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r19839-6153c530cc) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 Sun Jun 19 06:36:21 2022
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is ZTE MF286R
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 119996K/131072K available (6354K kernel code, 593K rwdata, 1332K rodata, 1248K init, 215K bss, 11076K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 775.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
[    0.000007] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
[    0.008281] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.074809] pid_max: default: 32768 minimum: 301
[    0.079809] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.087519] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.097643] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.108337] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.118743] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.126015] pinctrl core: initialized pinctrl subsystem
[    0.135477] NET: Registered protocol family 16
[    0.140794] thermal_sys: Registered thermal governor 'step_wise'
[    0.195075] clocksource: Switched to clocksource MIPS
[    0.207873] NET: Registered protocol family 2
[    0.212684] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.221069] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.229976] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.238071] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.245518] TCP: Hash tables configured (established 1024 bind 1024)
[    0.252348] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.259289] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.266926] NET: Registered protocol family 1
[    0.271541] PCI: CLS 0 bytes, default 32
[    0.281458] workingset: timestamp_bits=14 max_order=15 bucket_order=1
[    0.291872] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.298072] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.310090] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 252)
[    0.320419] pinctrl-single 1804002c.pinmux: 544 pins, size 68
[    0.327536] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.336318] printk: console [ttyS0] disabled
[    0.340889] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 9, base_baud = 1562500) is a 16550A
[    0.350031] printk: console [ttyS0] enabled
[... 219 lines cut ...]
E:  Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 5 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=42 Prot=01 Driver=(none)
E:  Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 6 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=06 Prot=00 Driver=cdc_ether
E:  Ad=88(I) Atr=03(Int.) MxPS=  16 Ivl=32ms
I:  If#= 7 Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_ether
I:* If#= 7 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_ether
E:  Ad=87(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=05(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
>
>T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.10
S:  Manufacturer=Linux 5.10.120 ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=1b400000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

Please let me know how else I could help. Thank you.

_Originally posted by @braveheartleo in https://github.com/openwrt/openwrt/issues/9670#issuecomment-1159693834_

closed 2022-06-19 by braveheartleo

Leo-PL, 2022-06-19:
I think you've found that yourself already, modem from MF286R does not expose such interface, it is only RNDIS + ACM at usb0 and /dev/ttyACM0 ;-)

braveheartleo, 2022-06-20:
@Leo-PL Yes, I got stuck following the guide of setting up NCM modems in Openwrt. Thank you for the support. Looks like this MF286RA model variant is not much different than MF286R.  BTW, do you know where to get modem statistics and connection information into Luci?

Leo-PL, 2022-06-20:
I use that: https://github.com/4IceG/luci-app-3ginfo, but haven't tried with 286R, I don't own one, only ones with Qualcomm modems.

braveheartleo, 2022-06-20:
@Leo-PL Thanks for the link.  Looks like I'll have to tinker with it so it works with the PXA1826.

Leo-PL, 2022-06-27:
@u-nel please see commit message: https://github.com/openwrt/openwrt/commit/7ac8da00609f42b8aba74b7efc6b0d055b7cef3e
One sidenote, that didn't make it there - busybox-static 1.16.1 (https://busybox.net/downloads/binaries/1.16.1/busybox-mips) used as telnetd works, 1.21.1 didn't. TL;DR is: use the webpage exploit to plant telnet server, overwrite "firmware" MTD with OpenWrt initramfs, reboot and proceed to proper installation.

u-nel, 2022-07-08:
> STEP 3: Booting initramfs image:

Method 1: using serial console (RECOMMENDED):
- Have TFTP server running, exposing the OpenWrt initramfs image, and
  set your computer's IP address as 192.168.0.22. This is the default
  expected by U-boot. You may wish to change that, and alter later
  commands accordingly.
- Connect the serial console if you haven't done so already,
- Interrupt boot sequence by pressing any key in U-boot when prompted
- Use the following commands to boot OpenWrt initramfs through TFTP:

  setenv serverip 192.168.0.22
  setenv ipaddr 192.168.0.1
  tftpboot 0x81000000 openwrt-ath79-nand-zte_mf286a-initramfs-kernel.bin
  bootm 0x81000000

  (Replace server IP and router IP as needed). There is no  emergency
  TFTP boot sequence triggered by buttons, contrary to MF283+.
- When OpenWrt initramfs finishes booting, proceed to actual
  installation.

Quick inquiry, Sorry, I'm new to this kind of stuff
What "serial console" is this USB or UTP?

Leo-PL, 2022-07-08:
None of that, there is an unpopulated 4-pin 3.3V UART connector on the board. Pinout is at the very beginning of instructions.

cunyiles, 2022-09-11:
> 

Hi, I have the following problem when overwriting the firmware device. How can I solve this?

```
admin@(none):~# nandwrite -p /dev/mtd16 /var/usb_disk/openwrt-22.03.0-ath79-nand-zte_mf286r-initramfs-kernel.bin
Writing data to block 0 at offset 0x0
libmtd: error!: cannot write 2048 bytes to mtd16 (eraseblock 0, offset 0)
        error 1 (Operation not permitted)
nandwrite: error!: /dev/mtd16: MTD write failure
           error 1 (Operation not permitted)
Data was only partially written due to error
: Operation not permitted
```

Leo-PL, 2022-09-11:
@cunyiles Some stock firmwares lock the erase blocks on the NAND for some reason, or flag the whole MTD partition as read-only. This was not known to me at the time the support was added.
I recommend using the "serial console+ initramfs" method or the "tftp factory recovery" method to install.