openwrt/openwrt #10098: 802.11 Mesh fails after upgrade in 5 GHz band

turboproc, 2022-06-20:
Upgraded yesterday to OpenWrt SNAPSHOT r19859-36e46c3c13 / LuCI Master git-22.167.28356-8effea5 to learn that the Mesh wouldn't come up any more. Checking some other places it was suggested to be related to DFS behaviour. Originally I had the mesh running on channel 100 in NL region. First change was replacing the region to US which made it work again but not really sustainable. Changing channel to 36 and region back to NL made it also work.

Wouldn't have the experience to quickly pinpoint where to fix this.

mundschenk-at, 2023-01-06:
@turboproc Did you find out anything about this?

mundschenk-at, 2023-02-28:
I tried to get someone more experienced to look at this issue via https://forum.openwrt.org/t/802-11s-stopped-working-with-dfs-channels-in-etsi-domain-between-21-02-2-and-21-02-3-what-changed/147582, but to no avail unfortunately.