openwrt/openwrt #10107: Enable 5/10 MHz on ath9k_htc

IakovlevAA, 2022-06-22:
I need to enable 5/10 MHz on my ar9721(ath9k_htc) on raspberry pi 4, but patches included in package/kernel/mac80211/patches/ath9k (i am interested in 512-ath9k_channelbw_debugfs.patch), seems to not working. I've looked in code, it has to create file 'chanbw' in /sys/kernel/debug/.../ath9k_htc. Changing /etc/config/wireless is useless as well. Before make, I put patch in /target/linux/bcm27xx/patches-5.15 and there are changes in build_dir code, but no changes in Openwrt system.

neheb, 2022-06-24:
ath9k is not ath9k_htc. different patch is needed.

IakovlevAA, 2022-06-26:
I did modify this patch for htc(htc_drv_debug.c and htc_drv_init.c), but there are no changes at all.

closed 2023-04-07 by IakovlevAA