openwrt/openwrt #10110: Belkin RT3200 - v22.03.0-rc4 - wifi failure

hpo13, 2022-06-23:
After upgrading my Belkin RT3200 to v22.03-rc4 wifi (2.4 and 5 GHz) does not work any more. 

On initialization the following messages appear in the logfile  (for both wifi devices):
```
ue Jun 21 16:40:41 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Tue Jun 21 16:40:41 2022 daemon.err hostapd: Line 42: unknown configuration item 'eap_server'
Tue Jun 21 16:40:41 2022 daemon.err hostapd: Line 43: unknown configuration item 'wps_state'
Tue Jun 21 16:40:41 2022 daemon.err hostapd: Line 44: unknown configuration item 'device_type'
Tue Jun 21 16:40:41 2022 daemon.err hostapd: Line 45: unknown configuration item 'device_name'
Tue Jun 21 16:40:41 2022 daemon.err hostapd: Line 46: unknown configuration item 'manufacturer'
Tue Jun 21 16:40:41 2022 daemon.err hostapd: Line 47: unknown configuration item 'config_methods'
Tue Jun 21 16:40:41 2022 daemon.err hostapd: Line 48: unknown configuration item 'wps_independent'
Tue Jun 21 16:40:41 2022 daemon.err hostapd: 7 errors found in configuration file '/var/run/hostapd-phy0.conf'
Tue Jun 21 16:40:41 2022 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy0.conf
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992): Command failed: ubus call hostapd config_add {"iface":"wlan0", "config":"/var/run/hostapd-phy0.conf"} (Invalid argument)
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992): Usage: ubus [<options>] <command> [arguments...]
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992): Options:
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):  -s <socket>:             Set the unix domain socket to connect to
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):  -t <timeout>:            Set the timeout (in seconds) for a command to complete
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):  -S:                      Use simplified output (for scripts)
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):  -v:                      More verbose output
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):  -m <type>:               (for monitor): include a specific message type
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):                   (can be used more than once)
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):  -M <r|t>         (for monitor): only capture received or transmitted traffic
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992): Commands:
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):  - list [<path>]                  List objects
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):  - call <path> <method> [<message>]       Call an object method
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):  - subscribe <path> [<path>...]   Subscribe to object(s) notifications
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):  - listen [<path>...]                     Listen for events
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):  - send <type> [<message>]                Send an event
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):  - wait_for <object> [<object>...]        Wait for multiple objects to appear on ubus
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):  - monitor                                Monitor ubus traffic
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992):
Tue Jun 21 16:40:41 2022 daemon.notice netifd: radio0 (1992): Device setup failed: HOSTAPD_START_FAILED
Tue Jun 21 16:40:41 2022 daemon.notice netifd: Wireless device 'radio0' set retry=0
```
The error persists after upgrading to a recent snapshot. Some more info about my current setup:
```
Model | Linksys E8450 (UBI)
Architecture | ARMv8 Processor rev 4
Target Platform | mediatek/mt7622
Firmware Version | OpenWrt SNAPSHOT r19870-5f10805094 / LuCI Master git-22.167.28356-8effea5
Kernel Version | 5.15.45
```
Upgrade was done from a March snapshot version (upgrade, not a new installation). With the old version wifi worked without problems.  

Please let me know if more information is needed or if there are workarounds I may try.

Regards, Hein

KA2107, 2022-06-24:
I am not a dev, but I think the devs would find it useful if you post your /etc/config/wireless and maybe also /etc/config/network.

hpo13, 2022-06-24:
Here are the relevant config files /etc/config/wireless, /etc/config/network, /var/run/hostapd-phy0.conf

/etc/config/wireless

```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/18000000.wmac'
        option band '2g'
        option htmode 'HT20'
        option country 'DE'
        option cell_density '0'
        option channel '8'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid '***'
        option encryption 'sae-mixed'
        option key '***'
        option wps_pushbutton '1'
        option ieee80211r '1'
        option mobility_domain 'f1ac'
        option ft_over_ds '1'
        option ft_psk_generate_local '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option band '5g'
        option country 'DE'
        option cell_density '0'
        option channel '124'
        option htmode 'HE160'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid '***'
        option encryption 'sae-mixed'
        option key '***'
        option wps_pushbutton '1'
        option ieee80211r '1'
        option mobility_domain 'f1ac'
        option ft_over_ds '1'
        option ft_psk_generate_local '1'

```
/etc/config/network
```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix '***'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.0.1'
        option netmask '255.255.255.0'
        option ip6hint '5'
        option ip6assign '64'

config interface 'wan'
        option proto 'pppoe'
        option device 'wan.40'
        option username '***'
        option password '***'
        option ipv6 'auto'

config interface 'wan6'
        option proto 'dhcpv6'
        option device 'wan.40'
        option reqaddress 'try'
        option reqprefix 'auto'

config device
        option type '8021q'
        option ifname 'wan'
        option vid '40'
        option name 'wan.40'

config interface 'DMZ'
        option proto 'static'
        option device 'lan4'
        option ipaddr '192.168.13.14'
        option netmask '255.255.255.0'
        option gateway '192.168.0.1'
        option ip6assign '64'
        option ip6hint '6'

config interface 'WIREGUARD'
        option proto 'wireguard'
        option private_key '***'
        option listen_port '1234'
        list addresses '10.14.0.0/16'

config wireguard_WIREGUARD
        option description '***'
        option public_key '***'
        list allowed_ips '10.14.0.3/32'
        option route_allowed_ips '1'
        option persistent_keepalive '25'
```
plus /var/run/hostaptd-phy0.conf
```
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=DE
ieee80211d=1
hw_mode=g
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 120 240
beacon_int=100
dtim_period=2
channel=8
chanlist=8

ieee80211n=1
ht_coex=0
ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1]

radio_config_id=***
interface=wlan0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
sae_require_mfp=1
sae_pwe=2
wpa_passphrase=***
wpa_psk_file=/var/run/hostapd-wlan0.psk
auth_algs=1
wpa=2
wpa_pairwise=CCMP
eap_server=1
wps_state=2
device_type=6-0050F204-1
device_name=OpenWrt AP
manufacturer=www.openwrt.org
config_methods=push_button
wps_independent=1
ssid=***
bridge=br-lan
wds_bridge=
snoop_iface=br-lan
ft_iface=br-lan
mobility_domain=f1ac
ft_psk_generate_local=1
ft_over_ds=1
reassociation_deadline=1000
nas_identifier=***
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK FT-PSK WPA-PSK-SHA256 SAE FT-SAE
okc=1
ieee80211w=1
group_mgmt_cipher=AES-128-CMAC
dynamic_vlan=0
vlan_naming=1
vlan_no_bridge=1
vlan_file=/var/run/hostapd-wlan0.vlan
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
config_id=***
bssid=***
```

neilsan1366, 2022-06-25:
seems resolve now with last commit

hpo13, 2022-06-26:
I just installed the current snapshot. Unfortunately, the error persists.