openwrt/openwrt #10111: wndap660 switch port to physical label and indicator LED mapping incorrect

tim-seoss, 2022-06-23:
WNDAP 660  OpenWrt 22.03.0-rc4, r19426-2b1941e47d

Viewing the 'Network' -> 'Switch' config page in LuCI:

- The LuCI "LAN 1" port corresponds to the port physically labelled "🖧2" at the rear of the device.
- The LuCI "LAN 2" port corresponds to the port physically labelled "🖧1" at the rear of the device.

When a patch cord is attached to the port labelled "🖧1" on the device, the LED labelled "🖧2" illuminates.
When a patch cord is attached to the port labelled "🖧2" on the device, no changes are seen on any LEDs.

I've only checked the LED behaviour with a with a 1000baseT device attached.

neheb, 2022-06-24:
Was this an issue with ar71xx (19.07)?

tim-seoss, 2022-06-24:
> Was this an issue with ar71xx (19.07)?

@neheb Although this access point has ath9k radios, the SoC is PowerPC 460 (APM82181), rather than MIPS, and it's always used Device Tree.

I haven't checked to see if this is a regression vs earlier released tho'.  I'll try and check that later today.

tim-seoss, 2022-06-24:
The LED behaviour is correct in 21.02.3 and also in 19.07.9.  It only seems to be broken in 22.03.0-rc4.

On all three releases, the LuCI `LAN 1` port is mapped to `🖧2` and vice versa.

I suppose it could be argued that fixing this could be a breaking change for some people, however in the default config both ports are configured as LAN ports, and also an upgrade from an earlier release to 22.03 requires writing the factory image (not sysupgrade due to partition sizing requirements), so I think pre 22.03 would be a good time to make this fix.  Any thoughts?

tim-seoss, 2022-06-24:
Hi @chunkeey firstly, thanks for your work porting OpenWRT to the WNDAP660! Any thoughts on the above?

tim-seoss, 2022-06-24:
The current downloadable snapshot `netgear_wndap660-initramfs-kernel.bin	10bba3a0bee879a378f73e2a1a90a07f91f5a852bc6c4983b71599a7b97295d3	7225.7 KB	Fri Jun 24 11:34:11 2022` shows the same behavour as 22.03.0-rc4.

closed 2022-06-25 by jow-

tim-seoss, 2022-06-25:
Confirmed fixed, thanks!

commit 36483600 (2022-06-25): apm821xx: WNDAP660: fix ethernet port ordering

Tim Small reported:
| Viewing the 'Network' -> 'Switch' config page in LuCI:
|
| The LuCI LAN 1 port corresponds to the port physically
|     labelled 2 at the rear of the device.
| [...]
|
| When a patch cord is attached to the port labelled 1 [...],
| the LED labelled 2 illuminates. [...]

=> Ports, LuCI and LEDs are reversed/don't match.

Reported-by: Tim Small
Fixes: #10111
commit a4390ea2 (2022-06-25): apm821xx: WNDAP660: fix ethernet port ordering

Tim Small reported:
| Viewing the 'Network' -> 'Switch' config page in LuCI:
|
| The LuCI LAN 1 port corresponds to the port physically
|     labelled 2 at the rear of the device.
| [...]
|
| When a patch cord is attached to the port labelled 1 [...],
| the LED labelled °2 illuminates.

=> essentially, the Ports and LEDs are reversed.

Reported-by: Tim Small
Fixes: #10111

(cherry picked from commit 36483600d337ab3ec8a6dabeaa160614d07f5588)