openwrt/openwrt #10115: YunCore AX820 stuck in recovery mode

matlab22, 2022-06-24:
I tried to upgrade from AX820-AP-V2.0-Build20220423095645-EN to openwrt-22.03.0-rc4-ramips-mt7621-yuncore_ax820-initramfs-kernel.bin. Therefore I renamed the bin to ubin and flashed it via web interface.
Now I am stuck in the MediaTek U-Boot System Recovery Mode (LAN-Port @ 192.168.1.1). I tried to reflash the bin but it loops to Upgrade failed!

@oe1rfc how did you manged to flash the initial firmware?

Vald3m, 2022-06-24:
I got similar issue on SNAPSHOT with YunCore FAP690 what I'm adapting to OpenWRT now. I still didn't find what changes lead to this but it can be fixed if you change next section in `target/linux/ramips/image/mt7621.mk`
```
define Device/yuncore_ax820
  $(Device/dsa-migration)
  $(Device/uimage-lzma-loader)
  IMAGE_SIZE := 15808k
  DEVICE_VENDOR := YunCore
  DEVICE_MODEL := AX820
  DEVICE_PACKAGES := kmod-mt7915e
endef
TARGET_DEVICES += yuncore_ax820
```

I found the point how to solve this in the next commit `fd72e595c2b2a46bab8cbc7e9415fbfeae7b5b0d`

` $(Device/uimage-lzma-loader)` what is  need to be added.
Some changes happened in the tree last week, because I was able to build bootable firmware on one week old SNAPSHOT. However it stopped to build after rebasing my code to latest commits.

matlab22, 2022-06-24:
via recovery mode I can not upload any image. It always fails.
I hooked up a serial console and flashed via TFTP

openwrt-22.03.0-rc4-ramips-mt7621-yuncore_ax820-initramfs-kernel.bin is getting this error
`
SF: Detected w25q128bv with page size 256 Bytes, erase size 64 KiB, total 16 MiB
Reading from flash 0x90000 to mem 0x80010000, size 0x572445 ...
## Booting kernel from Legacy Image at 80010000 ...
   Image Name:   MIPS OpenWrt Linux-5.10.120
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    5710853 Bytes = 5.4 MiB
   Load Address: 80001000
   Entry Point:  80001000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... lzma compressed: uncompress error 1
Must RESET board to recover

Web failsafe UI started

`

and openwrt-22.03.0-rc4-ramips-mt7621-yuncore_ax820-initramfs-kernel.bin is running in a boot loop

matlab22, 2022-06-25:
in the uboot benü there is also this option to upgrade the bootloader:
```
  *** U-Boot Boot Menu ***

     1. Startup system (Default)
     2. Upgrade firmware
     3. Upgrade bootloader
     4. Upgrade bootloader (advanced mode)
     5. Load image
     0. U-Boot console

  Press UP/DOWN to move, ENTER to select
```
For now I tried only option 2, should I try one of the others?

Vald3m, 2022-06-25:
I had the same error like you post earlier:
`SF: Detected w25q128bv with page size 256 Bytes, erase size 64 KiB, total 16 MiB
Reading from flash 0x90000 to mem 0x80010000, size 0x572445 ...
## Booting kernel from Legacy Image at 80010000 ...
   Image Name:   MIPS OpenWrt Linux-5.10.120
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    5710853 Bytes = 5.4 MiB
   Load Address: 80001000
   Entry Point:  80001000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... lzma compressed: uncompress error 1
Must RESET board to recover

Web failsafe UI started`

and I was able to solve it like I explained before.
I also tried to use:
` 5. Load image`
with `***-initramfs-kernel.bin` file, however error was the same.

oe1rfc, 2022-06-25:
```
Uncompressing Kernel Image ... lzma compressed: uncompress error 1
```
This indeed looks as if `$(Device/uimage-lzma-loader)` is needed in the device definition.
I prepared a [patch here](https://github.com/oe1rfc/openwrt/commit/503a9f0dda433d42b28ca81194016cd03575b7a1) but I haven't got an AX820 at hand currently to verify this before sending a patch/PR, but I'll see if I can get remote-hands on one in the next week.

matlab22, 2022-06-27:
I tried to set up a virtual machine to build, but ran in several time consuming problems. :(
If you have an image I can test it for you on the hardware.

Vald3m, 2022-06-27:
@matlab22 I create a build for this device with a patch from @oe1rfc. You can download this binary by the next [link](https://mega.nz/file/AMpXFBjC#uggXHzvOuJNFvTmP5Ptv4i9-PsAdVpDmVTg32r_l02E). Build is done on the latest SNAPSHOT commit at current moment (`9d06e5a7735508d793c3a125f9f7d623671e29e4`).

matlab22, 2022-06-27:
@Vald3m Thank you, I tested it (TFTP & Serial)
It is still the same behavior:
```
Please press Enter to activate this console.
[    8.156308] random: crng init done
[    8.159835] random: 19 urandom warning(s) missed due to ratelimiting
[    8.294276] kmodloader: loading kernel modules from /etc/modules.d/*
[    8.483081] Loading modules backported from Linux version v5.15.33-0-g06f50ca83ace
[    8.490678] Backport generated by backports.git v5.15.33-1-0-g183c4ab2
[    8.538892] urngd: v1.0.2 started.
[    8.642668] mt7621-pci 1e140000.pcie: bus=1 slot=0 irq=22
[    8.648093] pci 0000:00:00.0: enabling device (0004 -> 0007)
[    8.653758] mt7915e_hif 0000:01:00.0: enabling device (0000 -> 0002)
[    8.660575] mt7621-pci 1e140000.pcie: bus=2 slot=1 irq=23
[    8.666035] pci 0000:00:01.0: enabling device (0004 -> 0007)
[    8.671725] mt7915e 0000:02:00.0: enabling device (0000 -> 0002)
[    8.915086] mt7915e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20211222184017a
[    8.915086]
[    9.212310] mt7915e 0000:02:00.0: WM Firmware Version: ____000000, Build Time: 20211222184052
[    9.248790] mt7915e 0000:02:00.0: WA Firmware Version: DEV_000000, Build Time: 20211222184111
[   29.978996] mt7915e 0000:02:00.0: Message 00003aed (seq 13) timeout
[   50.458993] mt7915e 0000:02:00.0: Message 000021ed (seq 4) timeout
[   70.938983] mt7915e 0000:02:00.0: Message 00003aed (seq 5) timeout
[   91.418972] mt7915e 0000:02:00.0: Message 00001eed (seq 6) timeout
[  111.898980] mt7915e 0000:02:00.0: Message 00003aed (seq 7) timeout
[  111.968670] PPP generic driver version 2.4.2
[  111.974945] NET: Registered protocol family 24
[  111.987767] kmodloader: done loading kernel modules from /etc/modules.d/*
[  130.529596] jffs2_scan_eraseblock(): End of filesystem marker found at 0x0
[  130.536554] jffs2_build_filesystem(): unlocking the mtd device...
[  130.536565] done.
[  130.544825] jffs2_build_filesystem(): erasing all blocks after the end marker...
[  131.019934] mtk_soc_eth 1e100000.ethernet eth0: Link is Down
[  131.048792] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
[  131.057688] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  131.066921] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  131.077403] mt7530 mdio-bus:1f lan: configuring for phy/gmii link mode
[  131.085188] 8021q: adding VLAN 0 to HW filter on device lan
[  131.095286] br-lan: port 1(lan) entered blocking state
[  131.100710] br-lan: port 1(lan) entered disabled state
[  131.106704] device lan entered promiscuous mode
[  131.111506] device eth0 entered promiscuous mode
[  131.158342] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[  131.165465] 8021q: adding VLAN 0 to HW filter on device wan
[  133.686023] mt7530 mdio-bus:1f lan: Link is Up - 1Gbps/Full - flow control off
[  133.693323] br-lan: port 1(lan) entered blocking state
[  133.698441] br-lan: port 1(lan) entered forwarding state
[  133.704691] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[  134.696913] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[  134.704250] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
[  163.945165] done.
[  163.947134] jffs2: notice: (2072) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unc                             hecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[  164.103745] overlayfs: upper fs does not support tmpfile.

```

Vald3m, 2022-06-27:
But you log shows that device is booted.

matlab22, 2022-06-27:
yes but this was also the case, when I flash the previous one via TFTP. The bootloader problem happened after upgrading from webfrontend

Vald3m, 2022-06-27:
I stop understanding where is the issue now. 
Your initial post described incorrect usage of an image for initial flashing (`*-initramfs-kernel.bin` instead of `*-squashfs-sysupgrade.bin.`), that was pointed to you by @oe1rfc. 
Later you mention error with bootloader which I have with my device too (`Uncompressing Kernel Image ... lzma compressed: uncompress error 1`). 
However now your boot log looks correct and the device should be in working state. 
Where is the issue for you now?

matlab22, 2022-06-27:
I dont understand the problem that I am having. I can enter the TFTP bootloader and upload any image. But they all get stuck at some point doing ether a reboot loop or wait for ever.

matlab22, 2022-06-27:
I tried it once again: 
PoE on the WAN Port
LAN directly connected to my PC

I flashed the bin from [Vald3m](https://github.com/openwrt/openwrt/issues/10115#issuecomment-1166921629) I got this strange link up and down log
```
[  234.601872] mt7530 mdio-bus:1f wan: Link is Down
[  240.393906] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[  241.147998] mt7530 mdio-bus:1f wan: Link is Down
[  245.771931] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[  246.525994] mt7530 mdio-bus:1f wan: Link is Down
[  254.750302] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[  255.504536] mt7530 mdio-bus:1f wan: Link is Down
[  261.382178] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[  262.135822] mt7530 mdio-bus:1f wan: Link is Down
[  266.724311] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
```
I suspected a power problem and plugged in a 12V dc supply and accidentally hit the reset switch. The router restarted and now I am able to access the webinterface 👍 
Full boot-log attached
[bootlog.txt](https://github.com/openwrt/openwrt/files/8990598/bootlog.txt)

matlab22, 2022-06-27:
I updated the access point via openwrt webinterface again. I used the same image as before. Therefore I confirm that the patch is working. Here the full bootlog:
[update_log.txt](https://github.com/openwrt/openwrt/files/8991007/update_log.txt)

oe1rfc, 2022-06-27:
@matlab22 @Vald3m thanks for verifying, I'll send in the patch later.

matlab22, 2022-06-28:
I just checked again with "openwrt-22.03.0-rc4-ramips-mt7621-yuncore_ax820-squashfs-sysupgrade.bin" and it works when the router is properly powered. So no change needed. Thanks for your support

closed 2022-06-28 by matlab22