openwrt/openwrt #10131: YunCore AX820: GPIO LED not correct

matlab22, 2022-06-27:
In the "openwrt-22.03.0-rc4-ramips-mt7621-yuncore_ax820" firmware when I turn on the green led in the gui. The LED is red with some sort of light blue touch. The red and the blue LEDs are not available:

Green LED off:
```
root@OpenWrt:/# cat /sys/kernel/debug/gpio
gpiochip2: GPIOs 416-447, parent: platform/1e000600.gpio, 1e000600.gpio-bank2:
gpiochip1: GPIOs 448-479, parent: platform/1e000600.gpio, 1e000600.gpio-bank1:
gpiochip0: GPIOs 480-511, parent: platform/1e000600.gpio, 1e000600.gpio-bank0:
 gpio-480 (                    |watchdog            ) out hi ACTIVE LOW
 gpio-495 (                    |green:status        ) out hi ACTIVE LOW
 gpio-498 (                    |reset               ) in  hi IRQ ACTIVE LOW
 gpio-499 (                    |reset               ) out hi ACTIVE LOW
```
Green LED on:
root@OpenWrt:/# cat /sys/kernel/debug/gpio
gpiochip2: GPIOs 416-447, parent: platform/1e000600.gpio, 1e000600.gpio-bank2:
gpiochip1: GPIOs 448-479, parent: platform/1e000600.gpio, 1e000600.gpio-bank1:
gpiochip0: GPIOs 480-511, parent: platform/1e000600.gpio, 1e000600.gpio-bank0:
 gpio-480 (                    |watchdog            ) out lo ACTIVE LOW
 gpio-495 (                    |green:status        ) out lo ACTIVE LOW
 gpio-498 (                    |reset               ) in  hi IRQ ACTIVE LOW
 gpio-499 (                    |reset               ) out hi ACTIVE LOW

gpio base:
```
root@OpenWrt:/# cat /sys/class/gpio/gpiochip*/base | head -n1
416
```

On the vendors firmware I can observe the following:
```
gpio-14	out hi	
gpio-15	out hi	all off
gpio-16	out hi	
		
gpio-14	out lo	green
gpio-15	out hi	
gpio-16	out hi	
		
gpio-14	out hi	
gpio-15	out hi	
gpio-16	out lo	blue
		
gpio-14	out hi	
gpio-15	out lo	red
gpio-16	out hi	
```
gpio base:
```
root@OpenWrt:/# cat /sys/class/gpio/gpiochip*/base | head -n1
0
```

Perhaps I do not understand the GPIO base correctly, but the green LED should be on 432 ACTIVE LOW, but is on 495 and somehow alters the LED color anyway.

matlab22, 2022-06-27:
I just checked the following:

'''
echo "496" > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio496/direction
echo "1" > /sys/class/gpio/gpio496/value #blue off
echo "0" > /sys/class/gpio/gpio496/value #blue on

echo "494" > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio494/direction
echo "1" > /sys/class/gpio/gpio494/value #green off
echo "0" > /sys/class/gpio/gpio494/value #green on
'''

So I guess it should be like this:
'''
gpio-494 (                    |green:status     ) out lo ACTIVE LOW
gpio-495 (                    |red:status         ) out lo ACTIVE LOW
gpio-496 (                    |blue:status        ) out lo ACTIVE LOW
'''

f00b4r0, 2022-06-29:
I'm afraid this must be a different hardware revision. AFAICT my AX820 has only the green led and testing the other GPIOs as you mentioned does not light up any other color. If the different revisions are not easily distinguishable this is going to be a headache.

matlab22, 2022-06-29:
It looks like they are not a good example in therms of marking revisions. My device has 2 LEDs close to the ports indicating the Ethernet status and one RGB led linked with a lite pipe trough the top cover.
I uploaded my pictures to the [ToH AX820](https://openwrt.org/toh/yuncore/ax820#yuncore_ax820). Where and when did you get your sample?

f00b4r0, 2022-06-29:
I checked your pictures and the board looks visually identical to mine. Same revision (V1.2) and same looking LED, except mine only lights up green at the currently registered GPIO, as did that of @oe1rfc I suspect, since he submitted that device.
For where/when: I don't know: this device was sent to me by a 3rd party, I didn't buy it.

It wouldn't be so bad if the green color was on the same GPIO for your and our devices, but if that's not the case, *sigh*...

matlab22, 2022-06-29:
Is it posible that on your PCB D44 is mounted instead of LED1

oe1rfc, 2022-06-29:
That is true, but I don't think that different LED colors justifies adding additional hardware variants, especially if future changes are very likely. I do think that this (led colors) is primarily a cosmetic issue when it comes to board support, although the additional LED GPIOs should be supported, even if they are not populated.

f00b4r0, 2022-06-29:
> That is true, but I don't think that different LED colors justifies adding additional hardware variants, especially if future changes are very likely. I do think that this (led colors) is primarily [...]

Well technically the additional LED gpios are "supported", it's just a matter of exporting said GPIO.

IMHO there there are several problems with defining the extra GPIOs in DTS:
- if we do this assuming they are unconnected, but it eventually turns out that on some board revision they are connected to something else than the LED, we're in trouble (I'm generally uncomfortable defining output GPIOs that we don't know for sure they are always harmless: notice how these are active low GPIOs, so they will be high-stated if we define them).
- second, since the extra GPIOs do nothing on some boards (such as the ones you and I have) there will inevitably be people opening bug reports claiming the extra LEDs don't work;
- third, it will only take someone unhappy that the red led is the status led on their board to submit a patch to change that and disable the status led for the boards that have only the green wired.
(of course if at some point a third revision surfaces that has only one color and it is not on GPIO 15 (what is the green led for us), then we are screwed 😛)

Now what strikes me is: there has to be a way to tell the boards apart, otherwise how does vendor firmware do it?

oe1rfc, 2022-06-29:
I agree with you, but I do not think that there is a possibility that anything else than a LED will be populated on these boards, at least as long as it's the same board revision.

> Now what strikes me is: there has to be a way to tell the boards apart, otherwise how does vendor firmware do it?

In my (limited) experience with other YunCore firmwares, it doesn't. There are even some devices that run the same firmware on different (but electrically similar) boards where some SoC gpio pins are not connected to anything but still configured/used in the firmware.

f00b4r0, 2022-06-29:
> In my (limited) experience with other YunCore firmwares, it doesn't. There are even some devices that run the same firmware on different (but electrically similar) boards where some SoC gpio pins [...]

OK well if that's the case then we probably could define the extra LEDs, while keeping the current one as status LED for all devices. We'd also have to add a note in the DTS so that nobody changes that down the road. Should I prep a patch?

matlab22, 2022-06-29:
Here comes the crowning glory, I got the pin mapping from yuncore and it is different than on my board.
[AX820 GPIO列表-中性.xlsx](https://github.com/openwrt/openwrt/files/9014495/AX820.GPIO.-.xlsx)

f00b4r0, 2022-06-29:
Well at least this confirms that GPIO 15 is expected to be the system LED. For the other two, one thing I didn't try is to make sure they aren't pinmuxed to some other function by e.g. the bootloader, preventing their use as GPIO. I'll investigate later.

oe1rfc, 2022-06-29:
very nice, thanks for the datasheet, I've heard from the NAND option but didn't find any reference in the stock-lede.

I think it would be best to name the LEDs according to the datasheet `dome:2g`, `dome:sys` and `dome:5g` for example, what do you think? I'm on vacation until next week, @f00b4r0 if you would like to prepare/send a patch feel free to do so.

f00b4r0, 2022-06-29:
Sounds good, will do :)

oe1rfc, 2022-06-30:
looks good, thanks.

commit a0e1d3ab (2022-07-01): ramips: improve YunCore AX820 LEDs

At least two AX820 hardware variants are known to exist, but they cannot
be distinguished (same hardware revision, no specific markings).

They appear to have the same LED hardware, but wired differently:

- One has a red system LED at GPIO 15, a green wlan2g LED at GPIO 14 and
  a blue wlan5g LED at GPIO 16;
- The other only offers a green system LED at GPIO 15, with GPIO 14 and
  16 being apparently not connected

Finally, a Yuncore datasheet says the canonical wiring should be:
- Blue wlan2g GPIO 14, green system GPIO 15, red wlan5g GPIO 16

All GPIOs are [... cut]
commit 158a5af8 (2022-07-01): ramips: improve YunCore AX820 LEDs

At least two AX820 hardware variants are known to exist, but they cannot
be distinguished (same hardware revision, no specific markings).

They appear to have the same LED hardware, but wired differently:

- One has a red system LED at GPIO 15, a green wlan2g LED at GPIO 14 and
  a blue wlan5g LED at GPIO 16;
- The other only offers a green system LED at GPIO 15, with GPIO 14 and
  16 being apparently not connected

Finally, a Yuncore datasheet says the canonical wiring should be:
- Blue wlan2g GPIO 14, green system GPIO 15, red wlan5g GPIO 16

All GPIOs are [... cut]
closed 2022-07-01 by jow-