openwrt/openwrt #10147: Startup sequence need a look.

momothefox, 2022-06-29:
I am using openwrt 22.03-rc4 
and i am building my own custom image. 
I am using chilli on openwrt and i use external radius system for authentication and accounting. 

The thing is i noticed that users from some of my APs showing that start session is wrong dated.  
so i looked up accounting packets. and i could see clearly that chilli is reporting accounting packets with wrong date/time. 
while the AP date/time is correct. 

so i checked out startup sequence in openwrt found out that chilli start priority is 90 
while sysntpd start priority is 98 

so i think sysntpd start priority needs to be before chilli. 

below sample accounting packet dated Sat Jun 25 10:26:02 2022
while chilli reporting Event-Timestamp = "May 17 2022 23:02:13 EEST"

```
Sat Jun 25 10:26:02 2022
	ChilliSpot-Version = "1.6"
	Attr-26.14559.10 = 0x00000001
	Event-Timestamp = "May 17 2022 23:02:13 EEST"
	User-Name = "63578476"
	Acct-Status-Type = Start
	Acct-Session-Id = "165614193900000004"
	Class = 0x61693a3761363432373831363038363930333836633838656131616539663638386261
	Framed-IP-Address = 192.168.182.3
	NAS-Port-Type = Wireless-802.11
	NAS-Port = 4
	NAS-Port-Id = "00000004"
	Calling-Station-Id = "XX-XX-XX-XX-XX-XX"
	Called-Station-Id = "XX-XX-XX-XX-XX-XX"
	NAS-IP-Address = 192.168.182.1
	NAS-Identifier = "XX-XX-XX-XX-XX-XX"
	WISPr-Location-ID = "<XXXXX>"
	WISPr-Location-Name = "<XXXXXX>"
	FreeRADIUS-Acct-Session-Start-Time = "Jun 25 2022 10:26:02 EEST"
	Tmp-String-9 = "ai:"
	Acct-Unique-Session-Id = "789071c12892308bfc34c2e4e4f265a3"
	Timestamp = 1656141962
```

momothefox, 2022-07-03:
any Idea ?

jow-, 2022-07-13:
Time synchronization happens gradually and asynchronously so swapping the startup order will not help. You could consider disabling chilli start on boot and instead (re)start it from a [stratum NTP hotplug event handler](https://openwrt.org/docs/guide-user/base-system/hotplug#ntp). 

Furthermore, coova-chilli is maintained in https://github.com/openwrt/packages/tree/master/net/coova-chilli so consider opening an issue report there if you think it should be changed in the default packaging.

closed 2022-07-13 by jow-