openwrt/openwrt #10154: [22.03] belkin-rt-3200: mesh wifi does not work

PolynomialDivision, 2022-06-30:
The mesh wifi does not work anymore with 802.11s (`OpenWrt 22.03-SNAPSHOT, r19485-e4caacc887`). Dmesg shows:
```
[  473.582673] ------------[ cut here ]------------
[  473.587318] WARNING: CPU: 0 PID: 5525 at 0xffffffc0089fa570 [mac80211@00000000351eb123+0x79000]
[  473.596005] Modules linked in: nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_bridge nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT chacha_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY poly1305_neon nfnetlink nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic libcrc32c libchacha iptable_mangle iptable_filter ipt_ECN ip_tables hwmon compat ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT
[  473.596157]  x_tables nf_reject_ipv6 ip6_udp_tunnel udp_tunnel seqiv leds_gpio xhci_plat_hcd gpio_button_hotplug
[  473.693235] CPU: 0 PID: 5525 Comm: uhttpd Tainted: G S      W         5.10.125 #0
[  473.700706] Hardware name: Linksys E8450 (UBI) (DT)
[  473.705576] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO BTYPE=--)
[  473.711578] pc : 0xffffffc0089fa570 [mac80211@00000000351eb123+0x79000]
[  473.718183] lr : 0xffffffc0089fa4dc [mac80211@00000000351eb123+0x79000]
[  473.724786] sp : ffffffc010dfb660
[  473.728091] x29: ffffffc010dfb660 x28: ffffff8000192d00 
[  473.733396] x27: ffffffc010dfbdf0 x26: ffffff8003c1c2e0 
[  473.738699] x25: ffffff80022410a0 x24: 0000000000000011 
[  473.744004] x23: ffffff800264c8c0 x22: 0000000000000000 
[  473.749307] x21: ffffff80023d0d80 x20: 0000000000000000 
[  473.754611] x19: ffffff8003c1c000 x18: ffffff80022cfe60 
[  473.759915] x17: 00000000ffffffff x16: 0000000000000000 
[  473.765219] x15: 0000000042384800 x14: 0000000000003cc0 
[  473.770523] x13: 0000000000000225 x12: ffffffffffffc4d0 
[  473.775826] x11: 0000000000000002 x10: 0000000000000850 
[  473.781130] x9 : 0000000000000000 x8 : ffffff80023d1000 
[  473.786435] x7 : 0000000000000000 x6 : 000000000000003f 
[  473.791739] x5 : 0000000000000040 x4 : 0000000000000000 
[  473.797042] x3 : 0000000000000000 x2 : ffffff80022410a0 
[  473.802346] x1 : 0000000000000000 x0 : 0000000000000000 
[  473.807651] Call trace:
[  473.810090]  0xffffffc0089fa570 [mac80211@00000000351eb123+0x79000]
[  473.816349]  0xffffffc0089a4bd8 [mac80211@00000000351eb123+0x79000]
[  473.822608]  0xffffffc0089bb8b4 [mac80211@00000000351eb123+0x79000]
[  473.828867]  0xffffffc00896b170 [cfg80211@00000000b2916b15+0x44000]
[  473.835123]  0xffffffc0106bd6c4
[  473.838254]  0xffffffc0106be9b0
[  473.841385]  0xffffffc0106c16a4
[  473.844515]  0xffffffc0106c2108
[  473.847646]  0xffffffc0106c0ea8
[  473.850776]  0xffffffc0106c1624
[  473.853907]  0xffffffc0106c06d4
[  473.857038]  0xffffffc0106c0960
[  473.860168]  0xffffffc010631890
[  473.863300]  0xffffffc010634be8
[  473.866431]  0xffffffc010634cb4
[  473.869561]  0xffffffc010634d20
[  473.872692]  0xffffffc010010f40
[  473.875822]  0xffffffc01001100c
[  473.878953]  0xffffffc010800f44
[  473.882083]  0xffffffc010801360
[  473.885214]  0xffffffc0100025c8
[  473.888345] ---[ end trace 3f03cf97e6c88409 ]---
```

Probably related:
- https://github.com/openwrt/openwrt/commit/08e1812900b4f2ab0d7a33b0e72be2d377a5b3d9
- https://github.com/openwrt/openwrt/commit/aab535d2bbc414aecde5770152e7da6690462b9a
- https://github.com/openwrt/openwrt/commit/f608779f92b368de6f2e858e840bc29c717753c5

ynezz, 2022-06-30:
Does it happen only with 5.10 kernel or you're able to reproduce it with 5.15 as well using latest master snapshot?

ynezz, 2022-06-30:
Resolving those missing function symbols in your stack trace would help tracking down the issue significantly, so either rebuild the image with `CONFIG_KERNEL_DEBUG_INFO=y` or use master snapshot where it's enabled by default.

PolynomialDivision, 2022-07-02:
> Does it happen only with 5.10 kernel or you're able to reproduce it with 5.15 as well using latest master snapshot?

With latest snapshot:
```
[   19.205135] ------------[ cut here ]------------
[   19.209764] Device phy1 AC 0 pending airtime underflow: 4294967076, 220
[   19.209811] WARNING: CPU: 0 PID: 834 at ieee80211_sta_update_pending_airtime+0x1c0/0x1cc [mac80211]
[   19.225498] Modules linked in: nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_bridge nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT chacha_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY poly1305_neon nfnetlink nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic libcrc32c libchacha iptable_mangle iptable_filter ipt_ECN ip_tables hwmon compat ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6
[   19.225675]  ip6_udp_tunnel udp_tunnel seqiv leds_gpio xhci_plat_hcd gpio_button_hotplug
[   19.320662] CPU: 0 PID: 834 Comm: napi/phy1-9 Tainted: G S                5.15.49 #0
[   19.328397] Hardware name: Linksys E8450 (UBI) (DT)
[   19.333265] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   19.340219] pc : ieee80211_sta_update_pending_airtime+0x1c0/0x1cc [mac80211]
[   19.347285] lr : ieee80211_sta_update_pending_airtime+0x1c0/0x1cc [mac80211]
[   19.354346] sp : ffffffc008f53af0
[   19.357651] x29: ffffffc008f53af0 x28: 0000000000000001 x27: 0000000000000000
[   19.364783] x26: ffffff8003661928 x25: 000000000000011f x24: 0000000000000000
[   19.371914] x23: 0000000000000000 x22: ffffffc008f53d00 x21: 00000000ffffff24
[   19.379044] x20: ffffff8002201224 x19: ffffff8002201080 x18: 0000000000000146
[   19.386176] x17: 00000000ffffffff x16: 0000000000000000 x15: ffffffc008b6a570
[   19.393306] x14: 00000000000003d2 x13: 0000000000000146 x12: ffffffc008f53818
[   19.400437] x11: fffffffffffe5900 x10: fffffffffffe58b8 x9 : ffffffc008bc2570
[   19.407568] x8 : 0000000000001e90 x7 : ffffffc008b6a570 x6 : 0000000000000001
[   19.414698] x5 : ffffff801fea26f8 x4 : 0000000000000000 x3 : 0000000000000027
[   19.421828] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 000000000000003b
[   19.428960] Call trace:
[   19.431396]  ieee80211_sta_update_pending_airtime+0x1c0/0x1cc [mac80211]
[   19.438111]  ieee80211_tx_status_ext+0x234/0x8a0 [mac80211]
[   19.443696]  __mt76_tx_complete_skb+0x1d0/0x1ec [mt76]
[   19.448834]  mt7915_mcu_rf_regval+0x1e8/0x6f4 [mt7915e]
[   19.454058]  mt7915_mac_wtbl_lmac_addr+0x750/0x8d0 [mt7915e]
[   19.459714]  mt7915_rx_check+0x30/0xf4 [mt7915e]
[   19.464329]  mt76_dma_rx_poll+0x20c/0xc60 [mt76]
[   19.468943]  __napi_poll+0x34/0x174
[   19.472431]  napi_threaded_poll+0x88/0xec
[   19.476434]  kthread+0x11c/0x130
[   19.479657]  ret_from_fork+0x10/0x20
[   19.483226] ---[ end trace 1bc0206418c5834f ]---
[   90.218724] device br-uplink entered promiscuous mode
````

PolynomialDivision, 2022-07-03:
With latest snapshot I get all the time kernel traces:
```
[  424.757736] ------------[ cut here ]------------
[  424.762364] WARNING: CPU: 1 PID: 4442 at airtime_link_metric_get+0xd0/0x720 [mac80211]
[  424.770323] Modules linked in: nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_bridge nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT chacha_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY poly1305_neon nfnetlink nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic libcrc32c libchacha iptable_mangle iptable_filter ipt_ECN ip_tables hwmon compat ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6
[  424.770505]  ip6_udp_tunnel udp_tunnel seqiv leds_gpio xhci_plat_hcd gpio_button_hotplug
[  424.865491] CPU: 1 PID: 4442 Comm: ip Tainted: G S      W         5.15.49 #0
[  424.872533] Hardware name: Linksys E8450 (UBI) (DT)
[  424.877400] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  424.884354] pc : airtime_link_metric_get+0xd0/0x720 [mac80211]
[  424.890210] lr : airtime_link_metric_get+0x3c/0x720 [mac80211]
[  424.896055] sp : ffffffc009a12ef0
[  424.899361] x29: ffffffc009a12ef0 x28: ffffffc008b589e0 x27: ffffff80014902e0
[  424.906493] x26: ffffff8002369080 x25: ffffff8003604900 x24: ffffff8003734580
[  424.913623] x23: 000000000000001e x22: ffffffc009a13020 x21: 0000000000000000
[  424.920754] x20: 0000000000000000 x19: ffffff8001490000 x18: ffffff8001457bb0
[  424.927884] x17: 00000000ffffffff x16: 0000000000000000 x15: 000000004160f000
[  424.935015] x14: 0000000000001760 x13: 0000000000000281 x12: 0000000000000bb0
[  424.942145] x11: 000000e20cd662b6 x10: 0000000000000840 x9 : 0000000000000000
[  424.949276] x8 : ffffff8003734800 x7 : 0000000000000000 x6 : 000000000000003f
[  424.956406] x5 : 0000000000000040 x4 : 0000000000000000 x3 : 0000000000000000
[  424.963537] x2 : ffffff8002369080 x1 : 0000000000000000 x0 : 0000000000000000
[  424.970668] Call trace:
[  424.973106]  airtime_link_metric_get+0xd0/0x720 [mac80211]
[  424.978618]  sta_set_sinfo+0x5b4/0xbbc [mac80211]
[  424.983336]  sta_set_sinfo+0xb18/0xbbc [mac80211]
[  424.988054]  __sta_info_flush+0x104/0x154 [mac80211]
[  424.993031]  ieee80211_stop_mesh+0x30/0x104 [mac80211]
[  424.998182]  ieee80211_color_change_finish+0xce0/0x20c0 [mac80211]
[  425.004376]  __cfg80211_leave_mesh+0x4c/0xd0 [cfg80211]
[  425.009612]  __cfg80211_leave+0x15c/0x190 [cfg80211]
[  425.014580]  cfg80211_register_netdevice+0x1c0/0x6d0 [cfg80211]
[  425.020504]  raw_notifier_call_chain+0x50/0x70
[  425.024945]  call_netdevice_notifiers_info+0x50/0xa0
[  425.029905]  __dev_close_many+0x40/0x100
[  425.033821]  __dev_change_flags+0x98/0x190
[  425.037910]  dev_change_flags+0x20/0x60
[  425.041737]  do_setlink+0x218/0xda0
[  425.045222]  __rtnl_newlink+0x480/0x7cc
[  425.049053]  rtnl_newlink+0x48/0x70
[  425.052536]  rtnetlink_rcv_msg+0x104/0x334
[  425.056626]  netlink_rcv_skb+0x58/0x120
[  425.060456]  rtnetlink_rcv+0x14/0x20
[  425.064025]  netlink_unicast+0x1e0/0x2d0
[  425.067940]  netlink_sendmsg+0x1a0/0x3c4
[  425.071855]  ____sys_sendmsg+0x260/0x2a4
[  425.075771]  ___sys_sendmsg+0x78/0xc0
[  425.079427]  __sys_sendmsg+0x54/0xa0
[  425.082995]  __arm64_sys_sendmsg+0x20/0x30
[  425.087084]  invoke_syscall+0x44/0x110
[  425.090826]  el0_svc_common.constprop.0+0x3c/0xe4
[  425.095523]  do_el0_svc+0x1c/0x2c
[  425.098830]  el0_svc+0x14/0x50
[  425.101881]  el0t_64_sync_handler+0x9c/0x120
[  425.106144]  el0t_64_sync+0x158/0x15c
[  425.109799] ---[ end trace 793633cbb71ee091 ]---
```

closed 2022-07-06 by PolynomialDivision