openwrt/openwrt #10163: Lantiq: no cpu temperature in statistic

turboproc, 2022-06-30:
Noticed that the VRX200 temperature cannot be read by the collectd package as the temperature is not at the "standard" location in the sys fs.
After a bit of exploring I concluded that a simple patch of thermal.c in the collectd package will make it all work.

More than happy to contribute if I knew how to contribute a patch file. Happy to learn as I also survived my first commit earlier this week.

@xdarklight, any help would be appreciated. Your support on the first commit work quite well.

xdarklight, 2022-06-30:
I don't use collectd myself. Their `thermal.c` implementation doesn't look like it's using the hwmon sysfs interface.
It's probably best to discuss any code changes that you want to make with the collectd developers

nicefile, 2023-01-08:
cant confirm that . My device use collected with sensor plugin and had 0 issues.
```root@7362sl:~# sensors
ltq_cputemp-isa-0000
Adapter: ISA adapter
temp1:        +50.0°C

root@7362sl:~# sensors --v
sensors version 3.6.0 with libsensors version 3.6.0
root@7362sl:~#
```