openwrt/openwrt #10164: kmod-net-rtl8192su is empty

M95D, 2022-07-01:
Download this file:
https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/kmods/5.10.127-1-9016c742c67c216f0c1ec2e815af16d1/kmod-net-rtl8192su_5.10.127-1_arm_cortex-a9_vfpv3-d16.ipk
Try to extract the kernel module. You can't. It's not there. The package is empty.

I tried to build that module on my own build system. Same result. The module is not built. The resulting package is empty.

Second bug:
Building should fail, but it does not. It simply constructs an empty package.

chunkeey, 2022-07-01:
> See prism54.

oh, the prism54 could be dropped entirely. It was removed upstream.

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=d249ff28b1d8478419caf347c4836361a47b0c5b

Feel free to remove the driver+firmware package.

As for r8712u.ko. I think it's not being picked up because CONFIG_CFG80211 is not set in the main kernel's config.
This makes it so that the module isn't selectable. This is an interaction with the backport package that supplies our
out-of-tree wireless drivers. (Maybe it's possible to move r8712u to the backport package in order to fix this?)

chunkeey, 2022-08-12:
Update:

https://github.com/openwrt/openwrt/commit/7ebe1dca476ddb2c08f8a1cbbd0522e69c1edc82 removed the empty kmod-net-rtl8192su package. At this point it would need to be added through the mac80211 / backports package. I think this can be closed,
although this might not be the resolution we wanted :disappointed:.

closed 2022-08-12 by chunkeey