openwrt/openwrt #10167: openwrt x86_64 22.03.0-rc4 invalid opcodes reported in libwolfssl in uhttpd

systemdlete, 2022-07-01:
I am seeing errors on the openwrt console from what appears to be libwolfssl linked in uhttpd.  I am only seeing this in 22.03.0-rc4; the rc1 works.

```
 traps: uhttpd[1975] trap invalid opcode ip:7f3319be5c46 sp:7ffe08129168 error:0 in libwolfssl.so.5.3.0.ee39414e[7f3319b51000+d1000]
 traps: uhttpd[2755] trap invalid opcode ip:7f417fad8c46 sp:7ffcc7aa1388 error:0 in libwolfssl.so.5.3.0.ee39414e[7f417fa44000+d1000]
 traps: uhttpd[2766] trap invalid opcode ip:7f2e94c1dc46 sp:7ffc1f86f3b8 error:0 in libwolfssl.so.5.3.0.ee39414e[7f2e94b89000+d1000]
 traps: uhttpd[2777] trap invalid opcode ip:7f9e8015ac46 sp:7ffedd9c6128 error:0 in libwolfssl.so.5.3.0.ee39414e[7f9e800c6000+d1000]
 traps: uhttpd[2788] trap invalid opcode ip:7fd71bb39c46 sp:7ffe05dda8e8 error:0 in libwolfssl.so.5.3.0.ee39414e[7fd71baa5000+d1000]
 traps: uhttpd[2799] trap invalid opcode ip:7f3fbd00ac46 sp:7ffef2ce0b38 error:0 in libwolfssl.so.5.3.0.ee39414e[7f3fbcf76000+d1000]
```

I installed https://downloads.openwrt.org/releases/22.03.0-rc4/targets/x86/64/openwrt-22.03.0-rc4-x86-64-generic-ext4-combined.img.gz to a virtualbox VM (6.1.34, latest) which had been running https://downloads.openwrt.org/releases/22.03.0-rc1/targets/x86/64/openwrt-22.03.0-rc1-x86-64-generic-ext4-combined.img.gz without incident.   I did not change the VM configuration, which is using the Linux Kernel VM (kvm) for virtualization.

"service uhttpd status" reports that the web server is running.  But "ps" indicates it is not running.   Thus, the web interface does not work.

Here is the /proc/cpuinfo:
[cpuinfo.txt](https://github.com/openwrt/openwrt/files/9031603/cpuinfo.txt)

And here is the dmesg output:
[dmesg.txt](https://github.com/openwrt/openwrt/files/9031605/dmesg.txt)

I am able to restore the VM back to the rc1 (or prior) version of openwrt and it resumes working normally (i.e., I do not experience those errors above).

systemdlete, 2022-07-01:
I apologize for this late entry; I just got to testing rc4 today.

pkgadd, 2022-07-01:
That is known (depending on your, emulated, CPU capabilities) and should be fixed with f91b0d7a92c5927db9eb6610628d931e3104cd41.

closed 2022-07-03 by hauke