openwrt/openwrt #10183: [Feature Request] Disable IPv4 on this interface

nasbdh9, 2022-07-03:
Whether IPv6 is enabled on an interface can be controlled using the `option ipv6` option
Is it possible to add an `option ipv4` to control whether ipv4 is on the interface? or there is already such an implementation and I don't know

jow-, 2022-07-13:
In contrast to IPv6, IPv4 does not use link local addressing without additional services (avahi) so not configuring an IP or not running a DHCP client is enough to disable IPv4 on an interface.

closed 2022-07-13 by jow-

nasbdh9, 2022-07-13:
> In contrast to IPv6, IPv4 does not use link local addressing without additional services (avahi) so not configuring an IP or not running a DHCP client is enough to disable IPv4 on an interface.

But this does not disable dropping (disabled?) ipv4 addresses on the pppoe client interface, see commit https://git.openwrt.org/?p=project/netifd.git;a=commit;h=98ca67469da70accd710b14c28e77be2126170a8 to know disabling ipv6 It is implemented through the sysctl option, but currently there is no disable_ipv4 option in sysctl