openwrt/openwrt #10185: Linksys 3200ACM 21+ openwrt wifi.

papergirl, 2022-07-03:
Device: Linksys 3200ACM 
Software: OpenWrt 22.03.0-rc4 (issue also occurred on 21.02 ), issue did not occur on 19.07, reflashing back to 19.07 corrects the problem. 

Device Driver [mv88e6xxx-1:00] driver [Marvell 88E1540]

Dissociation of a wireless device, which happens far too quickly after a device has been connected, causes the web ui luci to lock up, the router it's self is still operational, (the lan and wan etc) the barring the wifi and web ui. rebooting it works the wireless comes up OK, up until a device connects successfully (gets and IP and everything), then is disconnected. 

It feels like there is multiple bugs at work here so I was unsure where to file it

1.  The driver problem.
2.  A problem luci/uhttpd becoming deadlocked. It logs in a user ok, but all calls to the endpoint ubus timeout, Unsure where to find logs for that or if it even logs at all tbh.

Relavant logread bit:
Sun Jul  3 17:02:28 2022 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 7c:87:ce:49:d5:38
Sun Jul  3 17:02:28 2022 daemon.info hostapd: wlan1: STA 7c:87:ce:49:d5:38 IEEE 802.11: disassociated due to inactivity
Sun Jul  3 17:02:48 2022 kern.err kernel: [  392.029132] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
Sun Jul  3 17:02:48 2022 kern.err kernel: [  392.035263] ieee80211 phy1: return code: 0x1122
Sun Jul  3 17:02:48 2022 kern.err kernel: [  392.039812] ieee80211 phy1: timeout: 0x1122
Sun Jul  3 17:02:48 2022 kern.err kernel: [  392.044020] wlan1: failed to remove key (0, 7c:87:ce:49:d5:38) from hardware (-5)
Sun Jul  3 17:02:48 2022 daemon.info hostapd: wlan1: STA 7c:87:ce:49:d5:38 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Jul  3 17:02:48 2022 kern.debug kernel: [  392.052158] ieee80211 phy1: MACREG_REG_INT_CODE: 0x0000
Sun Jul  3 17:03:08 2022 kern.err kernel: [  412.060435] ieee80211 phy1: cmd 0x9125=BAStream timed out
Sun Jul  3 17:03:08 2022 kern.err kernel: [  412.065860] ieee80211 phy1: return code: 0x1125
Sun Jul  3 17:03:08 2022 kern.err kernel: [  412.070409] ieee80211 phy1: timeout: 0x1125
Sun Jul  3 17:03:08 2022 kern.debug kernel: [  412.074746] ieee80211 phy1: MACREG_REG_INT_CODE: 0x0000
Sun Jul  3 17:03:28 2022 kern.err kernel: [  432.082155] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
Sun Jul  3 17:03:28 2022 kern.err kernel: [  432.088018] ieee80211 phy1: return code: 0x1111
... repeats ad infinitium.

Let me know if I can provide anymore infomation,  

tah.

papergirl, 2022-07-03:
oh the issue occurs on phy0 device also (which is identical, but its setup to be 5GHz stead of 2GHz),

nnune025, 2023-01-10:
Hi! I'm having the same issue wanted to know if you had any luck solving it? From some research I've found the following posts that may be relevant.

https://github.com/openwrt/openwrt/issues/9956
https://www.reddit.com/r/openwrt/comments/t3l92s/openwrt_21022_on_linksys_wrt3200acm/
https://openwrt.org/toh/linksys/wrt3200acm

I'm testing it out some more today.

nnune025, 2023-01-10:
OpenWrt SNAPSHOT r21753-2a3283643c / LuCI Master git-22.361.69865-deed682

[openwrt-0b431a6a6837-mvebu-cortexa9-linksys_wrt3200acm-squashfs-sysupgrade (stable).zip](https://github.com/openwrt/openwrt/files/10387150/openwrt-0b431a6a6837-mvebu-cortexa9-linksys_wrt3200acm-squashfs-sysupgrade.stable.zip)

I just installed and tested it, seems to be working fine. The attached is the bin I used for my WRT3200ACM, I custom build the firmware to not include opkg kmod-mwifiex-sdio so radio 2 does not show up in luci.

P.S.
Since its a snap shot build luci isnt included you will have to install via SSH with the following commands:

opkg update
opkg install luci
/etc/init.d/uhttpd enable
/etc/init.d/uhttpd restart

LaurenceGough, 2023-02-08:
Same issue here on 21.02.05, doing my head in. LuCI dies, completely unresponsive. SSH lags like mad but sometimes work.  I had to go back to 21.0.2.05 due to issues with the new 22.03 builds with this router's switch.

Seems to be an issue when enabling certain features such as KRACK counter measures or WPA3. 

All works great until the client disconnects from the OpenWRT router, then all goes mad.  

Perhaps it's to do with 802.11w or management frame protection, it does have a warning for our wonderful Marvell wifi.... (last time I'm buying one of their chipset routers).

Wed Feb  8 01:05:27 2023 daemon.info dnsmasq-dhcp[4298]: DHCPDISCOVER(br-lan.4) 64:xxxxxxxxxxxx
Wed Feb  8 01:05:27 2023 daemon.info dnsmasq-dhcp[4298]: DHCPOFFER(br-lan.4) 192.168.16.163 64:xxxxxxxxxxxx
Wed Feb  8 01:05:27 2023 daemon.info dnsmasq-dhcp[4298]: DHCPREQUEST(br-lan.4) 192.168.16.163 64:xxxxxxxxxxxx
Wed Feb  8 01:05:27 2023 daemon.info dnsmasq-dhcp[4298]: DHCPACK(br-lan.4) 192.168.16.163 64:xxxxxxxxxxxx 
Wed Feb  8 01:06:25 2023 daemon.info dnsmasq-dhcp[4298]: DHCPREQUEST(br-lan.4) 192.168.16.163 64:xxxxxxxxxxxx
Wed Feb  8 01:06:25 2023 daemon.info dnsmasq-dhcp[4298]: DHCPACK(br-lan.4) 192.168.16.163 64:xxxxxxxxxxxx 
Wed Feb  8 01:06:26 2023 daemon.info dnsmasq-dhcp[4298]: DHCPREQUEST(br-lan.4) 192.168.16.163 64:xxxxxxxxxxxx
Wed Feb  8 01:06:26 2023 daemon.info dnsmasq-dhcp[4298]: DHCPACK(br-lan.4) 192.168.16.163 64:xxxxxxxxxxxx 
Wed Feb  8 01:06:28 2023 daemon.info dnsmasq-dhcp[4298]: DHCPREQUEST(br-lan.4) 192.168.16.163 64:xxxxxxxxxxxx
Wed Feb  8 01:06:28 2023 daemon.info dnsmasq-dhcp[4298]: DHCPACK(br-lan.4) 192.168.16.163 64:xxxxxxxxxxxx 
Wed Feb  8 01:06:37 2023 daemon.info dnsmasq-dhcp[4298]: DHCPREQUEST(br-lan.4) 192.168.16.163 64:xxxxxxxxxxxx
Wed Feb  8 01:06:37 2023 daemon.info dnsmasq-dhcp[4298]: DHCPACK(br-lan.4) 192.168.16.163 64:xxxxxxxxxxxx 
Wed Feb  8 01:06:38 2023 daemon.info dnsmasq-dhcp[4298]: DHCPREQUEST(br-lan.4) 192.168.16.163 64:xxxxxxxxxxxx
Wed Feb  8 01:06:38 2023 daemon.info dnsmasq-dhcp[4298]: DHCPACK(br-lan.4) 192.168.16.163 64:xxxxxxxxxxxx