openwrt/openwrt #10188: Openwrt's odhcp6c configuration means that interface metric specification is not used for setup routes

d1b, 2022-07-04:
Seemingly Openwrt's odhcp6c configuration means that interface metric specification is not used for setup routes. This matters when a primary connection and a fallback connection:

* are active at the same time 
* provide IPv6 connectivity 

E.g. config in `/etc/config/network` 
```
config interface 'wan'
        option ifname 'eth1'
        option proto 'dhcp'
        option ipv6 '1'

config interface 'wan6'
        option ifname '@wan'
        option proto 'dhcpv6'

config interface 'lte'
        option ifname 'eth2'
        option proto 'dhcp'
        option ipv6 1

config interface 'lte6'
        option ifname '@lte'
        option proto 'dhcpv6'
        option metric 1000
``` 

```ip -6 route
default from 2001:<un-important-1> via fe80::<unimportant-2> dev eth2 proto static metric 384 pref medium
default from 2001:<unimportant-3>:/64 via fe80::<unimportant-2> dev eth2 proto static metric 384 pref medium
default from 2403:<unimportant-4>/48 via fe80::<unimportant-5> dev eth1 proto static metric 384 pref medium
default from 2403:<unimportant-6> via fe80::<unimportant-5> dev eth1 proto static metric 384 pref medium
2001:<unimportant-3>:/64 dev eth2 proto static metric 256 pref medium
unreachable 2001:<unimportant-3>:/64 dev lo proto static metric 2147483647 pref medium
2403:<unimportant-4>/64 dev br-lan proto kernel metric 256 expires 5290sec pref medium
2403:<unimportant-4>/64 dev br-lan proto static metric 1024 pref medium
unreachable 2403:<unimportant-4>/48 dev lo proto static metric 2147483647 pref medium
fd5b:<unimportant-9>/64 dev br-lan proto kernel metric 256 expires 6103sec pref medium
fd5b:<unimportant-9>/64 dev br-lan proto static metric 1024 pref medium
unreachable fd5b:<unimportant-9>/56 dev lo proto static metric 2147483647 pref medium
fe80::/64 dev eth2 proto kernel metric 256 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev br-lan proto kernel metric 256 pref medium
fe80::/64 dev eth1 proto kernel metric 256 pref medium
fe80::/64 dev wlan0 proto kernel metric 256 pref medium
fe80::/64 dev wlan0-1 proto kernel metric 256 pref medium
fe80::/64 dev wlan0-2 proto kernel metric 256 pref medium
fe80::/64 dev ifb4eth1 proto kernel metric 256 pref medium```
```

```
route -n -A inet6
Kernel IPv6 routing table
Destination                                 Next Hop                                Flags Metric Ref    Use Iface
::/0                                        fe80::<unimportant-20>               UG    384    1        0 eth2    
::/0                                        fe80::<unimportant-20>               UG    384    5        0 eth2    
::/0                                        fe80::<unimportant-21>                    UG    384    7        0 eth1    
::/0                                        fe80::<unimportant-21>                    UG    384    5        0 eth1    
<snip>
```

As far as I can tell the specified metric of `1000` for eth2 - `lte6`, the 2001 ... address, does not get applied. This seems to relate to the `dhcpv6` script openwrt has for `odhcp6c` -> https://github.com/openwrt/openwrt/blob/master/package/network/ipv6/odhcp6c/files/dhcpv6.script  .

jamesmacwhite, 2022-09-18:
It would appear the metric option does not work with dhcpv6 client interfaces, it does for static IPv6 interfaces.

mibofra, 2024-07-03:
Any news about this? Still affects 23.05.3