openwrt/openwrt #10200: Kernel log [backports-5.10.110-1/net/mac80211/tx.c:4859]

chadf, 2022-07-05:
I regularly get the following log message:
```
[  208.173167] ------------[ cut here ]------------
[  208.182450] WARNING: CPU: 0 PID: 63 at backports-5.10.110-1/net/mac80211/tx.c:4859 0x835b2eb0 [mac80211@36191169+0x7eb80]
[  208.204308] Modules linked in: rt2800soc rt2800mmio rt2800lib pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT rt2x00soc rt2x00mmio rt2x00lib pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio gpio_button_hotplug
[  208.321394] CPU: 0 PID: 63 Comm: kworker/u2:2 Not tainted 5.4.188 #0
[  208.334089] Workqueue: phy0 0x835a7bc0 [mac80211@36191169+0x7eb80]
[  208.346420] Stack : 805a0000 83ccfb00 80507cd4 835b2eb0 00000000 00000000 00000000 00000000
[  208.363092]         00000000 00000000 00000000 00000000 00000000 00000001 83dad990 5f7f58a7
[  208.379755]         83dada28 00000000 00000000 00000000 00000038 804ed8e4 39363131 3778302b
[  208.396418]         00000000 0000000a 00000000 00051909 00000000 83dad970 00000000 835b2eb0
[  208.413080]         00000009 00000000 834ba460 834a6730 00000001 802ac32c 00000000 806f0000
[  208.429743]         ...
[  208.434622] Call Trace:
[  208.434656] [<835b2eb0>] 0x835b2eb0 [mac80211@36191169+0x7eb80]
[  208.451349] [<804ed8e4>] 0x804ed8e4
[  208.458321] [<835b2eb0>] 0x835b2eb0 [mac80211@36191169+0x7eb80]
[  208.470131] [<802ac32c>] 0x802ac32c
[  208.477094] [<80009790>] 0x80009790
[  208.484051] [<80009798>] 0x80009798
[  208.491007] [<80023900>] 0x80023900
[  208.497976] [<835b2eb0>] 0x835b2eb0 [mac80211@36191169+0x7eb80]
[  208.509783] [<800239a8>] 0x800239a8
[  208.516761] [<835b2eb0>] 0x835b2eb0 [mac80211@36191169+0x7eb80]
[  208.528583] [<835b2f00>] 0x835b2f00 [mac80211@36191169+0x7eb80]
[  208.540402] [<835b75fc>] 0x835b75fc [mac80211@36191169+0x7eb80]
[  208.552228] [<835b91a0>] 0x835b91a0 [mac80211@36191169+0x7eb80]
[  208.564042] [<835aadbc>] 0x835aadbc [mac80211@36191169+0x7eb80]
[  208.575846] [<8033849c>] 0x8033849c
[  208.582826] [<835b765c>] 0x835b765c [mac80211@36191169+0x7eb80]
[  208.594630] [<8004c03c>] 0x8004c03c
[  208.601587] [<8011fc80>] 0x8011fc80
[  208.608561] [<83558110>] 0x83558110 [mt76x02_lib@833236f4+0x9ce0]
[  208.620712] [<8004c344>] 0x8004c344
[  208.627680] [<835beba0>] 0x835beba0 [mac80211@36191169+0x7eb80]
[  208.639497] [<8004cb18>] 0x8004cb18
[  208.646469] [<835580d0>] 0x835580d0 [mt76x02_lib@833236f4+0x9ce0]
[  208.658640] [<835bec9c>] 0x835bec9c [mac80211@36191169+0x7eb80]
[  208.670469] [<83554eb0>] 0x83554eb0 [mt76x02_lib@833236f4+0x9ce0]
[  208.682621] [<8003bd10>] 0x8003bd10
[  208.689580] [<80026790>] 0x80026790
[  208.696541] [<804f2d80>] 0x804f2d80
[  208.703497] [<80338e70>] 0x80338e70
[  208.710475] [<80026968>] 0x80026968
[  208.717438] [<83551ab4>] 0x83551ab4 [mt76x02_lib@833236f4+0x9ce0]
[  208.729597] [<834ed940>] 0x834ed940 [mt76x2_common@2770eb68+0x2da0]
[  208.742098] [<80026a4c>] 0x80026a4c
[  208.749079] [<834e6cc8>] 0x834e6cc8 [mt76@d0bb5786+0xa9c0]
[  208.760019] [<80039f1c>] 0x80039f1c
[  208.766994] [<834e6ec4>] 0x834e6ec4 [mt76@d0bb5786+0xa9c0]
[  208.777944] [<834e6d6c>] 0x834e6d6c [mt76@d0bb5786+0xa9c0]
[  208.788909] [<834e7074>] 0x834e7074 [mt76@d0bb5786+0xa9c0]
[  208.799875] [<834fc474>] 0x834fc474 [mt76x2e@c4cec529+0x27a0]
[  208.811350] [<83581ad8>] 0x83581ad8 [mac80211@36191169+0x7eb80]
[  208.823162] [<80357968>] 0x80357968
[  208.830144] [<835c9e78>] 0x835c9e78 [mac80211@36191169+0x7eb80]
[  208.841984] [<835cab60>] 0x835cab60 [mac80211@36191169+0x7eb80]
[  208.853800] [<8004a9bc>] 0x8004a9bc
[  208.860775] [<835a7c60>] 0x835a7c60 [mac80211@36191169+0x7eb80]
[  208.872579] [<804eeac0>] 0x804eeac0
[  208.879545] [<8003b02c>] 0x8003b02c
[  208.886529] [<8003b3f4>] 0x8003b3f4
[  208.893483] [<8003b28c>] 0x8003b28c
[  208.900439] [<804eecf8>] 0x804eecf8
[  208.907409] [<8003b28c>] 0x8003b28c
[  208.914365] [<80040144>] 0x80040144
[  208.921322] [<8004000c>] 0x8004000c
[  208.928285] [<8004000c>] 0x8004000c
[  208.935247] [<800050b8>] 0x800050b8
[  208.942211]
[  208.945188] ---[ end trace b91b09fb50778843 ]---
```
Is this a known/resolved issue, which will be fixed with the upcoming 22.03 release? I usually have to reboot the router at least once a day to resolve wifi & internet issues (which may or may not be related to this log). It's also hardware which has only supported by recent releases, so some issues are expected.

Model: Wavlink WL-WN530HG4
Firmware: OpenWrt 21.02.3 r16554-1d4dea6d4f / LuCI openwrt-21.02 branch git-22.167.28411-ee8170b

When I look at the referenced line in the mentioned backports source, it only contains a simple `if` statement:
```        } else if (sdata->vif.type == NL80211_IFTYPE_ADHOC) {```

It's even more confusing that while the log output gives lots of information, it doesn't actually seem to make it clear what the problem is, or where I should start looking to try and fix it. Otherwise, the only thing I can do is wait for 22.03 to get past RC state and hope it magically fixes things.

lukasz1992, 2022-07-10:
@chadf when you check line number, please remember that OpenWrt do many patches - you need to build image by yourself, to see what code is actually in specified line number.

chadf, 2022-07-16:
FYI, for anyone interested. The related code is:
```
static u8 __ieee80211_beacon_update_cntdwn(struct beacon_data *beacon)
{
        beacon->cntdwn_current_counter--;

        /* the counter should never reach 0 */
        WARN_ON_ONCE(!beacon->cntdwn_current_counter);

        return beacon->cntdwn_current_counter;
}
```
The `388-mac80211-add-support-for-BSS-color-change` patch does appear to touch some code dealing with the counter, or it may be broken upstream. I will take a look around to see if I stumble across the cause, if no one else figures it out first.

chadf, 2023-03-27:
After upgrading to 22.03.3, it still appears to be there. Though, I haven't done a build yet to verify it is the same code point.

[35353.452902] ------------[ cut here ]------------
[35353.462224] WARNING: CPU: 0 PID: 17624 at backports-5.15.81-1/net/mac80211/tx.c:4770 0x81a35e28 [mac80211@ce1d14ef+0x7f660]
[35353.484432] Modules linked in: rt2800soc rt2800mmio rt2800lib pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet rt2x00soc rt2x00mmio rt2x00lib pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug crc32c_generic
[35353.619840] CPU: 0 PID: 17624 Comm: kworker/u2:0 Not tainted 5.10.161 #0
[35353.633223] Workqueue: phy0 0x81a2888c [mac80211@ce1d14ef+0x7f660]
[35353.645555] Stack : 00000000 80598748 80640000 82458a80 00000000 00000000 00000000 00000000
[35353.662242]         00000000 00000000 00000000 00000000 00000000 00000001 82e3fa30 3e93ef31
[35353.678926]         82e3fac8 00000000 00000000 82e3f8b8 00000038 802c7124 00000000 ffffffea
[35353.695611]         00000000 82e3f8c4 00000108 80642fc8 82e3fa10 80598748 00000000 81a35e28
[35353.712295]         00000009 82e3b800 81ecaeb0 81b1a4c0 00000018 80315d48 00000000 80800000
[35353.728985]         ...
[35353.733851] Call Trace:
[35353.733868] [<802c7124>] 0x802c7124
[35353.745709] [<81a35e28>] 0x81a35e28 [mac80211@ce1d14ef+0x7f660]
[35353.757534] [<80315d48>] 0x80315d48
[35353.764476] [<80006d48>] 0x80006d48
[35353.771433] [<80006d50>] 0x80006d50
[35353.778393] [<800267d0>] 0x800267d0
[35353.785366] [<81a35e28>] 0x81a35e28 [mac80211@ce1d14ef+0x7f660]
[35353.797177] [<80026898>] 0x80026898
[35353.804142] [<81a35e28>] 0x81a35e28 [mac80211@ce1d14ef+0x7f660]
[35353.815954] [<803ca478>] 0x803ca478
[35353.822908] [<81a38800>] 0x81a38800 [mac80211@ce1d14ef+0x7f660]
[35353.834720] [<800511cc>] 0x800511cc
[35353.841690] [<81a388c4>] 0x81a388c4 [mac80211@ce1d14ef+0x7f660]
[35353.853498] [<80580000>] 0x80580000
[35353.860469] [<81968168>] 0x81968168 [mt76x02_lib@44b3b4ca+0x9da0]
[35353.872645] [<81a3f000>] 0x81a3f000 [mac80211@ce1d14ef+0x7f660]
[35353.884453] [<8003f750>] 0x8003f750
[35353.891420] [<81968128>] 0x81968128 [mt76x02_lib@44b3b4ca+0x9da0]
[35353.903599] [<81a3f0fc>] 0x81a3f0fc [mac80211@ce1d14ef+0x7f660]
[35353.915408] [<800711dc>] 0x800711dc
[35353.922366] [<81964f70>] 0x81964f70 [mt76x02_lib@44b3b4ca+0x9da0]
[35353.934525] [<806d0000>] 0x806d0000
[35353.941478] [<800298b8>] 0x800298b8
[35353.948450] [<805845f8>] 0x805845f8
[35353.955410] [<806d0000>] 0x806d0000
[35353.962363] [<80029b8c>] 0x80029b8c
[35353.969332] [<819545c0>] 0x819545c0 [mt76@b650df1f+0xada0]
[35353.980285] [<80029c64>] 0x80029c64
[35353.987260] [<819576b4>] 0x819576b4 [mt76@b650df1f+0xada0]
[35353.998228] [<8196c4e8>] 0x8196c4e8 [mt76x2e@2f0b23bc+0x2740]
[35354.009703] [<81a4a740>] 0x81a4a740 [mac80211@ce1d14ef+0x7f660]
[35354.021517] [<80583b64>] 0x80583b64
[35354.028486] [<81950488>] 0x81950488 [mt76@b650df1f+0xada0]
[35354.039444] [<819571b4>] 0x819571b4 [mt76@b650df1f+0xada0]
[35354.050406] [<81a4b504>] 0x81a4b504 [mac80211@ce1d14ef+0x7f660]
[35354.062237] [<81a2892c>] 0x81a2892c [mac80211@ce1d14ef+0x7f660]
[35354.074044] [<80580474>] 0x80580474
[35354.080995] [<8003dc48>] 0x8003dc48
[35354.087954] [<8003e038>] 0x8003e038
[35354.094912] [<8003deb8>] 0x8003deb8
[35354.101855] [<8003deb8>] 0x8003deb8
[35354.108810] [<80044344>] 0x80044344
[35354.115768] [<80044208>] 0x80044208
[35354.122709] [<80044208>] 0x80044208
[35354.129672] [<800025f8>] 0x800025f8
[35354.136636]
[35354.139593] ---[ end trace df91244f80290b98 ]---