openwrt/openwrt #10201: xdp-tools: fails to compile on powerpc_8540

PolynomialDivision, 2022-07-05:
```
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://codeload.github.com/xdp-project/xdp-tools/tar.gz/v1.2.5?/xdp-tools-1.2.5.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  247k    0  247k    0     0   491k      0 --:--:-- --:--:-- --:--:--  491k
Found clang binary '/builder/shared-workdir/build/sdk/staging_dir/host/llvm-bpf/bin/clang' with version 13 (from 'clang version 13.0.0 (https://git.openwrt.org/openwrt/openwrt.git 265f402fbdd8444835b43954bf27a83bc9369b91)')
libbpf support: system v0.8.0
  perf_buffer__consume support: yes
  btf__load_from_kernel_by_id support: yes
  btf__type_cnt support: yes
  bpf_object__next_map support: yes
  bpf_object__next_program support: yes
  bpf_program__insn_cnt support: yes
  bpf_map_create support: yes
  perf_buffer__new_raw support: yes
  bpf_xdp_attach support: yes
secure_getenv support: yes
make[4]: Entering directory '/builder/shared-workdir/build/sdk/build_dir/target-powerpc_8540_musl/xdp-tools-1.2.5'

lib

  libxdp
    M4       xdp-dispatcher.c
    CLANG    xsk_def_xdp_prog.o
xsk_def_xdp_prog.c:4:10: fatal error: 'bpf/bpf_helpers.h' file not found
#include <bpf/bpf_helpers.h>
         ^~~~~~~~~~~~~~~~~~~
1 error generated.
make[6]: *** [Makefile:139: xsk_def_xdp_prog.o] Error 1
make[5]: *** [Makefile:20: libxdp] Error 2
make[4]: *** [Makefile:25: lib] Error 2
make[4]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-powerpc_8540_musl/xdp-tools-1.2.5'
make[3]: *** [Makefile:131: /builder/shared-workdir/build/sdk/build_dir/target-powerpc_8540_musl/xdp-tools-1.2.5/.built] Error 2
time: package/feeds/base/xdp-tools/compile#3.97#0.49#7.57
```

@dangowrt @tohojo

PolynomialDivision, 2022-07-05:
Maybe it needs `PKG_FLAGS:=nonshared` like @nbd168 wrote here:
https://github.com/openwrt/openwrt/commit/a667f6b8dd4b0f47d9c0aacb900f1fd3045ec945

PolynomialDivision, 2022-07-05:
Or maybe we have to set `LIBDIR=$(STAGING_DIR)/usr/lib` or `HEADER_DIR=$(STAGING_DIR)/usr/include`?

PolynomialDivision, 2022-07-05:
The include directory is wrong
```
/home/build/openwrt/staging_dir/host/llvm-bpf/bin/clang -S \
    -target bpfel \
    -D __BPF_TRACING__ \
    -Wno-visibility -DBPF_DIR_MNT=\"/sys/fs/bpf\" -DBPF_OBJECT_PATH=\"/usr/lib/bpf\" -DMAX_DISPATCHER_ACTIONS=10 -DTOOLS_VERSION=\""1.2.5"\" -DLIBBPF_VERSION=\"0.8.0\" -DRUNDIR=\"/tmp/run\" -DHAVE_LIBBPF_PERF_BUFFER__CONSUME -DHAVE_LIBBPF_BTF__LOAD_FROM_KERNEL_BY_ID -DHAVE_LIBBPF_BTF__TYPE_CNT -DHAVE_LIBBPF_BPF_OBJECT__NEXT_MAP -DHAVE_LIBBPF_BPF_OBJECT__NEXT_PROGRAM -DHAVE_LIBBPF_BPF_PROGRAM__INSN_CNT -DHAVE_LIBBPF_BPF_MAP_CREATE -DHAVE_LIBBPF_PERF_BUFFER__NEW_RAW -DHAVE_LIBBPF_BPF_XDP_ATTACH -DHAVE_SECURE_GETENV -DDEBUG -DLIBBPF_DYNAMIC -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -I../../headers \
    -Wall \
    -Wno-unused-value \
    -Wno-pointer-sign \
    -Wno-compare-distinct-pointer-types \
    -Werror \
    -O2 -emit-llvm -c -g -o xdp-dispatcher.ll xdp-dispatcher.c
xdp-dispatcher.c:5:10: fatal error: 'bpf/bpf_helpers.h' file not found
#include <bpf/bpf_helpers.h>
         ^~~~~~~~~~~~~~~~~~~
1 error generated.
```

trippleflux, 2022-07-06:
Hmm also getting different errors with NLS enabled : 
```
libbpf support: FORCE_SYSTEM_LIBBPF is set, but no usable libbpf found on system
error: cc1: note: someone does not honour COPTS correctly, passed 0 times
/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.1.0_glibc/lib/gcc/x86_64-openwrt-linux-gnu/12.1.0/../../../../x86_64-openwrt-linux-gnu/bin/ld: warning: libintl.so.8, needed by /home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/libbpf.so, not found (try using -rpath or -rpath-link)
/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.1.0_glibc/lib/gcc/x86_64-openwrt-linux-gnu/12.1.0/../../../../x86_64-openwrt-linux-gnu/bin/ld: /home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/libelf.so.1: undefined reference to `libintl_dgettext'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:141: /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/xdp-tools-1.2.5/.configured_889556d2f423f99e091beece9c8d870a] Error 1
make[3]: Leaving directory '/home/username/works/openwrt/package/network/utils/xdp-tools'
time: package/network/utils/xdp-tools/compile#1.50#0.23#1.66
    ERROR: package/network/utils/xdp-tools failed to build.
make[2]: *** [package/Makefile:116: package/network/utils/xdp-tools/compile] Error 1
make[2]: Leaving directory '/home/username/works/openwrt'
make[1]: *** [package/Makefile:110: /home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/username/works/openwrt'
make: *** [/home/username/works/openwrt/include/toplevel.mk:230: world] Error 2
```
with CONFIG_BPF_TOOLCHAIN_BUILD_LLVM, x86_64 + glibc

tohojo, 2022-07-09:
> Hmm also getting different errors with NLS enabled :
>
> ```
> libbpf support: FORCE_SYSTEM_LIBBPF is set, but no usable libbpf found on system
> error: cc1: note: someone does not honour COPTS correctly, [...]

Ah, this looks like it could be because we're not honouring CFLAGS in the configure script...

dangowrt, 2022-07-09:
@tohojo I've already fixed this with a local patch, passing LDFLAGS was needed: https://github.com/openwrt/openwrt/commit/d29722e6ff764d78428c54a5618b7a0e191245f1

tohojo, 2022-07-09:
Ah, cool, care to submit a PR?

trippleflux, 2022-07-11:
Got further but still error on x86_64 master :
```
hecking dependency style of x86_64-openwrt-linux-gnu-g++... gcc3
checking for bison... bison -y
checking for a sed that does not truncate output... In file included from In file included from xsk_def_xdp_prog.cxdp-dispatcher.c::33:
:
In file included from In file included from ../../headers/linux/bpf.h../../headers/linux/bpf.h::1111:
:
/usr/include/linux/types.h/usr/include/linux/types.h::55::1010::  fatal error: fatal error: 'asm/types.h' file not found'asm/types.h' file not found

#include <asm/types.h>#include <asm/types.h>

         ^~~~~~~~~~~~~         ^~~~~~~~~~~~~

/home/username/works/openwrt/staging_dir/host/bin/sed
checking whether to build with code coverage support... no
checking whether to enable maintainer-specific portions of Makefiles... yes
checking for inline... inline
checking for ANSI C header files... (cached) yes
checking cpuid.h usability... yes
checking cpuid.h presence... yes
checking for cpuid.h... yes
checking for __get_cpuid_count... yes
checking for struct iovec.iov_base... yes
checking netinet/tcp.h usability... yes
checking netinet/tcp.h presence... yes
checking for netinet/tcp.h... yes
checking stdatomic.h usability... yes
checking stdatomic.h presence... yes
checking for stdatomic.h... yes
checking for library containing __atomic_load_4... -latomic
checking for threads.h... (cached) yes
checking sanitizer/asan_interface.h usability... 1 error generated.
make[6]: *** [Makefile:139: xsk_def_xdp_prog.o] Error 1
make[6]: *** Waiting for unfinished jobs....
  x86_64-openwrt-linux-gnu-g++ -mtune=skylake -march=skylake -O3 -funsafe-math-optimizations -ftree-vectorize -o /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/node-v16.15.1/out/Release/obj.target/v8_base_without_compiler/deps/v8/src/deoptimizer/deoptimize-reason.o ../deps/v8/src/deoptimizer/deoptimize-reason.cc '-D_GLIBCXX_USE_CXX11_ABI=1' '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-D__STDC_FORMAT_MACROS' '-DV8_TARGET_ARCH_X64' '-DV8_HAVE_TARGET_OS' '-DV8_TARGET_OS_LINUX' '-DV8_EMBEDDER_STRING="-node.21"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DOBJECT_PRINT' '-DV8_INTL_SUPPORT' '-DV8_ATOMIC_OBJECT_FIELD_WRITES' '-DV8_ATOMIC_MARKING_STATE' '-DV8_ENABLE_LAZY_SOURCE_POSITIONS' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' '-DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH' '-DV8_SNAPSHOT_COMPRESSION' '-DV8_ENABLE_WEBASSEMBLY' '-DV8_ALLOCATION_FOLDING' '-DV8_ALLOCATION_SITE_TRACKING' '-DV8_ADVANCED_BIGINT_ALGORITHMS' '-DICU_UTIL_DATA_IMPL=ICU_UTIL_DATA_STATIC' -I../deps/v8 -I../deps/v8/include -I/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/node-v16.15.1/out/Release/obj/gen/inspector-generated-output-root -I../deps/v8/third_party/inspector_protocol -I/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/node-v16.15.1/out/Release/obj/gen -I/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/node-v16.15.1/out/Release/obj/gen/generate-bytecode-output-root -I../deps/v8/third_party/zlib -I../deps/v8/third_party/zlib/google -I/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/include  -pthread -Wno-unused-parameter -m64 -Wno-return-type -fno-strict-aliasing -m64 -O3 -fno-omit-frame-pointer -fdata-sections -ffunction-sections -O3 -fno-rtti -fno-exceptions -std=gnu++14 -MMD -MF /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/node-v16.15.1/out/Release/.deps//home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/node-v16.15.1/out/Release/obj.target/v8_base_without_compiler/deps/v8/src/deoptimizer/deoptimize-reason.o.d.raw  -mtune=skylake -march=skylake -O3 -funsafe-math-optimizations -ftree-vectorize -fno-caller-saves -fno-plt -mtune=skylake -march=skylake -O3 -funsafe-math-optimizations -ftree-vectorize -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/node-v16.15.1=node-v16.15.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.1.0_glibc/usr/include -I/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.1.0_glibc/include  -c
  x86_64-openwrt-linux-gnu-g++ -mtune=skylake -march=skylake -O3 -funsafe-math-optimizations -ftree-vectorize -o /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/node-v16.15.1/out/Release/obj.target/v8_base_without_compiler/deps/v8/src/deoptimizer/deoptimized-frame-info.o ../deps/v8/src/deoptimizer/deoptimized-frame-info.cc '-D_GLIBCXX_USE_CXX11_ABI=1' '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-D__STDC_FORMAT_MACROS' '-DV8_TARGET_ARCH_X64' '-DV8_HAVE_TARGET_OS' '-DV8_TARGET_OS_LINUX' '-DV8_EMBEDDER_STRING="-node.21"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DOBJECT_PRINT' '-DV8_INTL_SUPPORT' '-DV8_ATOMIC_OBJECT_FIELD_WRITES' '-DV8_ATOMIC_MARKING_STATE' '-DV8_ENABLE_LAZY_SOURCE_POSITIONS' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' '-DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH' '-DV8_SNAPSHOT_COMPRESSION' '-DV8_ENABLE_WEBASSEMBLY' '-DV8_ALLOCATION_FOLDING' '-DV8_ALLOCATION_SITE_TRACKING' '-DV8_ADVANCED_BIGINT_ALGORITHMS' '-DICU_UTIL_DATA_IMPL=ICU_UTIL_DATA_STATIC' -I../deps/v8 -I../deps/v8/include - [... cut ...]
make[5]: *** [Makefile:20: libxdp] Error 2
make[4]: *** [Makefile:25: lib] Error 2
make[4]: Leaving directory '/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/xdp-tools-1.2.5'
make[3]: *** [Makefile:135: /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/xdp-tools-1.2.5/.built] Error 2
make[3]: Leaving directory '/home/username/works/openwrt/package/network/utils/xdp-tools'
time: package/network/utils/xdp-tools/compile#2.85#1.85#8.09
    ERROR: package/network/utils/xdp-tools failed to build.
```

dangowrt, 2022-07-11:
@trippleflux Do you have `CONFIG_BPF_TOOLCHAIN_BUILD_LLVM` or `CONFIG_BPF_TOOLCHAIN_HOST` set?

menmazqj, 2022-07-16:
> ```
> + curl -f --connect-timeout 20 --retry 5 --location --insecure https://codeload.github.com/xdp-project/xdp-tools/tar.gz/v1.2.5?/xdp-tools-1.2.5.tar.gz
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
> 100  247k    0  247k    0     0   491k      0 --:--:-- --:--:-- --:--:--  491k
> Found clang binary '/builder/shared-workdir/build/sdk/staging_dir/host/llvm-bpf/bin/clang' with version 13 (from 'clang version 13.0.0 (https://git.openwrt.org/openwrt/openwrt.git 265f402fbdd8444835b43954bf27a83bc9369b91)')
> libbpf support: system v0.8.0
>   perf_buffer__consume support: yes
>   btf__load_from_kernel_by_id support: yes
>   btf__type_cnt support: yes
>   bpf_object__next_map support: yes
>   bpf_object__next_program support: yes
>   bpf_program__insn_cnt support: yes
>   bpf_map_create support: yes
>   perf_buffer__new_raw support: yes
>   bpf_xdp_attach support: yes
> secure_getenv support: yes
> make[4]: Entering directory '/builder/shared-workdir/build/sdk/build_dir/target-powerpc_8540_musl/xdp-tools-1.2.5'
> 
> lib
> 
>   libxdp
>     M4       xdp-dispatcher.c
>     CLANG    xsk_def_xdp_prog.o
> xsk_def_xdp_prog.c:4:10: fatal error: 'bpf/bpf_helpers.h' file not found
> #include <bpf/bpf_helpers.h>
>          ^~~~~~~~~~~~~~~~~~~
> 1 error generated.
> make[6]: *** [Makefile:139: xsk_def_xdp_prog.o] Error 1
> make[5]: *** [Makefile:20: libxdp] Error 2
> make[4]: *** [Makefile:25: lib] Error 2
> make[4]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-powerpc_8540_musl/xdp-tools-1.2.5'
> make[3]: *** [Makefile:131: /builder/shared-workdir/build/sdk/build_dir/target-powerpc_8540_musl/xdp-tools-1.2.5/.built] Error 2
> time: package/feeds/base/xdp-tools/compile#3.97#0.49#7.57
> ```
> 
> @dangowrt @tohojo
> 
> Or ARM: https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a7/base/xdp-tools/compile.txt

I got the same error, but, emm... , I bypass this include error by adding include directory manually.
That is, locating at `openwrt/build_dir/target-x86_64_musl/xdp-tools-1.2.5/lib/libxdp/Makefile`, then adding
 `-I$(STAGING_DIR)/usr/include` to `BPF_CFLAGS` . Like this, `BPF_CFLAGS += -I$(HEADER_DIR) -I$(STAGING_DIR)/usr/include`
That is not elegant... but works.

PolynomialDivision, 2022-07-16:
@menmazqj https://github.com/openwrt/openwrt/pull/10223

jwh, 2023-10-12:
This is still broken btw, did anyone get to the bottom of the asm/types.h issue? I can't figure out how to fix this as the correct headers are just not installed in staging_dir.

x86_64 glibc host, x86_64 musl target