openwrt/openwrt #10209: Is there any plan for nanopi r5s?

org0000h, 2022-07-07:


org0000h, 2022-07-07:
I am friendly and I think you are friendly too. I know the forum.
I just want to know if the member of openwrt is curious about the this device instead of talking about it.

pkgadd, 2022-07-07:
As tapper suggested, this is a bugtracker, not a forum, nor a feature wishlist. Keeping the information lean and concise if the only way to keep it useful to the developers, otherwise real bugs are overseen and rot among badly disguised feature requests.

<hr>

There are, by definition, no 'plans' to support any device. Device support happens, if someone with the device on their desk starts working on porting it for their needs, there is no project budget to buy 'every device on the market' nor any work assignments to get a device supported 'by noon, or else'.

If someone with the device on their desk (perhaps you yourself) submits a pull request/ patch series to add support for this device to OpenWrt, chances should be good that it will be merged.

org0000h, 2022-07-08:
> As tapper suggested, this is a bugtracker, not a forum, nor a feature wishlist. Keeping the information lean and concise if the only way to keep it useful to the developers, otherwise real bugs are overseen and rot among badly disguised feature requests.
> 
> There are, by definition, no 'plans' to support any device. Device support happens, if someone with the device on their desk starts working on porting it for their needs, there is no project budget to buy 'every device on the market' nor any work assignments to get a device supported 'by noon, or else'.
> 
> If someone with the device on their desk (perhaps you yourself) submits a pull request/ patch series to add support for this device to OpenWrt, chances should be good that it will be merged.

org0000h, 2022-07-08:
Thanks for your reply very much.I think I got it.

closed 2022-07-08 by org0000h