openwrt/openwrt #10216: luci-app-wireguard: Invalid Error Message, Refresh

arrmo, 2022-07-08:
Hi,

I have noticed that if I have WireGuard configured, but handshake has not occurred, the status shows,
```
No WireGuard interfaces configured
```

But that's not really correct. It's not connected (vs. not configured). Perhaps update this wording? I don't mind updating it, if you'd prefer.

I have also noticed, that in this state refresh is not really updating. If it connects ... you have to manually refresh. This one seems a bit more odd. Any reason why it wouldn't refresh? And it seems that state changes are what cause this to not refresh properly?

Thanks!

arrmo, 2022-10-07:
Sure, NP! Is there a way to transfer it there? I admit, I wasn't sure quite where to open it.

Thanks!

f00b4r0, 2022-10-07:
There isn't. Best close here and open anew there. Thanks!

arrmo, 2022-10-07:
Will do! Just one thing ... move to packages, or https://github.com/openwrt/luci/issues?

Thanks!

f00b4r0, 2022-10-08:
> Will do! Just one thing ... move to packages, or https://github.com/openwrt/luci/issues?

The latter, good catch, my bad!
Thanks

arrmo, 2022-10-12:
Done!

closed 2022-10-12 by arrmo