openwrt/openwrt #10219: brcmfmac for bcm4366c0: Wi-Fi won't work on VLAN Filtering Enabled Bridge

arinc9, 2022-07-08:
When connecting to the Wi-Fi network, phone claims incorrect password was given. However, that's not the case.

I can see the Wi-Fi interface properly added to the VLAN filtering enabled bridge:

```
# bridge vlan show dev wlan0-1
port              vlan-id  
wlan0-1           1 PVID Egress Untagged
```

Config:
```
config wifi-iface 'wifinet1'
        option device 'radio0'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'psk2'
        option key 'password'
        option network 'internet'
```
```
config device
        option type 'bridge'
        option name 'br0'
        list ports 'extsw'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        list ports 'wan'

config bridge-vlan
        option device 'br0'
        list ports 'wan:t'
        option vlan '1'

config interface 'internet'
        option proto 'dhcp'
        option device 'br0.1'
```

Wi-Fi works on a bridge where VLAN filtering is not enabled.
As a workaround, I put the locally created VLAN interface (e.g. br0.1) on a new bridge where VLAN filtering is not enabled. Then I use that new bridge as the interface for LAN. Wi-Fi interface will be automatically added to the VLAN filtering disabled bridge interface of LAN.

Me and @scatman75 experience this issue on Asus RT-AC88U.

Tested on 22.03.0-rc5 and SNAPSHOT (r20023-21825af2da).

arinc9, 2022-07-08:
paging @hauke

asherman07, 2022-07-24:
I am also having this same issue on bcm53xx/generic platform. I also have is set up as a semi smart ap with no firewall and no dhcp/dns and just vlan filtering. I also have set up the interfaces  with DHCP and the interface are able to get addresses no problem. I run into the same issue with incorrect password with a wifi network attached to a vlan interface. I do notice i am able to get an address and browse the web with no security on the network. The work around posted by @arinc9 with creating a new bridge with the vlan device that was created by the original bridge does work.

asherman07, 2022-07-25:
Well after some more testing and working with @onemacrfifty we have determined it be a bug as well. The work around did work until I rebooted and lost everything, but this could have been my fault with a wrong config. I have since reverted back to V21. Since my Netgear R8000 looks like it doesn't fully support DSA, I create a new vlan device with the parent the same as the others. I then i create a new bridge with the vlan then attach to an interface. I then set up the swconfig with the correct vlan tagging and it works. 

I do hope this is fixed by the release as it is much easier to implement.

madmic1314, 2022-08-16:
@asherman07 are you able to share your config? I also have the same issue, but no option to revert to v21 as the Asus RT88U was only supported in v23.

scatman75, 2022-08-16:
@madmic1314 
You can install version 21.02.02. A pre-compiled image is available at https://github.com/arinc9/openwrt/releases/tag/v21.02.2-asus_rt-ac88u.

arinc9, 2022-08-16:
The issue is there on OpenWrt 21.02 as well, if I remember correctly.

scatman75, 2022-08-16:
@arinc9 You are absolutely right. @madmic1314 wanted to revert to v21...so be it. Hope, that the reverting-wish was not associated with the use of swconfig, as described from @asherman07 for the Netgear.

madmic1314, 2022-08-16:
> @madmic1314 You can install version 21.02.02. A pre-compiled image is available at https://github.com/arinc9/openwrt/releases/tag/v21.02.2-asus_rt-ac88u.

Wasn't aware of this option. Was looking at the [firmware selector](https://firmware-selector.openwrt.org/?version=22.03.0-rc4&target=bcm53xx%2Fgeneric&id=asus_rt-ac88u) for supported version. This implied 23.03.0-rc4 was the first supported version.

If the issue was also with v21 I'll stay where I am for the moment, hence me asking @asherman07 what config was used - but seems that is legacy now and not available in v23. Guess I'm stuck here for the moment until this is fixed.

madmic1314, 2023-04-21:
I figured it out - it's not too daunting once you get it going, but getting there took a bit of fiddling.

Prep:
A) Be connected via a cable on ports 1-4. Do not tag this port when doing step 1, else you'll be locked out. Do not use WIFi for this.

B) I will assume you're only creating a Guest VLAN - you can simply follow this guide from step 2 onwards for any additional VLANs.

1.) Enable VLANs on your main bridge device "br-lan" - this is my setup for my needs, your tagging / untagging may differ...
[image]

2a) Create a new Bridge Device "Guest_WiFi" on the Guest VLAN created in step 1.  In bridge VLAN filtering tab ensure VLAN filtering is **NOT** ticked!
[image]

[image]

2b) Save and apply config

3a) Create your new Guest interface, set the device to "Guest_WiFi", set all other options as normal (DHCP, IP space, etc.)
[image]

3b) Save and apply config

4a) Create new WiFi SSID, under Interface Configuration, set Network to "Guest"
[image]

4b) Save and apply config

What you're looking for is the Guest WiFi interface to show on the Guest interface:
[image]

I had to reboot my router for this to happen. Then all was working ok.

A bonus note from my experience - you'll need to set the firewall zone rule and firewall rules to get traffic to pass between zones. For Guest I have:
[image]

raenye, 2023-10-03:
Introducing the extra bridge indeed solves the handshake problem, but it seems to open the door for fdb issues ☹️ 
https://github.com/openwrt/openwrt/issues/11650

pfugl, 2024-12-14:
Same Problem still faces on  on 24.10rc2 RT-AC88U. Any change of a solution other then thr circumvention above?

zhanko73, 2024-12-19:
Quick question.  
**What about the VLAN tagging on WLANs? Does the VLAN tagging on wifi work?** 

I have Asus RT-AC88U. Once somehow I was able to use VLAN on wifi without password. But I changed a setting and suddenly I no longer was able to access the OpenWRT so I had to upload firmware again. (Next time I will connect a cable to serial pins to see what is happening there exactly.)
I tried several options even to create vlan interface for the radio0 and radio1 device (or whatever it called) and assign those to the trunk,  attempted with untagged, tagged, pvid setting  but as per traffic check it does not tag the packets it sends to the trunk. So again: I setup on bridge to tag it (I think the untagged and pvid setup together means that)  but it simple ignored it.
I spent many hours without success.
I think this was on the on of the  23.x  version but since that I replaced openwrt a few times.

sixerstyle, 2025-04-28:
April 2025 and the bug is still there :(

raintonr, 2025-05-03:
> Introducing the extra bridge indeed solves the handshake problem, but it seems to open the door for fdb issues ☹️ [#11650](https://github.com/openwrt/openwrt/issues/11650)

You actually don't have to create a bridge and can just add VLAN (802.1q) device with the correct base interface and VLAN ID then use that.

Funnily enough I have all this working just fine when using different WiFi networks (different ESSID) but when I want clients to roam between access points (ie. same ESSID & security) it doesn't work when one is untagged and another is on the other end of a VLAN connection like this. I get a `received packet on xxx-xxx with own address as source address` error. 

That error seems logical enough (erm... I think!) because packets with VLAN ID are probably coming into the bridge and being sent to the WiFi interface and somehow messing things up. What I think is required is a way to use VLAN filtering that includes the WiFi interface in that VLAN filtering setup so you can set that WiFi interface to untagged which (if I understand this) would remove the VLAN ID and make both WiFi networks just look like they are untagged. I'm guessing really, but sounds plausible. Sadly I cannot test this as when you add WiFi a bridge it doesn't appear in the VLAN filtering page.

Does anyone know any other way to tell OpenWRT that a WiFi interface should be untagged and have no VLAN ID?

raintonr, 2025-05-04:
> What I think is required is a way to use VLAN filtering that includes the WiFi interface in that VLAN filtering setup so you can set that WiFi interface to untagged which (if I understand this) would remove the VLAN ID and make both WiFi networks just look like they are untagged.

Please ignore this. Yes, I was guessing, and seems I was wrong. Even removing VLANs from the equation, I cannot get WiFi roaming to work and see the same `received packet on xxx-xxx with own address as source address` error.