openwrt/openwrt #10224: 22.03.0 IPv4 PPPoE software flow offload not working on Archer C7v5 and Deco M4Rv2

Ashraaf, 2022-07-08:
Upgraded from 22.03.0-rc1 to rc4 with fresh config on my Archer C7v5 and I noticed my speedtest dropped from 500/100 to 240/100.
Also occur on my Deco M4Rv2 with Openwrt 22.03.0-rc4 firmware installed. 

Then I flashed back my Deco to stock firmware, put it in router mode and the speed is back to normal. 
Later I tried connecting my Archer WAN port to stock Deco LAN port, and use Deco dhcp ip as Archer WAN ip with NAT and software flow offloading enabled, and I get 500/100 speed, the problem with offloading only occur when using pppoe and not dhcp.

I already tried with rc5 and latest snapshot 1 day before rc5 released and its still broken.

Network config
```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0.1'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option delegate '0'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'

config interface 'wan'
	option device 'eth0.500'
	option proto 'pppoe'
	option peerdns '0'
	option delegate '0'
	option ipv6 '0'
	option username '****'
	option password '****'
	list dns '8.8.8.8'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '0t 2 3 4 5'
	option vid '1'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '0t 1t'
	option vid '500'
```

Firewall config
```
config defaults
	option input 'REJECT'
	option output 'REJECT'
	option forward 'REJECT'
	option synflood_protect '1'
	option drop_invalid '1'
	option flow_offloading '1'

config zone
	option name 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	list network 'lan'

config zone
	option name 'wan'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'
	list network 'wan'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'
```

omarbotta, 2022-09-11:
After the update to 22.03.0, the bandwidth in the Speedtest.com dropped from 500 Mbps to 205 Mbps.

I had previously solved this by activating "Software flow offloading", but apparently this solution no longer works. 

I'm using a TP-Link Archer C6 v2 (US)

omarbotta, 2022-09-11:
Here I leave screenshots of the speed tests.

Connected directly to the ONT of the ISP.

[image]

And connected through the OpenWRT router

[image]

EchedeyLR, 2022-09-11:
@Ashraaf, would you please change the title to "22.03.0 IPv4 PPPoE software flow offload not working" to reflect that is affecting several devices and maybe more unknown by now?

hajosattila, 2022-09-17:
Same problem here. (Archer C7)

EchedeyLR, 2022-09-18:
I have seen that other issues related to 22.03 version were light-tagged as `[22.03]`. Maybe ours should be similar too.

EchedeyLR, 2022-09-24:
@dangowrt, @jow- and @stintel, I have seen you participated in flow offloading support, testing and backporting for this last released including solving issues related to Firewall4.

Do you know something that could have caused this issues related to, by now, TP-Link hardware? If yes, is this being taken into account and there is someone assigned to work around it?

muzuiget, 2022-09-30:
Hi, I think I also have this issue on my [WNDR4300v1](https://openwrt.org/toh/netgear/wndr4300) router.

My ISP bandwidth is 1000m and I do the speed-test on the ISP testing page with a modern Wifi-6 router, it can reach full 1000Mbps speed.

On my old WNDR4300v1 router, the WAN interface is 1000m, but there is a CPU bottleneck on NAT:

* OpenWrt 21.02.3 + "Disable Software flow offloading": about 150Mbps
* OpenWrt 21.02.3 + "Enable Software flow offloading": about 450Mpbs (300m faster)
* OpenWrt 22.03.0 + "Disable Software flow offloading": about 150Mbps (same as 21.02)
* OpenWrt 22.03.0 + "Enable Software flow offloading": about 170Mbps (only 30m faster)

Some status info on LuCI:

```
Model | Netgear WNDR4300
Architecture | Atheros AR9344 rev 2
Target Platform | ath79/nand
Firmware Version | OpenWrt 22.03.0 r19685-512e76967f / LuCI openwrt-22.03 branch git-22.245.77528-487e58a
Kernel Version | 5.10.138
```

goligo, 2022-10-07:
I can confirm the same issue with 22.03.0 on a FritzBox 4040 - with 21.03.1 I could max out my internet connection with 267MBit/s, after installing 22,03.0 I do get 155MBit/s max.

hajosattila, 2022-10-07:
I switched back to 22.03.0-rc1

omarbotta, 2022-10-07:
I switched back to 21.02.3 Old Stable series

hajosattila, 2022-10-08:
> I switched back to 21.02.3 Old Stable series

Software Flow Offload still works in 22.03.0-rc1.

goligo, 2022-10-08:
Actually I forgot that I had irqbalance installed before. After installing it again, I am back at full speed. Another option seems to enable packet steering in global network options, to make better use of multiple cores.

Ashraaf, 2022-10-08:
Using irqbalance and packet steering won't help me at all since both my device Archer c7 v5 and Deco M4r v2 use Qualcomm Atheros QCA9563 which is a single core cpu.

The problem here is with firewall4(nftables instead of iptables) which is is the default since Openwrt 22.03. Somehow its flow offload doesn't play nice with PPPoE. Back then when Im still on 22.03rc5, I tried manually downgrading firewall4 package from the archive to older rc's version until my speed is back to normal. Forgot which version it is, but its start to broke between rc1 to rc2.

Yodaregs, 2022-10-09:
I have the same issue on 22.03 on TP-Link Archer C7 v5, and downgraded to 21.02 until this is fixed.

KinteLiX, 2022-10-09:
I have the same issue on 22.03 on my Deco M4R v1 routers.
[image]

laduchesneau, 2022-10-10:
I have the same issue on x86/64build ( PC Engines apu4 with AMD GX-412TC SOC)

daschiller, 2022-10-10:
Is there a process for getting this onto the known issues section of the release notes?

TeRa23, 2022-10-12:
Same issue here...

Conn Type:	PPPoE /w IPv6 DualStack
Model	TP-Link Archer C7 v5
Architecture	Qualcomm Atheros QCA956X ver 1 rev 0
Target Platform	ath79/generic
Firmware Version	OpenWrt 22.03.0 r19685-512e76967f / LuCI openwrt-22.03 branch git-22.245.77528-487e58a
Kernel Version	5.10.138

laduchesneau, 2022-10-13:
So the issue was added between rc1 and rc2. Doing some basic troubleshooting by applying commit after commit, I was able to find that the issue was introduced in this commit: [kernel: fix flow offload issues with pppoe ](https://github.com/openwrt/openwrt/commit/53fc6e9edeae7364847bd1a7a81a331feeb56172)

I could be wrong.

laduchesneau, 2022-10-13:
Trying to find the root cause: Can others confirm that you are using PPPOE over VLAN ?

TeRa23, 2022-10-13:
> Trying to find the root cause: Can others confirm that you are using PPPOE over VLAN ? 
Yes

Protocol: PPPoE ; Device: eth0.3105

laduchesneau, 2022-10-14:
I have done some basic test, using a switch to untag the VLAN and configuring the PPPOE interface on openwrt without VLAN.

The issue persists.

Yodaregs, 2022-10-14:
I have this issue and am not usin any VLANs ...

whiskerp, 2022-10-24:
This is probably the same as #10828 which I reported on my Lantiq device (BB HH 5A)

whiskerp, 2022-11-15:
Is this problem not serious enough to warrant any attention? It is a blocker against adopting 22.03.

kozik47, 2022-11-18:
Same issue Archer C7 v2 after upgrade to 22.03.3. On my 250/125 line without software flow offload getting 170 Mb/s downstream and with it enabled just shy of 200 with it enabled (with CPU load at 100%).

Reverted to 21.02.5 where I get 270 Mb/s down with CPU load peaking at 70% with software flow offload enabled.

theodiem, 2022-11-26:
Can also confirm on a Fritzbox 7360SL (Lantiq XWAY VRX288).
22.03.2 firewall.cfg01e63d.flow_offloading=0  -> Speedtest from a computer in LAN->WAN downstream max. ~50Mbit
22.03.2 firewall.cfg01e63d.flow_offloading=1  -> Speedtest from a computer in LAN->WAN downstream max. ~68Mbit (flow_offloading_hw=0/1 made no difference)
21.02.5 firewall.cfg01e63d.flow_offloading=0  -> Speedtest from a computer in LAN->WAN downstream max. ~50Mbit
21.02.5 firewall.cfg01e63d.flow_offloading=1  -> Speedtest from a computer in LAN->WAN downstream max. ~110Mbit (contract speed).

Using dsl0.40 interface as I need VLAN tagging

mkopec, 2022-11-28:
I believe I'm seeing the same issue on a pcengines apu2 board (x86-64), with PPPoE over tagged VLAN, performance drops to circa 600Mbps, while on 21.02 i'm easily getting gigabit speeds.

randrrandr1, 2022-12-17:
It seems ` Xiaomi mi router 4A (MT7621DAT) ` has no working hw nat with pppoe either. 
High (90%) cpu usage under 300 Mbit/s load
` OpenWrt 22.03.2, r19803-9a599fee93 `

whiskerp, 2022-12-17:
> This is being actively worked on in master branch. Expect a backport to
> openwrt-22.03 once things have stabilized. In the meantime, you can help
> testing master branch and report if things are improving.

David, I've looked at the git and there doesn't seem to have been any pppoe
work since May. Maybe I'm looking in the wrong place?

Thanks
Peter

whiskerp, 2023-01-03:
The last pppoe flow offload commit I see was cb1dc49 on 6th May 2022. This is still a major issue which makes the use of openwrt 22.03 unusable on many hardwares.

https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=726ef8ba2dbe4d4a693c4d9300bc69e234e6d67d

whiskerp, 2023-01-12:
Very sad to see this is *still* not fixed on release 22.03.3

OpenWRT 21.02.5 (BT Homehub)
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  72.9 MBytes  61.2 Mbits/sec   65             sender
[  4]   0.00-10.00  sec  72.2 MBytes  60.6 Mbits/sec                  receiver

OpenWRT 22.03.3 (same hardware)
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  89.3 MBytes  74.9 Mbits/sec   21             sender
[  4]   0.00-10.00  sec  88.7 MBytes  74.4 Mbits/sec                  receiver

daschiller, 2023-01-12:
It looks like none of the core devs use flow offloading.

Spudz76, 2023-01-12:
I have a normal ISP that doesn't use PPPoE so... I guess you have to await a dev that has to use DSL or whatever (and doesn't just allow the modem to do the PPPoE and deliver a routed normal IP on the WAN).  Probably rare.  Difficult to make a test environment.

patrakov, 2023-01-13:
Testing without a "real" PPPoE provider is not difficult at all. All you need is a Linux box with `rp-pppoe` installed.

Spudz76, 2023-01-13:
Oh well if it's that simple then... *All you need is* a Dev that feels like doing that, and has affected hardware, which they wouldn't know unless they already used pppoe, which leads us back to my previous statement.

daschiller, 2023-01-13:
All FTTH setups in my country (Central Europe) use PPPoE from the ONT to the router, at least as far as I know.
It shouldn't be that uncommon.

whiskerp, 2023-01-13:
A lot of UK FTTC is also PPPoE over VDSL2.

Peter

Spudz76, 2023-01-14:
It actually should be uncommon, there is no reason to use PPPoE unless ISP is so lazy they must use RADIUS or something similarly outdated.  Especially on non-copper.  FTTC using it for the last ten feet sounds hilariously bad.

EchedeyLR, 2023-01-14:
> It actually should be uncommon

What do you mean?
* Both Telefonica and Vodafone uses PPPoE on both optic fiber connection VDSL and ADSL. Telefonica does it in all their installations and Vodafone at least in Spain.
* Antel, which is main provider in Uruguay does that for all connections in both ADSL and optic fiber.

Spudz76, 2023-01-14:
I didn't say it doesn't exist, I said it ought not exist by now.

Here is a blog representing my same opinion [from a decade ago](https://netequalizernews.com/2010/09/22/pppoe-may-be-outdated/)

It's a stupid layer of overhead that is not really needed, and I take it as a sign of a lazy ISP.  Especially if it's fiber.  It's like plain-old-telephone technology from the 1900s.  I find it about as dumb as electric vehicles having a speaker to make exhaust sounds.

For the most part, unless you're forced to use copper DSL (rural), PPPoE hasn't been a thing in the USA for a long time (last time I had to deal with it was... 2001?).  I associate PPP with dial-up modems, why would you do it "oE" and add useless overhead where it isn't needed?  Oh not to mention it's the layer you're all having issues with, imagine if you could just plug-in and DHCP and you're on?

Lawstorant, 2023-01-14:
I get what you're saying and agree wit you, but this is not a forum to discuss such topics. Orange is the biggest ISP in Poland (by numbers) and still uses PPPoE on their FFTH as they still probably operate upgraded infrastructure from the 90'.  OpenWRT supports PPPoE usecase (as it should because it's still widely used) and currently something that worked in 21.02 is broken and affects many users.

I can also report that Flow Offload is broken with PPPoE on PCEngines APU2 (x86 based).

Spudz76, 2023-01-14:
Just saying, maybe it would be better for everyone if you pressured these backward ISPs to stop with the ancient tech just because it's easier for them.  It's not easier for anyone else.

And back to an earlier question, why not just have PPPoE wrapping and unwrapping inside your "modem" (CPE) and then OpenWRT simply DHCP's like normal Ethernet ISP?  Customers should never need to know they even use PPPoE nor have to hassle with it.   The ISP should feel ashamed for using PPPoE and should attempt to hide it, and it should be a deal breaker for customers if they even need to know it's involved.

Spudz76, 2023-01-14:
Oh and, the whole reason it came up was the question "why hasn't this been fixed yet" so I think that makes this exactly the place to discuss such topics.

Same reason I haven't fixed my FAX machine... I haven't had or needed one since 20 years ago, and nobody should.

omarbotta, 2023-01-14:
> Oh and, the whole reason it came up was the question "why hasn't this been fixed yet" so I think that makes this exactly the place to discuss such topics.
> 
> Same reason I haven't fixed my FAX machine... I haven't had or needed one since 20 years ago, and nobody should.

But nevertheless it is still used a lot, but it seems that since it is not used in the US, those of us who depend on PPPoE connections should stop using OpenWRT.

Things that are well supported in proprietary firmwares, which since OpenWRT have decided are so modern that they don't want to support them properly.

omarbotta, 2023-01-14:
> 

> Just saying, maybe it would be better for everyone if you pressured these backward ISPs to stop with the ancient tech just because it's easier for them. It's not easier for anyone else.
>
> And back to [...]

So you suppose that it is easier to get all the ISPs in the world to stop using PPPoE than for you to fix what you broke in the new version?

Spudz76, 2023-01-14:
Nobody ever said `wontfix`, and I'm no official.  Just a lurking dev that fixes things that affect myself, mostly.  You know, how open source works.  I do not need PPPoE and will not be fixing it.  Apparently none of the others do either, or it would have been fixed by now.  Hey maybe one of you that has the problem could fix it, you know, how open source works?  Or work out another way around it (use one of those "well-supported" devices to unwrap the PPPoE and plug OpenWRT in as if you had a normal Ethernet ISP).

Interesting you would prefer to boycott in the wrong direction.  Customers must argue with their ISP to make PPPoE stop, not argue with technology that wants to move forward.

And it's nowhere near "all the ISPs in the world" since I haven't needed to even think about PPPoE in 20 years, and would not use an ISP if they used it unnecessarily.  Similar to if an ISP hadn't updated to PPP from SLIP back in oh 1990.  PPP is for serial lines (like SLIP was) and adapting it to work on Ethernet was a dumb stopgap in the first place, so that the same RADIUS database could authorize both dial-up and Ethernet customers.  Now that these ISPs don't have any dial-up services, PPP should be completely dead, including PPPoE.

EchedeyLR, 2023-01-14:
> Just saying, maybe it would be better for everyone if you pressured these backward ISPs to stop with the ancient tech just because it's easier for them. It's not easier for anyone else.

And meanwhile I need to be able to use it, don't you think?

It is not my fault that as things are easy for you, you consider magically that this should be in our side to be fixed. That is an elitist behavior and that is out of any "discussion" or how you want to call it.

EchedeyLR, 2023-01-14:
>  Hey maybe one of you that has the problem could fix it, you know, how open source works? Or work out another way around it (use one of those "well-supported" devices to unwrap the PPPoE and plug OpenWRT in as if you had a normal Ethernet ISP).

Yep, incredible seeing this elitist behavior.

The next pattern will be the next level ableist comment of "you are not a dev, just fluff yourself".

champtar, 2023-01-14:
Please all focus on the bug.
For people affected, maybe someone could try master with 5.15 kernel to see if there are any changes.
@dangowrt any good technical news ?

omarbotta, 2023-01-14:
> Nobody ever said `wontfix`, and I'm no official. Just a lurking dev that fixes things that affect myself, mostly. You know, how open source works. I do not need PPPoE and will not be fixing it. [...]

OK, your position is still ridiculous, because there are millions of potential OpenWRT users who will not be able to upgrade to the new version, because you see that it is used in a large part of Europe, Latin America, and other places.

I'm just a user, if I could fix it I would have already tried it... But it seems that there is a certain insensitivity in the OpenWRT development team with the issue.

And it is not true that the use of PPPoE is a minority, as far as I understand many ISPs maintain it to change from copper to fiber and have backward compatible configurations.

Maybe the solution is to stop using OpenWRT because it is elitist software

Spudz76, 2023-01-14:
LOL, whatever.  bye

omarbotta, 2023-01-14:
> Please all focus on the bug. For people affected, maybe someone could try master with 5.15 kernel to see if there are any changes. @dangowrt any good technical news ?

Thank you so much, finally someone trying to do something, and not just throw elitist comments

Spudz76, 2023-01-15:
Dang those elitists that think a Ford Model T is defunct and people should get real cars.

whiskerp, 2023-01-15:
When an ISP has rolled out VDSL2 with PPPoE to millions or tens of millions
of routers years ago it's not an overnight or cost-free change to
seamlessly re-architecht their network just because an open source third
party is a bit snooty about supporting PPPoE...

OpenWRT 22.03 is dead in the water for a considerable proportion of users
until this is fixed.

Peter

mhofstra, 2023-01-16:
> Please all focus on the bug. For people affected, maybe someone could try master with 5.15 kernel to see if there are any changes. @dangowrt any good technical news ?

This issue still exists with kernel 5.15 from latest snapshot. It is hardware agnostic unless you have hardware offload.

I hope to find more time to dig into this, but from what I can tell the flowtable is populated correctly. The subsequent packets aren't taking the "offload fastpath" route though.

champtar, 2023-01-16:
At least for software offloading I think OpenWrt now uses upstream code, so someone could reproduce the issue with a normal distro and simple nft rules and report upstream ?

SpyrosBarbatis, 2023-02-28:
I had the same issue on an archer c7 v5 running on 22.03.3, i wasted so many days to find out why isp provided CPE is faster, why archer's cpu hits 100% when downloading until i found this thread. I reverted back to 21.02.5 and now everything back to normal, faster speeds and less cpu.

kings-way, 2023-03-01:
It seems that PPPoE WAN connection is not the only one affected.

I have two devices, which run on  MT7621A (Redmi AC2100) and MT7622B (Redmi AX6S). No matter what the type of WAN connection is, either PPPoE or DHCP, they both perform much worse with sw flow offloading.

Althouth these two devices have hardware flow offloading and it works, but it does not support WLAN interfaces, so I have to fallback to sw flow offloading to accelerate WiFi connections.

Btw, fallback to 22.03.0-rc1 does solve the problem.

rafaelfini, 2023-03-08:
Weirdly, reverting to 22.03.0-rc1, as many in this thread, didn't solve the issue for my Archer C7. I had to go back to 21.02.5 for SFO to work. 

I hope someone fix this.

[236 more replies]