openwrt/openwrt #10225: ath10k "did not acknowledge authentication response"

XDjackieXD, 2022-07-08:
When running an IsoStation AC (identical hardware to the PowerBeam 5AC Gen2) in AP mode on DFS channels it happens that after a random time interval (I observed anything from seconds to half a day) it hangs and no clients can connect anymore.
The AP spams the line `Fri Jul  8 21:50:50 2022 daemon.notice hostapd: wlan0: STA de:ad:be:ef:01:23 IEEE 802.11: did not acknowledge authentication response` until the radio is restarted from LUCI (then everything works again until the problem occurs again).

I tested it with both the current master (fae3ac3560459320a88be86b31c572c4bca42645) and the current release candidate (2022.03.0-rc5) and will test with stable 21.02.03 this weekend.
I'm running ath10k-smallbuffers (non-ct) as this is what I had the best experience with so far (mostly on ipq4018 devices where I'm running openwrt 21.02 with ath10k non-ct since months without any problems).
Also I will also try ath10k-ct if downgrading to stable does not solve the issue.

XDjackieXD, 2022-07-09:
selecting ath10k-ct-smallbuffers on current master seems to not have this problem but also not flawless (a lot of DFS false positives resulting in huge downtimes of multiple minutes every few hours because it runs out of channels to use...)

XDjackieXD, 2022-09-11:
Just tested again with 22.03.0 with the same results. ath10k-smallbuffers has the mentioned problem, ath10k-ct-smallbuffers works.

cheng3100, 2023-06-01:
Hi @XDjackieXD ,
I have a similar issue recently on a ath10k pci router.  So from what you say it should be a driver issue?

The ath10k-ct-smallbuffer driver work but the ath10k-smallbuffer driver fail?
Do I understand correct?