openwrt/openwrt #10226: DFS issues with latest release candidate

XDjackieXD, 2022-07-08:
When selecting a DFS channel on the current rc (22.03.0-rc5) the radio will not switch channel on a DFS event but rather just disable the radio completely.
The workaround I found was to set a hand full of channels as option `channels` in uci and set the option `channel` to auto. With this DFS is working correctly and it switches to another channel of the given list (which LUCI sadly does not have an ui for).
This is a regression to 21.02.03 stable where a DFS channel switch is also happening (as expected) if a channel is manually selected and no channel-list is set.

example wireless config with workaround:
```
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'pci0000:00/0000:00:00.0'
	option band '5g'
	option channel 'auto'
	option channels '136, 140, 132, 116, 112, 108, 104, 100'
	option txpower '6'
	option country 'AT'
	option cell_density '3'
	option distance '7000'
	option htmode 'VHT20'
```

viniribeirossa, 2022-07-10:
Remove quotes between channels. Use spaces only
option channel 'auto'
option channels '136 140 132 116 112 108 104 100'

XDjackieXD, 2022-07-10:
The commas make absolutely no difference (I just tested both variants and they work exactly the same).
I also just tested if the 'auto' setting is really required but it behaves the same if I set some channel as the `channel` option while having the `channels` list set (but it does not seem to start at the selected channel but rather a random one from the list).
The problem is that DFS channels are defacto unuseable without manually setting this fallback list (there is no default value for this list set and in case of a DFS event it just disables the radio in this case) and there is no clear indication why that is the case.
As neither documentation nor LUCI make the user aware of this, it makes DFS channels (there are also indoor channels requiring DFS!) useless.

I'd suggest to either set a fallback list per default (for example if an indoor dfs channel is set, set all indoor channels as fallback and if an outdoor channel is set set all outdoor channels as fallback), have a LUCI GUI for configuring this list or at least have LUCI point out that this could be a problem when the user selects a DFS channel.

DragonBluep, 2022-08-13:
> When selecting a DFS channel on the current rc (22.03.0-rc5) the radio will not switch channel on a DFS event but rather just disable the radio completely.

Confirmed to have this issue on the MT7610E.
```
[   55.181204] br-lan: port 3(wlan0) entered blocking state
[   55.191964] br-lan: port 3(wlan0) entered disabled state
[   55.203108] device wlan0 entered promiscuous mode
[   55.212738] br-lan: port 3(wlan0) entered blocking state
[   55.223374] br-lan: port 3(wlan0) entered forwarding state
[   55.483680] br-lan: port 3(wlan0) entered disabled state
```